moggieuk/Happy-Hare #204: Happy-Hare with Klipper on Repetier-Server

jgbrown54, 2024-02-26:
I don't know if you have had any experience with Repetier-Server. It runs as the host instead of running Octoprint. I have an implementation of Klipper running on Repetier-Server. It installs slightly different than under Octoprint. The file locations are as follows:

Klipper Installation Path: /opt/klipper/LK5_Pro_ERCF
Printer configuration: /var/lib/Repetier-Server/database/klipper/LK5_Pro_ERCF.cfg
Printer socket: /var/lib/klipper/sockets/LK5_Pro_ERCF

I'm trying to install Happy-Hare with the following command:

./install.sh -k /opt/klipper/LK5_Pro_ERCF -c /var/lib/Repetier-Server/database/klipper -m /opt/klipper/LK5_Pro_ERCF/moonraker -i

And I get the following error:

Running on 'main' branch
Checking for updates...
Already the latest version: v2.4.2-90-gdfdf1ff
[sudo] password for pixie:
Klipper config directory (/var/lib/Repetier-Server/database/klipper) found
Klipper service not found! Please install Klipper first
pixie:Happy-Hare:%  

Klipper is installed. The printer works fine but it is unable to locate the Klipper Service.

Any assistance would be appreciated. If I can get it to work, I'll provide a writeup of how to install Happy-Hare on a Repetier-Server installation of Klipper.

jgbrown54, 2024-02-26:
Issues in nutshell. Since Repetier-Server supports multiple installations of Klipper, they separate them based on what they call a Printer Stub. The Stub is the name of the printer as added into Repetier with the spaces replaced with underscore. So in my case, the printer name is LK5 Pro ECRF so the Stub is LK5_Pro_ECRF. So the proper locations are:

Klipper Installation Path: /opt/klipper/LK5_Pro_ERCF
Printer configuration: /var/lib/Repetier-Server/database/klipper/LK5_Pro_ERCF.cfg
Printer socket: /var/lib/klipper/sockets/LK5_Pro_ERCF
Service klipper_LK5_Pro_ERCF.service

Notice that the printer config file is named LK5_Pro_ECRF.cfg instead of printer.cfg.

I'm not very familiar with Bash scripting but maybe it's time to learn unless you have a little time. I'll look at adding a -r parameter to specify a Repetier Stub. If a stub is specified, it will be used for the Installation Path. the Printer Configuration, and the Service. Also, Repetier does not use Moonraker but instead has it's own update process so the Moonraker sections can be bypassed.

I'm sure with your experience you could make these changes fairly quickly, but if you are unable to do so, I will muddle through and provide a copy for your review. I'll do some hard coding first to see if I run into any other issues and to see if the ERCF works.

jgbrown54, 2024-02-26:
I thought I had it. I used the -k and -c to specify the folders. I also changed the name of my config file to printer.cfg. I also changed the name of the service to look for to klipper_LK5_Pro_ERCF. But I still got the Klipper Service not found. It should be able to find it but it didn't. Then I ran across -s option for NOSERVICE. That appeared to allow everything to load. I rebooted the server but now I'm back to the Option 'step_pin' is not valid in section 'stepper_mmu_gear'.  Something didn't load correctly.

It's getting late tonight. I'll try some more tomorrow.

ningpj, 2024-02-26:
Try this.  quick hack to include -r arg to override printer.cfg and klipper.service.  I haven't tested it other than changing hard coded references for printer.cfg and klipper.service to variables and checking install.sh without args still runs through. You will need to test this on a repetier-server specific setup. 

Add -r LK5_Pro_ERCF to set KLIPPER_SERVICE to LK5_Pro_ERCF.service & PRINTER_CONFIG=LK5_Pro_ERCF.cfg.  

e.g. this "should" work - ./install.sh -k /opt/klipper/LK5_Pro_ERCF -c /var/lib/Repetier-Server/database/klipper -m /opt/klipper/LK5_Pro_ERCF/moonraker -r -r LK5_Pro_ERCF -i 

Make sure you have a backup of your config.

Usage: ./myinstall.sh [-k <klipper_home_dir>] [-c <klipper_config_dir>] [-m <moonraker_home_dir>] [-b <branch>] [-r <Repetier-Server stub>] [-i] [-d] [-z]

-i for interactive install
-d for uninstall
-z skip github check
-r specify Repetier-Server <stub> to override printer.cfg and klipper.service
(no flags for safe re-install / upgrade)
[repetier_install.sh.zip](https://github.com/moggieuk/Happy-Hare/files/14402005/repetier_install.sh.zip)

ningpj, 2024-02-26:
Thats usually an issue with symbolic links for mmu klipper modules not getting setup in <klipperhome>/klippy/extras dir.  Assuming install.sh is pointing at the correct directory, run install.sh again and restart klipper.

These are the symbolic links install.sh is trying to setup:

/home/pi/klipper/klippy/extras

lrwxrwxrwx 1 pi pi    46 Feb 26 21:37 mmu_config_setup.py -> /home/pi/Happy-Hare/extras/mmu_config_setup.py
lrwxrwxrwx 1 pi pi    41 Feb 26 21:37 mmu_encoder.py -> /home/pi/Happy-Hare/extras/mmu_encoder.py
lrwxrwxrwx 1 pi pi    44 Feb 26 21:37 mmu_led_effect.py -> /home/pi/Happy-Hare/extras/mmu_led_effect.py
lrwxrwxrwx 1 pi pi    38 Feb 26 21:37 mmu_leds.py -> /home/pi/Happy-Hare/extras/mmu_leds.py
lrwxrwxrwx 1 pi pi    33 Feb 26 21:37 mmu.py -> /home/pi/Happy-Hare/extras/mmu.py
lrwxrwxrwx 1 pi pi    41 Feb 26 21:37 mmu_sensors.py -> /home/pi/Happy-Hare/extras/mmu_sensors.py
lrwxrwxrwx 1 pi pi    39 Feb 26 21:37 mmu_servo.py -> /home/pi/Happy-Hare/extras/mmu_servo.py
lrwxrwxrwx 1 pi pi    42 Feb 26 21:37 mmu_toolhead.py -> /home/pi/Happy-Hare/extras/mmu_toolhead.py

jgbrown54, 2024-02-26:
I tried the repetier_install,sh you sent. It seems to install but I get the same issue Option 'step_pin' is not valid in section 'stepper_mmu_gear'. 

My Happy-Hare folder is in /home/pixie instead of /home/pi. Where are these links set?

jgbrown54, 2024-02-26:
Step by step. Inch by inch.

I set up a folder /home/pi and put Happy-Hare in it and ran the repetier_install.sh from there. Now I get a new error. Internal error during connect: module 'toolhead' has no attribute 'BUFFER_TIME_HIGH'

I just made the folder and didn't set a user for it. I'll try that.

ningpj, 2024-02-26:
The klipper team have been working on the python tool head object lately creating a dependency between klipper and Happy Hare releases. Update klipper to work around this. This will be less of an issue moving forward but in the meantime changes to Happy Hare have been required to maintain compatibility so both need to be updated. There may also be a few other hardwired assumptions in install.sh that need to be handled differently

ningpj, 2024-02-27:
Did you manage to get a functional install for repetier-server?  I have created a PR to merge changes for install.sh but cant verify if this is all that's required

jgbrown54, 2024-02-27:
Unfortunately I got sidetracked over the last few days with an ER visit and doctor appointments for a kidney stone.

You provided some symbolic links above that should be set up. They are set pointing to a /home/pi/Happy-Hare. My Happy-Hare folder is in /home/pixie/Happy-Hare. Does it need to be in /home/pi or doe it use the installation folder? How can I check these links?

ningpj, 2024-02-27:
OK, HH assumes its installed under the users home dir.

ningpj, 2024-02-27:
it "should" base it of the directory you run install.sh from.  rerun install.sh from the happy hare directory again

closed 2024-02-29 by moggieuk

jgbrown54, 2024-03-06:
I'm sorry my health interfered with testing this. I finally got a chance to try it. Installation does work after setting some permissions. I used the install.sh as follows:

./install.sh -k /opt/klipper/LK5_Pro_ERCF -c /var/lib/Repetier-Server/database/klipper -r LK5_Pro_ERCF -i  

The "stub" in Repetier is LK5_Pro_ERCF which is the name of the printer in Repetier-Server. Notice that I had to specify the stub as part of the -k option. Repetier installs a separate copy of Klipper for each printer. You should use the stub as specified in the -r parameter as part of the Klipper installation folder.

I will document what protections needed to be changed. For now I just set them to 777 but they should probably be set for just owner and group.

I do still have one issue which is probably another protection issue. After running the MMU_CALIBRATE_SELECTOR, it is unable to save the variables. Where ar they saved?

ningpj, 2024-03-06:
mmu_software.cfg defines the persistent save location as {klipper_config_home}/mmu/mmu_vars.cfg so may have a write permission issue or an incorrectly replaced reference.  Moggieuk has already merged the PR and closed this as complete. however you will still need the -k override as klipper is usually installed into the users home so need to be able to point it at a different target. again, if this is hardwired and the only option you have when installing repetier then we could set klipper home to always point to /opt.klipper/<stub> when -r is specified and -k is null.

jgbrown54, 2024-03-06:
Changing the permissions on the configuration folder eliminated the "Unable to save variables" issue. But it also brings up something new. Repetier-Server allows multiple installations of Klipper each using it's own .cfg file. With the mmu .cfg files in a folder off of the configuration folder, there is no way to specify separate mmu .cfg files for each printer. This is not currently an issue for me as I only have the one printer running Happy-Hare. But for complete compatibility, the mmu .cfg files will need to be separated into folders for each printer based on the "stub". I'm wondering if I change the mmu folder to mmu_LK5_Pro_ERCF and change the includes in the LK5_Pro_ERCF.cfg file to the new folder name, will the settings for the variables reference the new folder or are they hard coded to be at the config file location /mmu? I'll try that tomorrow and see what happens. We might need some additional work on the -r option of the install.sh to keep the mmu .cfg files separated for each printer stub.

Thanks for all your work and assistance so far. I appreciate anything that makes something easier and better. My favorite quote is, "There is always a better way to do something. Find it. -- Thomas Edison"

jgbrown54, 2024-03-06:
Okay. I'll drop it. The changes you have made do allow a single printer running Happy-Hare to be be added in Repetier. It's just that you are limiting it to only one printer with Happy-Hare instead of two or more which could be required for a printer farm. I have verified that I can make the changes necessary to allow multiple printers myself if I have the need which at the moment I do not.

PR #205 (2024-02-29): Add support for Repetier-Server install overrides
Add new ``-r`` install.sh arg to allow ``printer.cfg`` and ``klipper.service`` references to be overridden when installing HH onto Repetier-Server.  

Repetier-Server supports multiple klipper instances and generates unique names for ``klipper.service`` and ``printer.cfg`` using a stub/instance reference. This allows users to specify the "stub" used to set this correctly.

For example - use ``-r LK5_Pro_ERCF`` to set klipper.service to ``klipper_LK5_Pro_ERCF.service`` & printer.cfg to ``LK5_Pro_ERCF.cfg``

``./install.sh -k /opt/klipper/LK5_Pro [... cut]