moggieuk/Happy-Hare #209: Suggestion to change the order in _MMU_RESTORE_POSITION

Zephyrus67, 2024-03-02:
It seems pretty problematic to me that the restore order in the _MMU_RESTORE_POSITION macro is done with the X and Y first instead of restoring the Z position first.

When I do a print where I do some toolchange in the very top of the print I am breaking the print because the process I am using is with a filament cutter + purge/wipe bucket but NOT with a prime/purge tower.

So when it is done priming the new filament into the nozzle then it goes down to z=3 to wipe the nozzle on the brush. Then it moves towards the original x and y positions and actually goes right into the printed part.
If it restores the Z pos first things like this would not happen. So is there a very good reason for NOT restoring Z first?

moggieuk, 2024-03-03:
Well the reason is that the this is a reversal of a z_hop done at the start of the sequence.  It is last to prevent crashing (or grazing) of the part on restore.  Really the purge/wipe brush extension should restore position when done.  I recommend a SAVE_GCODE_STATE at the beginning of the macro and a RESTORE_GCODE_STATE MOVE=1 at the end.

That way each step is responsible for restoration. 

Yes, I could restore z-height first but then you would loose the safety of being above the print

Zephyrus67, 2024-03-04:
restoring z position first would in my logic actually KEEP safety of being above the print. But I probably just dont understand the way this is put together. the moves I am talking about is in the MMU_POST_LOAD macro in sequences.

Anyway. So you say that changing the:
RESTORE_GCODE_STATE NAME=MMU_POST_LOAD_state
to:
RESTORE_GCODE_STATE NAME=MMU_POST_LOAD_state MOVE=1

will fix this issue?

Zephyrus67, 2024-03-04:
Well it definitely doesnt work the way you say.

I now only have a call to my nozzle scrub macro and the change you just described and what it does now is move to the purge area as supposed....then it moves to the cutter as supposed and make the cut. Then it moves back to the purge area to unload old filament and loads new filament..all as expected.....but THEN it moves to the saved pos...and now after I added the MOVE=1 it moves back to the purge area...and THEN back to saved pos again. but this time in something that looks like a linear Z along the distance of the X,Y restore.

I have had a look in the nozzle scrub macro I am using and that ALSO have a SAVE_GCODE_STATE and a RESTORE_GCODE_STATE command in it.

I dont know if that can mess up things.Should I maybe make a "copy" of the nozzle scrub code that I will use along with the toolchange process rather than use the old one that has its own gcode state calls?

moggieuk, 2024-03-10:
I need a quite space to think through what might be happening here, but but default HH does not move the toolhead back to starting  X,Y position now. 

BTW the convention for motor sync and servo position is that every step in the process should ensure these are correct and not assume starting state.  The reason for this is to minimize thrashing of the servo where movement is not necessary.

jwhite, 2024-03-14:
@Zephyrus67 I had the same problem.  It was surprisingly hard to figure out, as most of the time I would return to the corner of the wipe tower and it would work, but I noticed I had plastic building up there and every so often either an object or the wipe tower would be in the return path and become disloged.

I was using the default nozzle scrub macro provided with the brush/purge bucket.

Here is my wipe routine:

<script src="https://gist.github.com/jwhite/5e56b33776e62b8ea000441d1decce68.js"></script>

you can see the changes I made here:

[image]

[image]

Be aware that you want to be in absolute G90 positioning when you get this variable and restore it.

jwhite, 2024-03-14:
> Btw: it seems like the software keeps the mmu_servo in move position while the MMU_POST_LOAD is processed. As I see it then it should change to the print state it is supposed to use before this...I can of course also just add a mmu_servo pos=up myself but would it not be more correct that the software actually goes back to the upp position right after it is done loading the filament?

I have this as well (tracked here : https://github.com/moggieuk/Happy-Hare/issues/159)

I fixed it with the mmu_servo pos=up added in my tool change.

moggieuk, 2026-10-21:
Closing this as v4 handles it. In Happy Hare v4, `_MMU_RESTORE_POSITION` first raises the toolhead back to the toolchange movement height (the z-hop plane set when it parked) before making any XY move, and only drops to the original print height at the end. So if your purge/wipe macro lowers the nozzle to the brush, the return travel still happens above the print rather than at brush height.

If you're adding a custom wipe, the cleanest place for it in v4 is `variable_user_post_load_extension` in `mmu_macro_vars.cfg`, which runs after loading but before the toolhead position is restored.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk