moggieuk/Happy-Hare #210: Endless Spool - reloaded

LostFool, 2024-03-03:
Without Klipper screen, how do I tell Happy Hare that I have successfully reloaded a gate?

So say big print is going, group 0 (tool 0 and 1) has filament.
It switches from T0 to T1 after run-out. I replace tool T0 while it is printing. 
Now I want HappyHare to know that T0 is reloaded and usable, but I don't see how.

moggieuk, 2024-03-04:
`MMU_GATE_MAP` can be used.  With the pre-gate sensor it will be automatic, but this would work:
```
MMU_GATE_MAP GATE=0 AVAILABLE=1
```
this will tell HH the gate is available.
But the TTG map would have have mapped T0 to gate1, so to switch that back:
```
MMU_TTG_MAP TOOL=0 GATE=0
```
It will/should fix itself on the next tool change.

LostFool, 2024-03-04:
Awesome! Thanks! I think I also messed up my gate assignment because now T0
only calls T1, which is labeled as T0+
The gate assignment in the parameters.cfg is a bit confusing. But I believe
I was just supposed to do grouping from the command line ey.

moggieuk, 2024-03-10:
The "T0+" means that a particular gate is mapped to more than one tool.  The `MMU_TTG_MAP` command will give you more details. BTW to force a change, assuming our group is setup correctly, you can:
```
MMU_GATE_MAP GATE=xx AVAILABLE=0
```
This will/should cause the fallover to another gate in the group.

Let me know if this doesn't work as expected - I'll leave this Issue open for a while.

moggieuk, 2024-04-08:
Going to close the issue as fixed.