moggieuk/Happy-Hare #211: mmu in error but print don't stop

salocain, 2024-03-03:
i have a fews error on my ercf , and i see few times , HH say " MMU issue detected whilst printer is paused......" but the print start or restart after a filament load ( a fake filament load because in error)

moggieuk, 2024-03-04:
I need to see the 'mmu.log'.  Happy Hare does not resume a print unless resume is called -- it's the only way, but let's look at the log first...

salocain, 2024-03-05:
here a link of a video with bugs.
[https://youtube.com/shorts/dgH0mp3gIq8?feature=share](url)

21:53:40 MMU Last Print Statistics:
         1 swaps completed
         1 minutes 1 seconds spent loading (average: 1 minutes 1 seconds)
         0 seconds spent unloading (average: 0 seconds)
         0 seconds spent paused (total pauses: 1)
         
         Gate Statistics:
         #0: Perfect, #1: Great, #2: Degraded, #3: Perfect, #4: Terrible, #5: Marginal, #6: Great, #7: Poor, #8: Good, #9: n/a, #10: n/a, #11: Perfect
21:53:41 - - TRACE: Enabled runout detection
21:53:44 > MMU_CHANGE_TOOL TOOL=4
21:53:44 - DEBUG: Asked to save toolhead position for change_tool but it is already saved for change_tool. Ignored
21:53:44 - - TRACE: Disabled runout detection
21:53:44 - DEBUG: Tool change initiated with standalone MMU tip forming
21:53:44 Tool change requested, from T3 to T4
21:53:44 - DEBUG: Unloading tool T3
21:53:44 - - TRACE: Running macro: _MMU_PRE_UNLOAD
21:53:48 Unloading filament...
21:53:48 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Idle'
21:53:48 [T3] < En <<<<<<<<< [Ex <<< Nz] LOADED 0.0mm (e:0.0mm)
21:53:48 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Forming Tip' OLD_ACTION='Unloading'
21:53:48 - - TRACE: _ensure_safe_extruder_temperature: current_temp=199.48, paused_extruder_temp=0.0, current_target_temp=200.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
21:53:49 Forming tip...
21:53:49 - - TRACE: Running macro: filament_cutting
21:53:52 - - TRACE: After tip formation, extruder moved (park_pos): 6.0mm, encoder measured 6.7mm
21:53:52 [T3] < En <<<<<<<<< [Ex <<. Nz] -6.0mm (e:5.8mm)
21:53:52 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Forming Tip'
21:53:52 - - TRACE: Running macro: _MMU_POST_FORM_TIP
21:53:52 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Exiting Ext' OLD_ACTION='Unloading'
21:53:52 - DEBUG: Extracting filament from extruder
21:53:52 - - TRACE: _ensure_safe_extruder_temperature: current_temp=199.7, paused_extruder_temp=0.0, current_target_temp=200.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
21:53:52 - DEBUG: Unloading last 594.0mm to exit the extruder
21:54:30 - - TRACE: Unloading extruder. Stepper: 'extruder' moved -594.0mm, encoder measured 561.7mm (delta 32.3mm). Pos: @-600.0, (567.5mm)
21:54:30 - DEBUG: Total measured movement: 561.7mm, total delta: 32.3mm
21:54:30 [T3] < En <<<<<<<<< [Ex ... Nz] -590.0mm (e:567.5mm)
21:54:30 - DEBUG: Filament should be out of extruder
21:54:30 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Exiting Ext'
21:54:30 - DEBUG: Unloading bowden tube
21:54:30 - DEBUG: Setting servo to down (filament drive) position at angle: 0
21:54:30 - DEBUG: Performing bowden pre-unload test
21:54:31 - - TRACE: Bowden pre-unload test. Stepper: 'gear' moved -15.0mm, encoder measured 15.4mm (delta -0.4mm). Pos: @-605.0, (583.9mm)
21:54:31 [T3] < En <<<<<.... [Ex ... Nz] -605.0mm (e:583.9mm)
21:54:35 - - TRACE: Course unloading move from bowden. Stepper: 'gear' moved -470.0mm, encoder measured 471.1mm (delta -1.1mm). Pos: @-1075.0, (1055.0mm)
21:54:35 [T3] < En <........ [Ex ... Nz] -1075.0mm (e:1055.0mm)
21:54:35 - DEBUG: Slow unload of the encoder
21:54:36 - - TRACE: Unloading step #1 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 14.5mm (delta 0.5mm). Pos: @-1090.0, (1069.4mm)
21:54:37 - - TRACE: Unloading step #2 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 15.4mm (delta -0.4mm). Pos: @-1105.0, (1084.9mm)
21:54:37 - - TRACE: Unloading step #3 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 14.5mm (delta 0.5mm). Pos: @-1120.0, (1099.3mm)
21:54:38 - - TRACE: Unloading step #4 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 15.4mm (delta -0.4mm). Pos: @-1135.0, (1114.7mm)
21:54:38 - - TRACE: Unloading step #5 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 15.4mm (delta -0.4mm). Pos: @-1150.0, (1130.1mm)
21:54:39 - - TRACE: Unloading step #6 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 14.5mm (delta 0.5mm). Pos: @-1165.0, (1144.6mm)
21:54:40 - - TRACE: Unloading step #7 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 13.5mm (delta 1.5mm). Pos: @-1180.0, (1158.1mm)
21:54:40 - - TRACE: Unloading step #8 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 1.0mm (delta 14.0mm). Pos: @-1195.0, (1159.1mm)
21:54:41 - - TRACE: Final parking. Stepper: 'gear' moved -8.0mm, encoder measured 0.0mm (delta 8.0mm). Pos: @-1187.0, (1159.1mm)
21:54:41 [T3] < En ......... [Ex ... Nz] UNLOADED -1179.1mm (e:1159.1mm)
21:54:41 - DEBUG: Setting servo to up (filament released) position at angle: 120
21:54:42 Unload of -1179.1mm filament successful (encoder measured 1159.1mm)
21:54:42 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Unloading'
21:54:42 - - TRACE: Running macro: _MMU_POST_UNLOAD
21:54:42 - DEBUG: Loading tool T4...
21:54:42 - - TRACE: Running macro: _MMU_PRE_LOAD
21:54:42 - DEBUG: Selecting tool T4 on gate #4...
21:54:42 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Selecting' OLD_ACTION='Idle'
21:54:42 - DEBUG: Setting servo to move (filament hold) position at angle: 120
21:54:42 - - TRACE: Positioning selector
21:54:42 - - TRACE: Setting gear motor rotation distance: 21.893519 (ratio: 0.946098)
[... cut ...]
21:56:20 - DEBUG: Job State: PAUSE_LOCKED -> CANCELLED (MMU State: Encoder: Disabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_hop @0.0mm, pause_resume: False, Idle timeout: 600.00s)
21:56:20 - - TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='cancelled' OLD_STATE='pause_locked'
21:56:21 - - TRACE: Processing idle_timeout 'ready' event
21:57:38 > MMU_HOME
21:57:38 - - TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Homing' OLD_ACTION='Idle'
21:57:38 - - TRACE: Processing idle_timeout 'printing' event
21:57:38 Homing MMU...

moggieuk, 2024-03-10:
Sorry for the delay and thanks for video and log..

So obviously this shouldn't happen, but the clue is here `21:54:48 MMU issue detected whilst printer is paused`

The mmu is already "paused" when the error occurs and so does do something similar to:
```
2:14 PM TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='pause_locked' OLD_STATE='initialized'
2:14 PM TRACE: Running macro: PAUSE
2:14 PM TRACE: MMU PAUSE wrapper called
2:14 PM DEBUG: Cannot save toolhead position or z-hop for pause because not homed
2:14 PM TRACE: Running macro: __PAUSE
```

Are you running via Octoprint by any chance?  If so there are a couple of steps that must be taken (calling _MMU_START and _MMU_END for example).

But assuming that the very first mmu "error" in the logs did cause the PAUSE macro, then we should look there as to why the print continues...

salocain, 2024-03-10:
no i'm on mainsail ^^
I see the problem become after a bad load between extruder and hotend (bowden style here).
i just update to 2.5 , i test if the same mistake became ( i modify many things in my config now) .
(i try to configure "touch" with tmc2209 and mmb .... no good results for now, because no sensibility....)

salocain, 2024-03-10:
another issue in same line with precedent.

`12:08:55 (\_/)
         ( *,*)
         (")_(") Happy Hare v2.5.0 Ready
12:08:55 Gates: | 0 | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10| 11|
         Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |T7 |T8 |T9 |T10|T11|
         Avail: | B | B | B | B | B | S | S | S | B | S | S | S |
         Selct: ------------| * |-------------------------------- T3
12:08:55 [T3] . En ......... [Ex ... Nz] UNKNOWN 0.0mm (e:0.0mm)
12:08:55   DEBUG: Job State: READY -> INITIALIZED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
12:08:55     TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='initialized' OLD_STATE='ready'
12:08:55     TRACE: Disabled runout detection
12:08:55   DEBUG: Setting servo to move (filament hold) position at angle: 120
12:08:55     TRACE: Processing idle_timeout 'printing' event
12:08:55     TRACE: (mmu_pre_gate_0 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_1 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_2 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_3 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_4 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_5 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_6 sensor detects filament)
12:08:55     TRACE: (mmu_pre_gate_7 sensor detects filament)
12:08:57     TRACE: Processing idle_timeout 'ready' event
12:08:58     TRACE: Processing idle_timeout 'printing' event
12:08:59     TRACE: Processing idle_timeout 'ready' event
12:09:04     TRACE: Processing idle_timeout 'printing' event
12:11:35 > _MMU_PRINT_START
12:11:35     TRACE: _on_print_start(->started)
12:11:35     TRACE: Enabled runout detection
12:11:35   DEBUG: Job State: INITIALIZED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
12:11:35     TRACE: _on_print_start(->printing)
12:11:35   DEBUG: Setting servo to up (filament released) position at angle: 120
12:11:35 Happy Hare initialized ready for print (no filament preloaded)
12:11:35   DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
12:11:35     TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='printing' OLD_STATE='started'
12:11:35     TRACE: Running macro: _MMU_CLEAR_POSITION
12:11:46 > MMU_LED RESET=1	QUIET=1
12:11:46     TRACE: Running macro: _MMU_SET_LED EXIT_EFFECT=default ENTRY_EFFECT=default STATUS_EFFECT=default
12:11:46 > MMU_SLICER_TOOL_MAP INITIAL_TOOL=0 PURGE_VOLUMES=
12:11:46 > MMU_SLICER_TOOL_MAP
12:11:46 --------- Slicer MMU Tool Summary ---------
         Single color print
         Initial Tool: T0
         -------------------------------------------
12:11:46 > MMU_CHANGE_TOOL STANDALONE=1 TOOL=0
12:11:46 Attempting to recover filament position...
12:11:46   DEBUG: Checking for filament in MMU...
12:11:46   DEBUG: Setting servo to down (filament drive) position at angle: 10
12:11:48     TRACE: After buzzing gear motor, encoder measured 0.00
12:11:48   DEBUG: Filament not detected in encoder after buzzing gear motor
12:11:48 [T3] > En ......... [Ex ... Nz] UNLOADED 0.0mm (e:0.0mm)
12:11:48   DEBUG: Saving toolhead gcode state and position (X:0.0 Y:-25.0 Z:0.0 E:0.0) for change_tool
12:11:48   DEBUG: Lifting toolhead 0.2mm
12:11:48     TRACE: Disabled runout detection
12:11:48   DEBUG: Tool change initiated with standalone MMU tip forming
12:11:48     TRACE: track times: {'total': 1710069108.643215}
12:11:48 Tool change requested: T0
12:11:48   DEBUG: Loading tool T0...
12:11:48     TRACE: track times: {'total': 1710069108.643215, 'pre_load': 1710069108.6464}
12:11:48     TRACE: Running macro: _MMU_PRE_LOAD
12:11:48     TRACE: statistics: {'total_swaps': 175, 'load': 9983.34, 'unload': 8884.0, 'total_pauses': 70, 'pause': 14669.59, 'pre_load': 0.46, 'swaps_since_pause': 3, 'post_load': 155.33, 'swaps_since_pause_record': 10, 'total': 2154.29, 'pre_unload': 16.68, 'post_unload': 0.08}
12:11:48   DEBUG: Selecting tool T0 on Gate 0...
12:11:48     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Selecting' OLD_ACTION='Idle'
12:11:48   DEBUG: Setting servo to move (filament hold) position at angle: 120
12:11:51     TRACE: Attempting selector 'touch' movement
12:11:53     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Selecting'
12:11:53 MMU issue detected. Print will be paused
         Reason: Selector path is probably externally blocked.
         Occured when changing tool: > T0
[... 263 lines cut ...]
12:21:08     TRACE: statistics: {'total_swaps': 176, 'load': 10109.21, 'unload': 8952.915935516357, 'total_pauses': 71, 'pause': 14669.59, 'pre_load': 0.5032942199707031, 'swaps_since_pause': 1, 'post_load': 181.58, 'swaps_since_pause_record': 10, 'total': 2314.05, 'pre_unload': 17.348656587600708, 'post_unload': 0.0838909912109375}
12:21:08     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Loading'
12:21:08 MMU issue detected whilst printer is paused
         Reason: Load sequence failed: Error loading filament at gate - not enough movement detected at encoder.
         Occured when changing tool: T0 > T6
[... cut ...]
12:22:54     TRACE: Running macro: __CLEAR_PAUSE
12:22:54     TRACE: _on_print_end(cancelled)
12:22:59     TRACE: Disabled runout detection
12:22:59   DEBUG: Job State: PAUSE_LOCKED -> CANCELLED (MMU State: Encoder: Disabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
12:22:59     TRACE: Running macro: _MMU_PRINT_STATE_CHANGED STATE='cancelled' OLD_STATE='pause_locked'
12:23:00     TRACE: Processing idle_timeout 'ready' event`

salocain, 2024-03-10:
[image]

moggieuk, 2024-03-10:
I'm not sure what exactly in the error in here you want me to look at but I did notice something strange:
```
12:11:53     TRACE: Running macro: __PAUSE
12:11:53 Error running PAUSE: Extrude below minimum temp
```

Is this my pause macro?    This failure might be the key to solving this problem .. if the pause macro fails it will never call BASE_PAUSE which actually pauses the print...

salocain, 2024-03-11:
i change the macro pause/resume/cancel by your , i cross fingers , i tell you if the problem come back

moggieuk, 2024-03-12:
Ok. Sounds good.  If you look at my reference macros, I make sure the PAUSE can't fail.   BASE_PAUSE must be called.  If you are going to try to extruder you need to put it in a conditional that checks the extruder temp to ensure it can extrude...

moggieuk, 2024-03-19:
I'm going to close out this issue, but feel free to re-open if you experience it again.  I'm confident that it was because of the pause within a long running macro...

closed 2024-03-19 by moggieuk