moggieuk/Happy-Hare #212: Over Extrusion during first filament load

HoweCarson, 2024-03-05:
I have ran into an issue with a filametrix setup on my ERCF. For some reason during the initial load for the MMU (Typically after a reboot)
The MMU will attempt to extrude clear to the nozzle instead of right at the filament cutter location. 

My fix at first was to set ooze prevention to 62mm, but this causes every other tool change to error out due to loading the final amount as 0mm (Error is thrown as encoder did not sense any movement)

I haven't been able to figure out why it's over extruding on the first load.

moggieuk, 2024-03-10:
If you are using a cutting this is indeed possible.  I was considering persisting the "amount of filament" left in the nozzle across reboots but I'm not sure how reliable that will be..

The issue is that HH keeps an in-memory value for the amount of filament left in the nozzle between tool changes. Obviously this is lost on reboot.

moggieuk, 2024-03-10:
I'm going to label this an enhancement because if `persistence_level` is set to 4, then this position should probably be persisted.

HoweCarson, 2024-03-10:
Sounds good, now knowing if `persistence_level` is set to 4 it should not be an issue. I will verify on my current build. I believe I it is currently set to 3.

moggieuk, 2024-03-10:
Ok. I've created a fix. It is in a branch called "issue212" so you can switch to it and test.  Let me know and I'll merge with main.
To switch to issue212 branch:
```
cd ~/Happy-Hare
./install.sh -b issue212
```
To switch back to main at any point, just repeat the above with "main" instead of "issue212"

moggieuk, 2024-03-12:
I've tested and confident it is addressed.  Works with any persistence level.  Pushed to main

HoweCarson, 2024-03-12:
I have tested as well and did not experience an issue either. Thank you.

closed 2024-11-29 by github-actions[bot]