moggieuk/Happy-Hare #215: Variable printer['mmu'].last_tool is not populated at the beginning of a new print

barsamgr, 2024-03-07:
I am using the Variables Branch and I have set it up to eject the filament at the end of the print.
What I have noticed at the beginning of a new print is that the tool used at the last print is not moved to the last_tool variable.  I am using this information so that if the new print is with a different color, I need to purge the appropriate amount of filament to clear the nozzle.
If I am not mistaken, this was working in the main branch, but I may be wrong.

I can see that in the mmu_vars.cfg file has saved the mmu_state_tool_selected from the previews print, but it is not used to populate the last_tool variable during the initialization.

This is what I use at the start g-code before my start_print:

MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK

Benoitone, 2024-03-08:
maybe if you can switch to https://github.com/Frix-x/klippain/tree/Happy_Hare_v2.5 branch of klippain you can help me to see if it's OK for you...
It's seems OK for me.
for info to change branch:
```
cd ~/klippain_config && git checkout Happy_Hare_v2.5 && git pull
```

moggieuk, 2024-03-10:
I'm pretty confident that `last_tool` would have always been reset on the start of the print (main branch also) but I'm not 100% sure.  I'll take a look.

But you might find all the new "purge volume" work added to the v2.5.0 useful in the meantime...

moggieuk, 2024-03-10:
I took a closer look and there are only two places I set "last_tool".  The first is during initialization (like MMU_INIT, reboot, etc), and the other is on MMU_CHANGE_TOOL.  It is set to current selected tool.

Is it possible that something is causing the selected gate to be set to unknown?  An `MMU_HOME` perhaps?

moggieuk, 2024-03-10:
I've made a partial fix (which restores previously functionality)...
If the MMU_CHECK_GATE call returns without pause, the current tool will be reselected.
If the checking fails (and mmu pauses) the current tool will be unknown.

The current tool is important because it is what last_tool will be set to when CHANGE_TOOL is called.

It's a little tricker to do better than this former functionality because after a pause, really anything is possible..

barsamgr, 2024-03-10:
I can see now that after the gates check the selector goes back to the last tool and it is kind of working, but...
maybe it is better to introduce a new variable in mmu_vars.cfg that will hold the value of the tool used in the last print and it will be filled each time a tool change is called while self._is_printing() is true.  The mmu_state_tool_selected variable can be changed in many situations and then the filament left in the nozzle from the last print is not known.  Of course there is a way to set it manually, but it is not convenient.

moggieuk, 2024-03-10:
This isn't a bad idea. Really the "last_tool" was initially designed to only to be valid in the toolchange for the purpose of having sufficient info to do intelligent tip forming and purging.  That was a long time ago...

I think this might be very similar to issue212.  I created a branch for that that persists the filament length left in the extruder across restarts.

moggieuk, 2024-03-11:
Ok. I've created a fix. It is in a branch called "issue212" (because it was so similar to this issue) so you can switch to it and test. Let me know and I'll merge with main.

To switch to issue212 branch:
```
cd ~/Happy-Hare
./install.sh -b issue212
```

To switch back to main at any point, just repeat the above with "main" instead of "issue212"

moggieuk, 2024-03-12:
Ok. I tested and was satisfied so it has been pushed to main

barsamgr, 2024-03-12:
Looks good now

closed 2024-11-29 by github-actions[bot]