moggieuk/Happy-Hare #218: Moonraker gcode preprocessing fails on upload and print

Neristance, 2024-03-07:
Hi,

First of all great work, my setup was flawless so far with happy-hare!

I today upgraded to v2.5.0 which adds the new macro for the start gcode (my version):

```
MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK

START_PRINT BED_TEMP=[bed_temperature_initial_layer]  EXTRUDER_TEMP=[nozzle_temperature_initial_layer] SKIRTLOOPS=[skirt_loops] USE_BED_MESH_CALIBRATE=0 USE_QUAD_GANTRY_LEVEL=0

MMU_START_LOAD_INITIAL_TOOL
```

I noticed that the mmu_server.py does run and process the file, its also twice checking the code its because moonraker acts also on file changes... anyway.

The usecase is following I use OrcaSlicer and have setup a multifilament print (two colors) then I hit `PRINT` -> `Upload and Print ` just to have it easy and let my Voron 2 make his job automatically.

What happens now from my observation is that the file gets uploaded its processed by moonraker and mmu_server.py the job is started, then the file is saved and moonraker runs again. 

During the execution of the GCode the Happy-Hare macro `MMU_START_SETUP` reports that the preprocessor is not correctly setup:

```
19:59:08 
// --------- Slicer MMU Tool Summary ---------
// Single color print
// Initial Tool: T1
// -------------------------------------------
19:59:08 
echo: Happy Hare gcode pre-processor is probably disabled or not setup correctly
```

But this seems to be a raise condition the print starts and the file contents are not yet updated with the recent version and by the time the macro is executed it still has the non replaced version like in the start gcode:

`MMU_START_SETUP INITIAL_TOOL=1 REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!
`

Checking the gcode file on the PI running Klipper I can see it is processed:

`MMU_START_SETUP INITIAL_TOOL=1 REFERENCED_TOOLS=1 TOOL_COLORS=0000FE,000000 TOOL_TEMPS= TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!`

I also noticed that it does not fill the temperature which is empty and ignores the materials as well as purge volumes fully. I was checking the metadata that is in the gcode file and compared it "fast" with the mmu_server.py code but the regex and strings used seem to be okay and I can find all of this in the gcode file from the orca slicer. 

Sadly I did not find a fast fix myself due to the time to open a PR but I hoped for your help.

If you need more info let me know!

In short how to reproduce:

1. Setup Multimaterial Print in OrcaSlicer
2. Hit Upload & Print
3. Check console output 
4. MMU_SLICER_TOOL_MAP DETAIL=1 reports single color print even there are T1 and T0 gcode in the file

Neristance, 2024-03-07:
Some note:

I did write a python unit test that runs `_write_mmu_metadata` with my gcode file and the end result is fine.

However, it looks like to fail as above when uploading via OrcaSlicer or by manually dropping the file for upload to Fluidd.

Result is this:

`MMU_START_SETUP INITIAL_TOOL=1 REFERENCED_TOOLS=1 TOOL_COLORS=0000FE,000000 TOOL_TEMPS= TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!`

moggieuk, 2024-03-10:
Upload and print functionality was not supported and documented (somewhere :-) )

BUT.  I just pushed a patch that changes the way the internal work so now the atomic "upload & print" is possible.  Force and update an check it out...  (tested well with Prusa Slicer, less well with Orca).

Side note. You have to "persuade" the slicer to include the purge volumes -- if wipe tower is enabled, then disabled, the purge volumes will be in the print.  The blobifier author @Dendrowen probably has more info on this.

Neristance, 2024-03-10:
@moggieuk great thanks for the info and the patch! I will check and provide feedback about OrcaSlicer then 👍 
I normally use a wipetower and checked for the slicer metadata which had all information attached and as mentioned even wrote a small unittest which looked okay, but then it did puzzle my mind :'D

iamnotahippy, 2024-03-10:
Running HH v2.5.0-37-g31639895, upload and print does not seem to work yet. The old method of upload only still works.

moggieuk, 2024-03-11:
Interesting.  Did you restart moonraker?

iamnotahippy, 2024-03-11:
I had restarted moonraker per the changelog. I've since rerun the HH install script and can confirm it's working properly with Prusa Slicer. Thanks!

moggieuk, 2024-03-12:
Great to hear.  I'm going to close this issue. Thanks.

closed 2024-03-12 by moggieuk