moggieuk/Happy-Hare #221: [FR] Motor current & accel tuning for _MMU_CUT_TIP_VAR

SHKinsem, 2024-03-10:
While printing with acceleration control enabled in slicer settings, different acceleration values are applied for various movements. In my case, I had set a relatively high travel acceleration. Unfortunately, during the tip-cutting process, this high acceleration caused a loss of a few motor steps—even though I had configured a relatively low velocity for the cutting process.

To address this issue, I experimented by adding a line to the tool change G-code. This line sets the acceleration before the tool change occurs. It turned out this modification significantly improved reliability—at least for my setup.

So, I’m requesting the implementation of these two functionalities.

moggieuk, 2024-03-10:
Understand your setup but I think this is a very unique case that you can fix is a sustainable way. Here's how:
```
[gcode_macro MY_MMU_CUT_TIP]
gcode:
     add your acceleration control here..
     _MMU_CUT_TIP {rawparams}
     reset accelaration control here...
```
Then in `mmu_parameters.cfg`:
```
form_tip_macro: MY_MMU_FORM_TIP
```

This structure will survive future updates..

SHKinsem, 2024-03-13:
Thanks for the reply.

> Then in `mmu_parameters.cfg`:
> 
> ```
> form_tip_macro: MY_MMU_CUT_TIP
> ```

It is pretty cool that the custom implementation could be done in this way because of the code structure you made.

After a few more rounds of print test, to exaggerate and quantify the effect I would say changing the accel raised the success rate from 50% to 75%, while directly changing the current can raise it from 50% to 100%. Me either think it is weird of changing the accel could also contributes to the success rate, probably it is because of the high decelerating back EMF (IDK lol 

>If this becomes a common request I will reconsider adding default functionality because the use of a toolhead driven cutter is still relatively new.

Yes I am also new to this so I am still testing it, not yet ready for a real multi-color printing job.

I am closing this for now since this issue could be addressed with only a small amount of macro customization.

closed 2024-03-13 by SHKinsem