moggieuk/Happy-Hare #225: Feature request: the ability to set your own base colors in the configuration.

deathfrozen, 2024-03-11:
[image]
I would like to be able to set my colors in the configs as in the screenshots above. Previously, I set these colors in a macro, but now this file is read-only. I am especially interested in the black color, because it looks purple for me (I use a very weak white and it looked very good).

moggieuk, 2024-03-11:
Obviously you can still copy the file (break the symlink) and then edit. The problem is that it them won't be able to upgrade.  I may add this but need to take a break from updates for a few days.  Can you tell me what RGB you used instead? The concept was to use a "black light" to signify black.

deathfrozen, 2024-03-13:
Yes, I also thought that it doesn't look complicated and just wanted to share an idea that doesn't really affect the main functionality. I use 0.01 0.01 0.01 for black. The color that is currently set by default looks like purple for me (I have a purple filament and they are very similar). But perhaps this is subjective and for different people it is the color that he will choose black (and maybe other colors) - different. It is also possible that this is influenced by the plastic used to print the diffusers (I have a white opaque plastic).
Thanks for the amazing software, I really appreciate your work.

moggieuk, 2024-03-18:
Thanks.  Ok. It's all live now of the push of v2.5.1. to main...

deathfrozen, 2024-03-18:
Thanks, it works perfectly!

closed 2024-11-29 by github-actions[bot]