moggieuk/Happy-Hare #229: Internal error during connect: 'MmuToolHead' object has no attribute 'min_cruise_ratio'

pvdl101, 2024-03-14:
After the most recent update I was immediately faced with this error:
"Internal error during connect: 'MmuToolHead' object has no attribute 'min_cruise_ratio'"
Reinstall of Happy-Hare does not resolve the issue
[klippy (1).log](https://github.com/moggieuk/Happy-Hare/files/14596527/klippy.1.log)
[mmu (1).log](https://github.com/moggieuk/Happy-Hare/files/14596528/mmu.1.log)

pvdl101, 2024-03-14:
Thanks for the fix, updating again resolved the issue.

Tribute06, 2024-03-14:
just updated and now getting this error Internal error during connect: 'MmuToolHead' object has no attribute 'requested_accel_to_decel'   tried updating second time and still getting same error.

jdlarry, 2024-03-15:
same issue now getting Internal error during connect: 'MmuToolHead' object has no attribute 'requested_accel_to_decel'

Sparkss, 2024-03-15:
I am running Klippain + Happy Hare and had the same 'requested_accel_to_decel' error. I found that I had mcu mmu defined in both the HH and KP files. I commented out the ones in the KP files (they were in my mcu.cfg) and updated the ones in HH (in mmu/mmu.cfg) to reflect the proper comm (canbus id in my instance) and no longer get the error.

Tribute06, 2024-03-16:
I will give that a try.      Thanks for the help

779124428, 2024-04-09:
same issue happend, "Internal error during connect: 'MmuToolHead' object has no attribute 'requested_accel_to_decel'" I checked the config files and installed HH again, but it does not work, how could I solve the problem?

jdlarry, 2024-04-09:
Look in all your config files and comment out any accel to decel lines. This fixed it for me.

closed 2024-11-29 by github-actions[bot]