moggieuk/Happy-Hare #233: getting Internal error during connect: 'MmuToolHead' object has no attribute 'requested_accel_to_decel'

jdlarry, 2024-03-16:
after last update get this eroor

JoaquinBerrios, 2024-03-17:
I believe this is due to a change in the latest version of Klipper to deprecate ACCEL_TO_DECEL in favor of MINIMUM_CRUISE_RATIO.  I didn't see ACCEL_TO_DECEL referenced in any HappyHare macros except for some in blobifier.cfg (which I'm not using) which would need to be updated.

The other instance in my configuration was in klicky-macros.cfg. The newest version of the macro on their github includes updates to address deprecation of ACCEL_TO_DECEL in Klipper. Grabbing the latest version of the macro solved the problem for me. 

You can also try adding MAX_ACCEL_TO_DECEL parameter in the [printer] section of your printer.cfg file as a workaround. It will give you an error that the parameter is deprecated, but allowed my printer to at least home/level, although it's probably best to make proper updates.

Sparkss, 2024-03-17:
I had mentioned this error and how I fixed it here: https://github.com/moggieuk/Happy-Hare/issues/229#issuecomment-1999965210  Not sure if that is the same root cause, but I hope that info helps.

moggieuk, 2024-03-18:
If a reference to and error in HH then you simply need to update klipper and HH to fix.

The Bloblifer macro has been updated (I'm not the author) and will be release n v2.5.1 (due today)

jdlarry, 2024-03-19:
i got the latest kilicky probe cfg and then everything worked. thanks.

closed 2024-04-08 by moggieuk