moggieuk/Happy-Hare #235: Use TEMPERATURE_WAIT for Extruder Temp

SkiBikePrint, 2024-03-17:
Tool changes could be shortened if the logic for checking/waiting for the target extruder temperature used Klipper TEMPERATURE_WAIT SENSOR=extruder MINIMUM={var1} MAXIMUM={var2} where var1/2 are user settable via parameters or macro_vars.

For me, sometimes there is no wait for the hotend temperature and other times it could be up to 20 seconds.  I likely am the worse case as the Mellow Fly-SB2040 CAN toolhead board seems to be a bit bouncy with its hotend temperature control but I suspect that this affecting mst users at varying degrees.  

I personally would be tolerant to +/- 3-5 degrees temp variation during a tool change.  Others might want it at +/-1 and could set this accordingly.

Thanks for you consideration of this Moggie!

Glenn

SkiBikePrint, 2024-03-18:
I looked at the code and you do use Temperature_wait so the request is to expose the min/max variables to be user adjustable with defaults as many won't need to change them, but some of us will.

Code block under discussion below from mmu.py:

            # Optionally wait until temperature is stable or at minimum safe temp so extruder can move
            if wait and new_target_temp >= klipper_minimum_temp and abs(new_target_temp - current_temp) > 2:
                with self._wrap_action(self.ACTION_HEATING):
                    self._log_info("Waiting for extruder to reach target (%s) temperature: %.1f%sC" % (source, new_target_temp, UI_DEGREE))
                    self.gcode.run_script_from_command("TEMPERATURE_WAIT SENSOR=extruder MINIMUM=%.1f MAXIMUM=%.1f" % (new_target_temp - 1, new_target_temp + 1))

moggieuk, 2024-03-18:
So you are basically asking to make the 2 degrees, a `mmu_parameters.cfg` configuration value?

Sounds reasonable..

moggieuk, 2024-03-18:
Will be in the v2.5.1 release:
```
default_extruder_temp: 200      # The baseline temperature for performing swaps and forming tips outside of a print
extruder_temp_variance: 2       # When waiting for extruder temperature this is the +/- permissible variance in degrees (>= 1)
```

SkiBikePrint, 2024-03-18:
Yes, and we will either need to have min/max be variables or more logically min=target-range/2 and max=target+range/2 where range is the user settable parameter.

I tried to test this by making changes to mmu.py under HH/extras (I think) but it didn't seem to make a difference.

I don't know what I'm doing with the Python stuff so maybe that is not the right way to do it??

Sent from my Verizon, Samsung Galaxy smartphone

moggieuk, 2024-03-18:
It's now released to main.  So it should work.  It is a symmetrical "variance" so the default of 2.  So +/- 2 degrees.
Let me know...

SkiBikePrint, 2024-03-18:
Thanks so much!  I'll give it a try.

Updates are a bit of a pain for me as I am controlling the ERCF directly from my main MCU pins (Octopus 1.1).

I can't figure out how to use mmu.cfg when no MMU MCU is defined. Is there a way to do that?  It would make updates much quicker.

Thanks again,

Glenn

Sent from my Verizon, Samsung Galaxy smartphone

SkiBikePrint, 2024-10-28:
Request is not relevant

closed 2024-10-28 by SkiBikePrint