moggieuk/Happy-Hare #237: SKEW Profile causes filament cutter to "Miss"

arnoldstrife, 2024-03-19:
Having a SKEW profile (https://www.klipper3d.org/Skew_Correction.html) can cause the toolhead to miss the filament cutter. I get

Reason: Unload sequence failed: Error running _MMU_CUT_TIP: Move out of range: -1.854 355.000 2.817 [5.000]

Because with the skew correction it's trying to move the toolhead to a negative x axis. I fixed it myself by adding

SET_SKEW CLEAR=1
to the beginning of the gcode macro and
SKEW_PROFILE LOAD=CaliFlower //Califlower is the name of my skew profile
at the end.
Probably should have something built in to fix it.

moggieuk, 2024-04-05:
Yes, this is true. The solution is to move the location of the cutter slightly away from the edge.  It is always risky to move/print right at the limit with skew correction on because even a 0.00001 mm movement causing an overhang in the print volume will cause klipper to stop.  This is how klipper is designed and the purpose of a the fixed build volume.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 the cutter moves stay inside the axis limits when skew correction is on, and I log a warning if a move had to be nudged, so clearing the skew profile around the cut shouldn't be needed.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk