moggieuk/Happy-Hare #239: Gate Map doesn't catch info from Spoolman

ac1dwalk3r, 2024-03-21:
I have Spoolman installed, followed the Instructions.
Spoolman shows on my Fluidd Dashboard, i can select Filaments.

But when i try to do a MMU_GATE_MAP GATE=0 SPOOLID=1
it just Changes SPOOLID, all the other values stay empty.
persistence_level: 3
enable_spoolman: 1

I didn't do the Calibration yet, does that have influence?
Or do i need costum fields in spoolman?

Either i'm blind somehow or something in the spoolman db has changed or there's an extra step thats not documented.

please help!
[mmu.log](https://github.com/moggieuk/Happy-Hare/files/14705341/mmu.log)

ac1dwalk3r, 2024-03-22:
Update: I've found the issue(s)
1: localhost in moonraker.cfg didn't work.
2.:
when i update the spool with
 MMU_GATE_MAP GATE=0 SPOOLID=1
it doesn't refresh automatically.
MMU_GATE
afterwards still shows just the spoolID
MMU_GATE REFRESH=1
refreshes, prints the filament list, still no change.
after repeating the last steps one or 2 times, MMU_GATE shows the correct filament.

later i figured out:
2 Times Repeating MMU_GATE REFRESH=1
means  one refresh in the console message. after the first, its in the MMU_VARS.cfg
at the second time it prints the correct values.

So may lets transform this into a feature request: Reinitialize the Data Before printing The Filament list?

Generally, there's definitly a better way to implement that from a user perspective. how about a dropdown list with available spoolman rolls for each Gate?

moggieuk, 2024-04-05:
Gald you were able to sort it out.  Remember that HH isn't an interactive UI, so when you run
```yml
MMU_GATE_MAP GATE=0 SPOOLID=1
```
it updated the spoolid in the gatemap immediately and issues an ASYNCHRONOUS request to moonraker to request that the reset of the gate map should be updated.  This request returns sometime later (or times out) and then the rest of the gate map is updated.  It is not possible (well, possible but not a good idea) to wait in a klipper gcode command for an  async communication request.

The consequence of this is that the gate map will be updated as soon as possible but it may not be (won't be) available at the time of running the MMU_GATE_MAP command.

closed 2026-10-21 by moggieuk