moggieuk/Happy-Hare #241: mmu_Sequence.cfg are Read only

afllak, 2024-03-22:
After updating to v2.5.1 I was trying to add a custom Gcode to the post-loading to have more filament purging but the file is in read-only and I can't save the changes. Is there a way to add the Gcode? Thank you.

121Jigawatt, 2024-04-02:
Not speaking for the creator or have worked on it, but that is probably by design. Most of the macros that get run during a toolchange have been moved to the mmu_macro_vars.cfg file. Around like 114 you can give it a macro to run. You may need to make a new macro as it doesn't seem like it supports multi-line g-code, just a single macro

moggieuk, 2024-04-05:
This is by design and follow the pattern established by other macro plugins like Mainsail, Klippain, etc.

You can extend the default macros through the user hooks, e.g.
```yml
variable_user_pre_unload_extension  : ''        ; Executed after default logic
variable_user_post_unload_extension : ''        ; Executed after default logic
variable_user_pre_load_extension    : ''        ; Executed after default logic
variable_user_post_load_extension   : ''        ; Executed after default logic but before restoring toolhead position
```
just insert a call to your macro (defined in another file).

Secondly, if these hooks are not enough, then you can completely define your own macro (under your own name and in separate file) and then simply tell HH to execute these instead:
```yml
pause_macro: PAUSE
action_changed_macro: _MMU_ACTION_CHANGED           # Called when action (printer.mmu.action) changes
print_state_changed_macro: _MMU_PRINT_STATE_CHANGED # Called when print state (printer.mmu.print_state) changes
pre_unload_macro: _MMU_PRE_UNLOAD                   # Called before starting the unload
post_form_tip_macro: _MMU_POST_FORM_TIP             # Called immediately after tip forming
post_unload_macro: _MMU_POST_UNLOAD                 # Called after unload completes
pre_load_macro: _MMU_PRE_LOAD                       # Called before starting the load
post_load_macro: _MMU_POST_LOAD                     # Called after the load is complete
unload_sequence_macro: _MMU_UNLOAD_SEQUENCE         # VERY ADVANCED: Optionally called based on 'gcode_unload_sequence'
load_sequence_macro: _MMU_LOAD_SEQUENCE             # VERY ADVANCED: Optionally called based on 'gcode_load_sequence'
```

afllak, 2024-04-08:
Thank's

closed 2024-04-08 by afllak