moggieuk/Happy-Hare #242: gcode_unload_sequence: 1 will be ingorized

Turboralle83, 2024-03-23:
The user-unload macro has been ignored since the update to 2.5.1.
ERCF 1.1
2 gates

Parameters.cfg:
gcode_unload_sequence: 1	# VERY ADVANCED: Gcode unloading sequence, 1=enabled, 0=internal logic (default)

unload_sequence_macro: UNLOAD_SEQUENCE	    # VERY ADVANCED: Optionally called based on 'gcode_unload_sequence'   ###############################################################
###############################################################

[gcode_macro UNLOAD_SEQUENCE]
description: Called when MMU is asked to unload filament
gcode:
    {% set filament_pos = params.FILAMENT_POS|float %}
    {% set length = params.LENGTH|float %}
    {% set extruder_only = params.EXTRUDER_ONLY|int %}
    {% set park_pos = params.PARK_POS|float %}

    {% set AKTIVES_GATE = params.GATE|default(printer['mmu']['gate'])|int %}           #############################                                                          

    {% if extruder_only %}
        {% if filament_pos >= 7 %}                      # FILAMENT_POS_PAST_EXTRUDER
            _MMU_STEP_UNLOAD_TOOLHEAD EXTRUDER_ONLY=1 PARK_POS={park_pos}
        {% else %}
            {action_raise_error("Can't unload extruder - already unloaded!")}
        {% endif %}

    {% elif filament_pos == 0 %}
        {action_raise_error("Can't unload - already unloaded!")}

    {% else %}
        {% if filament_pos >= 7 %}                      # FILAMENT_POS_PAST_EXTRUDER
            # Exit extruder, fast unload of bowden, then slow unload encoder
            _MMU_STEP_UNLOAD_TOOLHEAD PARK_POS={park_pos}
        {% endif %}

        {% if filament_pos >= 4 %}                      # FILAMENT_POS_END_BOWDEN
            # Fast unload of bowden, then slow unload encoder
  
                 MANUAL_STEPPER STEPPER=AUFROLLER ENABLE=1               #############################

            {% if AKTIVES_GATE == 0 %} 
                SET_SERVO SERVO=aufroller ANGLE=160                                  ###############################
             {% endif %}

            {% if AKTIVES_GATE == 1 %}                                                            ###############################
                SET_SERVO SERVO=aufroller ANGLE=20
             {% endif %}

                    MANUAL_STEPPER STEPPER=AUFROLLER MOVE=5 SPEED=2 SET_POSITION=0   #############################                         
                   _MMU_STEP_UNLOAD_BOWDEN FULL=1
                    SET_SERVO SERVO=aufroller ANGLE=90    #############################
                   MANUAL_STEPPER STEPPER=AUFROLLER MOVE=-0.5 SPEED=2 SET_POSITION=0    ##########################                       
                   _MMU_STEP_UNLOAD_GATE
                   MANUAL_STEPPER STEPPER=AUFROLLER ENABLE=0                     #############################                                      

      {% elif filament_pos >= 2 %}                    # FILAMENT_POS_START_BOWDEN
            # Have to do slow unload because we don't know exactly where in the bowden we are
            _MMU_STEP_UNLOAD_GATE FULL=1
        {% endif %}

    {% endif %}

#####################################################################
it worked like in the video before the update

https://github.com/moggieuk/Happy-Hare/assets/136199923/ee755df6-ea75-4f41-94d3-19b98204ba2c

moggieuk, 2024-04-05:
Let me check. Nothing was changed there, but maybe a bug crept in...

moggieuk, 2024-04-05:
One thing to check is that the filament position integers are still correct:
```yml
    FILAMENT_POS_UNKNOWN = -1
    FILAMENT_POS_UNLOADED = 0
    FILAMENT_POS_HOMED_GATE = 1
    FILAMENT_POS_START_BOWDEN = 2
    FILAMENT_POS_IN_BOWDEN = 3
    FILAMENT_POS_END_BOWDEN = 4
    FILAMENT_POS_HOMED_ENTRY = 5
    FILAMENT_POS_HOMED_EXTRUDER = 6
    FILAMENT_POS_EXTRUDER_ENTRY = 7
    FILAMENT_POS_HOMED_TS = 8
    FILAMENT_POS_IN_EXTRUDER = 9 # AKA FILAMENT_POS_PAST_TS
    FILAMENT_POS_LOADED = 10     # AKA FILAMENT_POS_HOMED_NOZZLE
    ```

moggieuk, 2024-04-05:
I turned on the advanced load/unload macro sequence and testing my supplied examples (which duplicate the built in HH logic) and they worked as expected.   I haven't reviewed your macro but I think it is because the integer states changed slightly.  I'm confident that if you review and compare against the new examples in `mmu_sequence.cfg` it will probably be a simple fix...
LMK

moggieuk, 2024-04-05:
You can debug by looking at `mmu.log` and searching for the "_MMU_STEP_xxx" calls:
```
23:53:22 > _MMU_STEP_UNLOAD_BOWDEN FULL=1
23:53:22   DEBUG: Unloading bowden tube
23:53:22   DEBUG: Setting servo to down (filament drive) position at angle: 45
23:53:22   DEBUG: Performing bowden pre-unload test
23:53:25     TRACE: Bowden pre-unload test. Stepper: 'gear' moved -15.0mm, encoder measured 14.9mm (delta 0.1mm). Pos: @-87.0, (91.6mm)
23:53:25     TRACE: (toolhead sensor does not detect filament)
23:53:25 [T0] < En <<<<<.... [Ex .. (t) .. Nz] -87.0mm (e:91.6mm)
23:53:31     TRACE: Course unloading move from bowden. Stepper: 'gear' moved -643.0mm, encoder measured 689.0mm (delta -46.0mm). Pos: @-730.0, (780.5mm)
23:53:31     TRACE: (toolhead sensor does not detect filament)
23:53:31 [T0] < En <........ [Ex .. (t) .. Nz] -730.0mm (e:780.5mm)
23:53:31 > _MMU_STEP_UNLOAD_GATE
```

Turboralle83, 2024-04-05:
My macro is from the mmu_sequence.cfg, only the marked parts are inserted

19:02:43 [T0] < En <<<<<<<<< [Ex <<. Nz] -2.0mm
19:02:43     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Forming Tip'
19:02:43     TRACE: Running macro: _MMU_POST_FORM_TIP
19:02:43   DEBUG: Calling external user defined unloading sequence macro
19:02:43     TRACE: Running macro: _MMU_UNLOAD_SEQUENCE FILAMENT_POS=9 LENGTH=905.0 EXTRUDER_ONLY=0 PARK_POS=-2.0
19:02:43 > _MMU_STEP_UNLOAD_TOOLHEAD PARK_POS=-2.0
19:02:43     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Exiting Ext' OLD_ACTION='Unloading'
19:02:43   DEBUG: Extracting filament from extruder
19:02:43     TRACE: _ensure_safe_extruder_temperature: current_temp=229.84, paused_extruder_temp=None, current_target_temp=230.0, klipper_minimum_temp=0.0, default_extruder_temp=250.0, source=auto
19:02:43   DEBUG: Unloading last 57.0mm to exit the extruder
19:02:46     TRACE: Unloading extruder. Stepper: 'extruder' moved -57.0mm, encoder measured 0.0mm (delta 57.0mm). Pos: @-55.0, (0.0mm)
19:02:46 [T0] < En <<<<<<<<< [Ex ... Nz] -50.0mm
19:02:46   DEBUG: Filament should be out of extruder
19:02:46     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Exiting Ext'
19:02:46 > _MMU_STEP_UNLOAD_BOWDEN FULL=1

moggieuk, 2024-04-05:
The other thought is that your macro is the same name as the default one I supply.  This will cause confusion in klipper as to which one actually runs.  Best to call it something else and adjust:
```yml
unload_sequence_macro: _MMU_UNLOAD_SEQUENCE         # VERY ADVANCED: Optionally called based on 'gcode_unload_sequence'
load_sequence_macro: _MMU_LOAD_SEQUENCE 
```
to point to your version

moggieuk, 2024-04-05:
Ok. It's my mistake.  It assumes the original names..  I will patch tomorrow and push an update... sorry

Turboralle83, 2024-04-05:
That would be great!
THANKS!!!

moggieuk, 2024-04-05:
It was a simple fix so I just pushed an update.  Please let me know if this works as expected, it should...
You may need to force update_manager to see the update or you can run:
```
cd ~/Happy-Hare
./install.sh
```
Which will force the update.

Turboralle83, 2024-04-12:
it works!!!!!

Thanks!!!!!!!!!!!!

closed 2024-04-12 by Turboralle83