moggieuk/Happy-Hare #245: Need some way to satisfy conditions for extensions before MMU command can be run

Saikedo, 2024-03-24:
I frequently encounter an issue with the Blobifier tool. It relies on the conditions specified by variable_user_post_load_extension and variable_user_pre_unload_extension. The problem arises with Blobifier's requirement for QGL (Quality of Service Guarantee Level). If QGL is not previously applied, Blobifier terminates the operation. Unfortunately, this issue usually becomes evident too late during the tool change process. The filament is already exchanged by the hardware handler (HH), and only at the end, when variable_user_pre_unload_extension is executed, does it abort the operation due to the absence of QGL.

Not sure if possible but it would be beneficial to implement a mechanism that can verify whether all necessary conditions for the extensions are satisfied before initiating the filament change operation.

moggieuk, 2024-04-05:
I'm not the author of blobifer, but yes this could be done... the pre-unload  hook would be used to validate conditions, pausing if necessary telling the user to perform the necessary steps, then resuming, then when the post_load hook runs it should have all pre-conditions. I'll ping the author to see what they say..

Dendrowen, 2024-04-05:
Saikedo and I already discussed this in Discord and was fixed with the latest PR.

moggieuk, 2024-04-05:
Fantastic.  Closing as fixed. ;-)

closed 2024-04-05 by moggieuk