moggieuk/Happy-Hare #246: Endless spool don’t work properly

33martin, 2024-03-25:
Hy
I have a ERCF V2 software version v2.5.1-16-g774bc871  and when I simulate a runout with MMU_TEST_RUNOUT it works without any problems. But if I have a real runout then it switches correctly to the next filament and the printer stops at pause state. The interface shows the printer as paused, but clicking resume doesn't work as I get an error saying that I cannot resume as the printer is not paused. I'm uploading the mmu log if anyone is interested. At 11:23:20 I triggered > CANCEL_PRINT
The ERCF works flawless on filament swaps.

best regards Martin
[mmu_1.log](https://github.com/moggieuk/Happy-Hare/files/14742708/mmu_1.log)

Alohacajun, 2024-03-27:
I encountered a similar issue on the previous version, v2.5.1-15-ge6cdbddf. EndlessSpool triggered and successfully loaded the next spool but failed to resume the print. Tried clicking to resume (Mainsail v2.10 ; not using HH optional client_macros.cfg) and was told the same thing as OP. 
I don't remember the specific of how I did it, but I managed to get it to resume by manually calling one of the PAUSE macros (PAUSE_PRINT, PAUSE, etc.) then manually calling a RESUME macro (RESUME_PRINT, RESUME, etc). I don't have the log sadly, but I hope this helps provide some insight.

moggieuk, 2024-04-05:
Back from vacation. I'll take a look at this.   Can you confirm what sensor is triggering the runout.  Also I need a 'mmu.log' ...

33martin, 2024-04-08:
Hy

the pre gate switch,

mmu log is in the first post

thx

igiannakas, 2024-04-17:
Same just happened to me today. Runout triggered by pre gate sensor. Secondary gate selected and loaded. Nozzle moved to print but it didn’t resume. Calling the resume function says it was already printing when it clearly was not. The workaround was to press pause and resume manually.

[image]

closed 2024-04-22 by moggieuk