moggieuk/Happy-Hare #248: Feature Request: T0(x) Macro retrieve filament colours

mochoandre, 2024-03-26:
Hello,
Is it possible for tools macros in mmu_macro_vars.cfgretrieve filament colours so mainsail and fluid can automatically retrieve those colors?
like 

```
[gcode_macro T0]
gcode: MMU_CHANGE_TOOL TOOL=0
variable_color: "000000"
[gcode_macro T1]
gcode: MMU_CHANGE_TOOL TOOL=1
variable_color: "ffffff"
[gcode_macro T2]
gcode: MMU_CHANGE_TOOL TOOL=2
variable_color: "f20303"
[gcode_macro T3]
gcode: MMU_CHANGE_TOOL TOOL=3
variable_color: "2d8f02"
...

```

I can do it but during update Happy Hare rewrites file.
Any other way to make mainsail and fluid present the colors?

moggieuk, 2024-04-05:
I had no idea that Mainsail/Fluidd tried to obtain filament color this way...!?!

Let me look at what these UI's do to see how to address this -- honestly it is a strange way to look for color because it assumes that "Txx" is a macro.  What if it was a built in command.  Anyway, let me have a look..

moggieuk, 2024-04-05:
I've been playing a bit with this:
```yml
--------- Slicer MMU Tool Summary ---------
3 color print
T7 (Gate 7, ABS, ff5500, 245°C)
T0 (Gate 0, ABS, ff0000, 240°C)
T1 (Gate 1, ABS, 00ff00, 240°C)
Initial Tool: T0
-------------------------------------------
```
[image]

But one problem is that you have to refresh Mainsail (at least) because it doesn't monitor these variables.

moggieuk, 2024-04-05:
I also need to add a way to set slicer color for each tool without making the tool part of the print.

mochoandre, 2024-04-05:
> Let me look at what these UI's do to see how to address this -- honestly it is a strange way to look for color because it assumes that "Txx" is a macro. What if it was a built in command? Also, it should be static in this way, it needs to be dynamic based on either the gate_map or slicer_tool_map. Anyway, let me have a look..

I Wish I could help you because you have done a really magnificent work With Happy Hare and Klipercreen but I just found out this searching trough google. I Have no coding Skills unfortunatly. 

Thank you very much for all, You Rock!

moggieuk, 2024-04-07:
Ok. It's implemented.  Just posted in the forums.  Copied here:
Still v2.5.1 but in fixing a couple bugs reported under Issues I added a new "feature" to automatically set the tool colors in Mainsail / Fluidd.  Now all slicer tool info is stored in HH although only those used in print with be reported with `MMU_SLICER_TOOL_MAP` unless you specify the `DETAIL=1` option:
```yml
> MMU_SLICER_TOOL_MAP DETAIL=1
--------- Slicer MMU Tool Summary ---------
2 color print
T0 (Gate 0, PLA, red, 200°C)
T1 (Gate 1, ABS+, green, 201°C)
T2 (Gate 2, ABS, grey, 202°C) Not used
T3 (Gate 3, ASA, blue, 203°C) Not used
T4 (Gate 4, ABS, orange, 240°C) Not used
T5 (Gate 5, ABS, lt_green, 240°C) Not used
T6 (Gate 6, ABS, black, 240°C) Not used
T7 (Gate 7, ABS, yellow, 240°C) Not used
T8 (Gate 8, PETG, grey, 208°C) Not used
Initial Tool: T0
-------------------------------------------
```
Note that owing to the way Mainsail works it will not dynamically adjust the extruder colors... I going to see if I can twist an arm, but until then you can just refresh your browser and the colors will appear as soon as the print starts...

moggieuk, 2024-04-07:
[image]

ningpj, 2024-04-07:
nice....i had been manually overwriting these post updates.  works fine with fluidd env and can see all gate colour references are hex now so happy days :-)

moggieuk, 2024-04-08:
Does Fluidd dynamically update the colors when they change or do you need to refresh your browser?

zhangxiaokang7788, 2024-04-09:
Good idea.If slicer program can surport for ttg ，i think it will be better.

closed 2024-11-28 by github-actions[bot]