moggieuk/Happy-Hare #249: Resuming print after pause and filament change sometimes ends up resuming print with incorrect z offset.

Saikedo, 2024-03-30:
Hi, I am often leveraging the functionality where I pause the print at a certain height, change the filament and resume again. I notice that when happy hare is involved, sometimes resuming the print starts printing in the air, 1 mm above where it should have been. I was able to consistently replicate this issue by doing this.

1. In Cura, add pause at height (Calls M0 that is basically PAUSE)
2. set the temp back to what I was (Pause disables extruder heater)
3. Perform a toolchange (I think both just pressing on Tn in mainsail and  MMU_EJECT -> MMU_SELECT GATE=n -> MMU_LOAD) did the same thing
4. Manually purge a little bit of filament
5. Resume print

In a lot of cases, this ends up printing above the print surface by 1 mm for me. Tried the same with all the MMU includes commented out and the issue was not observed. I use custom PAUSE and RESUME macros, not sure if that affects anything but here they are just in case if it`s relevant.

Please let me know if I can help and test things to narrow down this issue.

Thanks

```
[gcode_macro RESUME]
rename_existing: BASE_RESUME
variable_zhop: 0
variable_etemp: 0
gcode:
    # Parameters
    {% set e = params.E|default(2.5)|int %}                   

    {% if printer['pause_resume'].is_paused|int == 1 %}
        SET_FILAMENT_SENSOR SENSOR=toolhead_sensor ENABLE=1                          ; enable filament sensor
        #INITIAL_RGB                                                                    ; reset LCD color
        SET_IDLE_TIMEOUT TIMEOUT={printer.configfile.settings.idle_timeout.timeout}  ; set timeout back to configured value
        {% if etemp > 0 %}
            M109 S{etemp|int}                                                        ; wait for hotend to heat back up
        {% endif %}
        RESTORE_GCODE_STATE NAME=PAUSEPARK MOVE=1 MOVE_SPEED=100                     ; go back to parked position in case toolhead was moved during pause (otherwise the return zhop can error)
        G91                                                                          ; relative positioning
        M83                                                                          ; relative extruder positioning
        {% if printer[printer.toolhead.extruder].temperature >= printer.configfile.settings.extruder.min_extrude_temp %}
            G1 Z{zhop * -1} E{e} F900                                                ; prime nozzle by E, lower Z back down
        {% else %}
            G1 Z{zhop * -1} F900                                                     ; lower Z back down without priming (just in case we are testing the macro with cold hotend)
        {% endif %}
        RESTORE_GCODE_STATE NAME=PAUSE MOVE=1 MOVE_SPEED=60                          ; restore position
        BASE_RESUME                                                                  ; resume print
    {% endif %}

[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
gcode:
    # Parameters
    {% set z = params.Z|default(10)|int %}         

    {% if printer['pause_resume'].is_paused|int == 0 %}
        SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=zhop VALUE={z}                              ; set z hop variable for reference in resume macro
        SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=etemp VALUE={printer['extruder'].target}    ; set hotend temp variable for reference in resume macro

        SET_FILAMENT_SENSOR SENSOR=toolhead_sensor ENABLE=0                                  ; disable filament sensor
        SAVE_GCODE_STATE NAME=PAUSE                                                          ; save current print position for resume
        BASE_PAUSE                                                                           ; pause print
        {% if (printer.gcode_move.position.z + z) < printer.toolhead.axis_maximum.z %}       ; check that zhop doesn't exceed z max
            G91                                                                              ; relative positioning
            G1 Z{z} F900                                                                     ; raise Z up by z hop amount
        {% else %}
            { action_respond_info("Pause zhop exceeds maximum Z height.") }                  ; if z max is exceeded, show message and set zhop value for resume to 0
            SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=zhop VALUE=0
        {% endif %}
        G90                                                                                  ; absolute positioning
        G1 X{printer.toolhead.axis_maximum.x/2} Y{printer.toolhead.axis_minimum.y+5} F6000   ; park toolhead at front center
        SAVE_GCODE_STATE NAME=PAUSEPARK; save parked position in case toolhead is moved during the pause (otherwise the return zhop can error)
        M104 S0                                                                              ; turn off hotend
        SET_IDLE_TIMEOUT TIMEOUT=43200                                                       ; set timeout to 12 hours
    {% endif %}

[gcode_macro M0]
gcode:
  PAUSE
```

moggieuk, 2024-04-05:
Can you tell me the version of HH you are running and whether you are running the default sequence macros?  (I assume because you have your own PAUSE/RESUME that you are not including mmu/optional/client_macros.cfg)

moggieuk, 2024-04-05:
Also, have a read of this: https://github.com/moggieuk/Happy-Hare/blob/main/doc/toolchange_movement.md#---z-hop-moves

It explains the way HH works with z-hop which might help debug the source of the incompatibility

Saikedo, 2024-04-08:
@moggieuk I am using my own macros but I just realized that I was also including `mmu/optional/client_macros.cfg`. I think the client_macros were being overwritten by my macros though since I was importing my macros after the client ones.

For version, I am using v2.5.1-16. I noticed that there is a new version but I did not had the chance to try with the new version yet.

Will also take a look at the zHop to see if I can debug this any further.

closed 2024-11-28 by github-actions[bot]