moggieuk/Happy-Hare #251: TTG Map Status "AVAILABLE = -1 " not to be placed above command

TTom1088, 2024-04-01:
Every now and then I had the error that a pre_gate_sensor was triggered briefly by movement of the filament. Therefore, Happy Hare didn't load the gate and went on pause. Now I have written a query macro that checks the limit switches before each load and then sets the gates back to AVAIABLE.
Is it perhaps possible to adapt the program so that the command
 MMU_TTG_MAP GATE=0 AVAILABLE=**-1** QUIET=0 is possible to set the gate with ? to mark?

here is the macro

[gcode_macro Gate_Abfrage_ES]
gcode:
 {% set runout = printer['mmu'].runout %}
 
 {% set G0 = printer['mmu'].gate_status [0] %}
 {% set G1 = printer['mmu'].gate_status [1] %}
 {% set G2 = printer['mmu'].gate_status [2] %}
 {% set G3 = printer['mmu'].gate_status [3] %}
 {% set G4 = printer['mmu'].gate_status [4] %}
 {% set G5 = printer['mmu'].gate_status [5] %}
 {% set G6 = printer['mmu'].gate_status [6] %}
 {% set G7 = printer['mmu'].gate_status [7] %}
 {% set G8 = printer['mmu'].gate_status [8] %}

 {% set ES0 = printer['filament_switch_sensor mmu_pre_gate_0'].filament_detected %}
 {% set ES1 = printer['filament_switch_sensor mmu_pre_gate_1'].filament_detected %}
 {% set ES2 = printer['filament_switch_sensor mmu_pre_gate_2'].filament_detected %}
 {% set ES3 = printer['filament_switch_sensor mmu_pre_gate_3'].filament_detected %}
 {% set ES4 = printer['filament_switch_sensor mmu_pre_gate_4'].filament_detected %}
 {% set ES5 = printer['filament_switch_sensor mmu_pre_gate_5'].filament_detected %}
 {% set ES6 = printer['filament_switch_sensor mmu_pre_gate_6'].filament_detected %}
 {% set ES7 = printer['filament_switch_sensor mmu_pre_gate_7'].filament_detected %}
 {% set ES8 = printer['filament_switch_sensor mmu_pre_gate_8'].filament_detected %}

 {% if runout != True %}

   {% if G0 == 0 and ES0 == True %}
      MMU_TTG_MAP GATE=0 AVAILABLE=1 QUIET=0
      M118 ES0 Pruefen 
   {% endif %}

   {% if G1 == 0 and ES1 == True %}
      MMU_TTG_MAP GATE=1 AVAILABLE=1 QUIET=0
      M118 ES1 Pruefen 
   {% endif %}

   {% if G2 == 0 and ES2 == True %}
      MMU_TTG_MAP GATE=2 AVAILABLE=1 QUIET=0
      M118 ES2 Pruefen
   {% endif %}

   {% if G3 == 0 and ES3 == True %}
      MMU_TTG_MAP GATE=3 AVAILABLE=1 QUIET=0
      M118 ES3 Pruefen 
   {% endif %}

   {% if G4 == 0 and ES4 == True %}
      MMU_TTG_MAP GATE=4 AVAILABLE=1 QUIET=0
      M118 ES4 Pruefen
   {% endif %}

   {% if G5 == 0 and ES5 == True %}
      MMU_TTG_MAP GATE=5 AVAILABLE=1 QUIET=0
      M118 ES5 Pruefen 
   {% endif %}

   {% if G6 == 0 and ES6 == True %}
      MMU_TTG_MAP GATE=6 AVAILABLE=1 QUIET=0
      M118 ES6 Pruefen
   {% endif %}
  
   {% if G7 == 0 and ES7 == True %}
      MMU_TTG_MAP GATE=7 AVAILABLE=1 QUIET=0
      M118 ES7 Pruefen
   {% endif %}

   {% if G8 == 0 and ES8 == True %}
      MMU_TTG_MAP GATE=8 AVAILABLE=1 QUIET=0
      M118 ES8 Pruefen
   {% endif %}
 {% endif %}

moggieuk, 2024-04-05:
It's a shame you have to do this.  I know there is a rework of the pre-gate sensor to make it less prone to flat spots causing a false trigger.  JFYI the reason that it doesn't reset after a false trigger is because of a built in delay (I'm wondering if that delay makes sense).

Anyway, it should be possible to call:
`MMU_TTG_MAP GATE=x AVAILABLE=-1`
to set back to an unknown state.  At least it is in the v2.5.1 release. 

Can you confirm you still have a problem..

TTom1088, 2024-04-05:
I had installed version 2.5.0-63 when the error occurred. I'll update, deactivate my macro and see if it works.
Thanks for the tip.

I'll let you know then

TTom1088, 2024-04-06:
I have now updated to happy hare v2.5.1-33-gf4673ecb. But writing the variable to AVAIABLE=-1 doesn't work yet. Later I'll test an unwanted short runout with the pre_gate_sensor

[image]

TTom1088, 2024-04-06:
After the update I have massive problems with the runout macro.
1. my runout RESTORE_GCODE_STATE is now subsequently overwritten and the correct Z height is no longer achieved.

2. The speed after the runout is too slow (because the command RESTORE_GCODE_STATE NAME=role change MOVE=1 MOVE_SPEED=300 is subsequently overwritten)

3. The Runout Variable doesn't seem to be reset properly. After a "Runout Resume" the printer executed the runout macro the next time it was unloaded.

Therefore I will now have to adapt the mmu.sequence macros for my config or use the old ones.

I would be happy to provide you with my cfg for the runout macro, maybe you can find an easy way to get around such err
[config-20240406-130953.zip](https://github.com/moggieuk/Happy-Hare/files/14893330/config-20240406-130953.zip)
ors.

moggieuk, 2024-04-07:
Ah, `MMU_TTG_MAP` does not support `AVAILABLE=-1`, but the raw gate map manipulation command does:
```yml
MMU_GATE_MAP GATE=0 AVAILABLE=-1
```

moggieuk, 2024-04-07:
As for z-height, it might just need to turn off the appropriate options.  Have a read of this to see if it answers your questions:
https://github.com/moggieuk/Happy-Hare/blob/main/doc/toolchange_movement.md#---z-hop-moves

TTom1088, 2024-04-07:
Now it's working again, but I noticed 1 small software bug with the pregate sensors. I'll try to isolate the error and find out the reason for it. I will contact you.

I thank you in advance for the help.

Unfortunately I cannot use the MMU_GATE_MAP GATE=0 AVAILABLE=-1 command for my macro because it then sets the color and material to "unknown". But MMU_TTG_MAP works quite well there.

Greetings Thomas

closed 2024-04-07 by TTom1088