moggieuk/Happy-Hare #252: Deprecation of max_accel_to_decel fails _MMU_PRE_LOAD

Neristance, 2024-04-03:
Hi,

Something I observed while updating to the recent version of Klipper: v0.12.0-138-g5e280680 + Happy Hare: v2.5.1-16-g774bc871

The config setting max_accel_to_decel is deprecated, so I recently did remove it from my printer config file. Until some versions it seems that Klipper had defaults, now with the recent version it looks like the defaults are gone which makes the _MMU_PRE_LOAD macro fail, I was checking the code seems its reading the value and trying to execute `SET_VELOCITY_LIMITS ACCEL_TO_DECEL=` with no value which results in a parsing error. 

[image]

Workaround:
Adding the config value back is a workaround until klipper fully removes the value from the config.

moggieuk, 2024-04-05:
I'm not understanding where this is coming from.  Yes, klipper did remove this setting and I thought I had logic to make Happy Hare tolerant to older klipper versions.  Any way, the latest update removes all the legacy support (and as a consequence, klipper must me up-to-date).  That means there is nothing that HH is doing WRT SET_VELOCITY_LIMIT, so behavior will be 100% the same as raw klipper.  Let me know if you still have problems with this.

Neristance, 2024-04-08:
I will check during the next days. I mostly put it as a reference since I hit this point and before others face the same they have at least a workaround.

moggieuk, 2024-04-25:
ANy updates on this?

Neristance, 2024-04-25:
Hi, I pulled latest changes and installed it, but it reports me the same issue still if I do omit `max_accel_to_decel`

Neristance, 2024-04-25:
Okay I found the bad guy, the one from Happy Hare is gone, sorry about that but seems like my Klicky Probe has also an issue ;) 

Can be closed!

closed 2024-04-25 by Neristance