moggieuk/Happy-Hare #254: All gcode files are modified after a print is finished

eMapleLuke, 2024-04-05:
I use the ERCF V2 with orca slicer. When I after a print is finished, all gcode files sliced with more than one color are modified. All have the same date and time, and the tracker of printed parts dont work anymore, because the files are recognized as new files.

iamnotahippy, 2024-04-05:
I have the same issue. It's breaking my gcode archiving script as well.

iamnotahippy, 2024-04-05:
> why do you believe happy hare is updating the g-code?

In moonraker.log it shows mmu_server.py attempting to preprocess every gcode file on the system.

ningpj, 2024-04-05:
Ok, the preprocessor & parsing and expanding placeholders/tokens like !referenced_tools!
This is triggered on gcode upload and was trying to work out why file timestamps on the gcode would be updated post print.  Do you use place holders like !referenced_tools! as you can disable to preprocessor in moonraker.conf

```[mmu_server]
enable_file_preprocessor: False
enable_toolchange_next_pos: False```

eMapleLuke, 2024-04-05:
> Ok, the preprocessor & parsing and expanding placeholders/tokens like !referenced_tools! This is triggered on gcode upload and was trying to work out why file timestamps on the gcode would be updated post print. Do you use place holders like !referenced_tools! as you can disable to preprocessor in moonraker.conf
> 
> ```
> enable_file_preprocessor: False
> enable_toolchange_next_pos: False
> ```

When I change these to false, what does it change? Do I have any disadvantages with it?

ningpj, 2024-04-05:
Your log doesnt seem to have the message "Processing Uploaded File:...." preceding the shell_command.py call out...
What version of moonraker are you running?  I'm current and running v0.8.0-325-g99b97afc and only see moonraker triggering the preprocess script each time i upload a new gcode prior to printing it.

eMapleLuke, 2024-04-05:
> It turns off the preprocessor so HH wont be able to validate required gates have filament ready to go. more detail here. https://github.com/moggieuk/Happy-Hare?tab=readme-ov-file#11-useful-pre-print-functionality https://github.com/moggieuk/Happy-Hare/blob/main/doc/gcode_preprocessing.md
> 
> So the process flow was you uploaded and printed a gcode file and at the conclusion of the print all mmu gcode files in the directory were updated again? Are you using fluidd/mainsail or octoprint to upload and print your files?

I use mainsail. Yes, after a print is finished, all the files I uploaded with more than one filament is updated, mot the old ones.

ningpj, 2024-04-05:
can you please upload you full moonraker log, or at the very least log entries from around the time you uploaded the gcode file.  Can you also please confirm your HH and moonraker versions

eMapleLuke, 2024-04-05:
> can you please upload you full moonraker log, or at the very least log entries from around the time you uploaded the gcode file. Can you also please confirm your HH and moonraker versions

[moonraker.log](https://github.com/moggieuk/Happy-Hare/files/14883450/moonraker.log)

ningpj, 2024-04-07:
Something odd is going on here as gcode files I havent touched for 7+ days are getting reprocessed.  
Stealthbourner gcodes should be dated march some time when I reprinted them but have clearly been reprocessed again today when my printer started and before I uploaded Plate-0.4.... to print.

``-rw-r--r--  1 pi pi 12258964 Apr  7 11:31 '[a]_stealthburner_main_body-0.4mm-34.8474g-2h19m.gcode'
-rw-r--r--  1 pi pi  2553866 Apr  6 20:15  MM_Test_print_9_colors-0.4mm-39.0408g-3h28m.gcode
-rw-r--r--  1 pi pi  2524523 Apr  6 20:15  MM_Test_print_9_colors-0.4mm-52.3465g-3h21m.gcode
-rw-r--r--  1 pi pi  4334138 Apr  6 20:15 'my2020 Bracket-0.4mm-46.3627g-1h48m.gcode'
-rw-r--r--  1 pi pi  1927348 Apr  6 20:15 'my2020 Bracket - colour logo-0.4mm-10.7182g-36m.gcode'
-rw-r--r--  1 pi pi  4335870 Apr  6 20:15 'my2020 Bracket - colour logo-0.4mm-45.1772g-1h46m.gcode'
-rw-r--r--  1 pi pi  8632515 Apr  7 12:39  Plate-0.4mm-60.2445g-2h40m.gcode
-rw-r--r--  1 pi pi  9338527 Apr  7 11:31  SB_CW2_Plate-0.4mm-27.9305g-1h46m.gcode
-rw-r--r--  1 pi pi 10897355 Apr  6 20:15  stealthburner_printhead_rapido_front-0.4mm-28.2149g-1h46m.gcode``

Moonraker log post startup clearly shows them being processed again....unsure if the INFO message re `currently supports cancellation, processing aborted` is relevant or not.  So net-net, these old files shouldnt be triggering 
 reprocessing. 

``Klipper Version: v0.12.0-131-gd9043345-dirty
2024-04-07 11:31:52,683 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Inserting next position to tool changes took 58566.49s

2024-04-07 11:31:52,737 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/mmu_server.py -m -n -p /home/pi/printer_data/gcodes -f "[a]_stealthburner_main_body-0.4mm-34.8474g-2h19m.gcode" --check-objects) successfully finished
2024-04-07 11:31:52,946 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2024-04-07 11:31:52,948 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is enabled

2024-04-07 11:31:53,121 [shell_command.py:pipe_data_received()] - INFO:metadata:File 'SB_CW2_Plate-0.4mm-27.9305g-1h46m.gcode' currently supports cancellation, processing aborted

2024-04-07 11:31:53,173 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/pi/printer_data/gcodes/SB_CW2_Plate-0.4mm-27.9305g-1h46m.gcode

2024-04-07 11:31:54,654 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 1.49s. Detected gcode by slicer: superslicer

2024-04-07 11:31:55,644 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Inserting next position to tool changes took 0.99s

2024-04-07 11:31:55,690 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/mmu_server.py -m -n -p /home/pi/printer_data/gcodes -f "SB_CW2_Plate-0.4mm-27.9305g-1h46m.gcode" --check-objects) successfully finished``

moggieuk, 2024-04-07:
Interesting.  This is definitely moonraker's decision to reprocess .. I just latch onto that (and thus you see some more log message).  Any change to a file (even a unix 'touch') will cause a file to be re-processed but I wonder what Moonraker is doing..?  If you have any clues let me know, but the extra processing that HH does isn't going to add much work - the bulk of the pre-processing time is the object processing..

ningpj, 2024-04-07:
yes, need to work out why its only triggering on some files as other old gcode files in the directory are ignored and will see what I can find out and will see if I can have a dig around in the moonraker db

eMapleLuke, 2024-04-08:
I have investigated the problem a little bit more, and found out, that on startup of the raspberry pi, the files were modified. And always the ones, were more than one filament were in the slicer configured.

iamnotahippy, 2024-04-08:
> I have investigated the problem a little bit more, and found out, that on startup of the raspberry pi, the files were modified. And always the ones, were more than one filament were in the slicer configured.

To add to this, restarting Moonraker under some unknown specific circumstances triggers the issue

ningpj, 2024-04-09:
Could be a moonraker bug?  If you disable (exclude) object processing and both mmu preprocessing options and restart moonraker, it appears to update its metadata in its db and leaves existing gcode files alone even when these options are subsequently reenabled.  When reenabled, any gcode file uploaded afterward this  will be preprocessed again whenever moonraker is restarted.  

```
[file_manager]
enable_object_processing: False
...
[mmu_server]
enable_file_preprocessor: False
enable_toolchange_next_pos: False
```

gneu42, 2024-04-17:
In my case, since a few days/week all file dates are changed to the current date when I start the printer.
I will disable file processor to see.

closed 2024-04-18 by moggieuk