moggieuk/Happy-Hare #263: Can not display thumbnail afer add MMU_START_SETUP MMU_START_CHECK to start g_Code

zhangxiaokang7788, 2024-04-09:
after I add MMU_START_SETUP、 MMU_START_CHECK to start g_Code like this:

MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes! ; Initialize MMU and save info from gcode file     
MMU_START_CHECK ; Check MMU is setup for the slicer defined print 
M115 U3.9.3 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode
G28 W ; home all without mesh bed level
G1 X0.1 Y10 Z40 F5000.0 ;Move to start position
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
MMU_START_LOAD_INITIAL_TOOL ; Load slicer defined initial tool into MMU 
G92 E0.0
G1 Z0.28 F3000.0 ;Move to start position
G1 Y180.0  F1500.0 E15 ;Draw the first line
G1 X0.4  F5000.0 ;Move to side a little
G1 Y20  F1500.0 E30 ;Draw the second line
G92 E0.0
G1 Z2.0 F3000 ;Move Z Axis up

My thumbnail can not display.I use Orcaslicer and fluid.Refresh meta does not work:
[image]

moggieuk, 2024-04-09:
Does this occur with every gcode file you upload?   If you have a small gcode file that has the error, can you attach to this issue.  Inspecting it might highlight the cause...

zhangxiaokang7788, 2024-04-10:
Every time when I upload the same test g-code. It is two cube model.
[image]

I have updated the comment , thank you for your answer.

moggieuk, 2024-04-10:
Very strange.  The file you sent has not been pre-processed by Happy Hare - I assume it is the original file.

I loaded into my system with HH pre-processor enabled and it correctly parsed the placeholders and `Tx tool changes`: E.g.
```yml
diff CUBE_PETG_1h1m.gcode /tmp/CUBE_PETG_1h1m.gcode
8d7
< ;
116c115
< MMU_START_SETUP INITIAL_TOOL=0 REFERENCED_TOOLS=0,1 TOOL_COLORS=80FFFF,000000,FFFFFF,FF0000,FF8040 TOOL_TEMPS=215,215,215,215,215 TOOL_MATERIALS=PETG,PETG,PETG,PETG,PETG PURGE_VOLUMES=0,118.75,225,250,150,750,0,750,406.25,150,118.75,75,0,256.25,106.25,556.25,87.5,581.25,0,118.75,543.75,56.25,550,331.25,0 ; Initialize MMU and save info from gcode file     
---
> MMU_START_SETUP INITIAL_TOOL=0 REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes! ; Initialize MMU and save info from gcode file     
135c134
< MMU_CHANGE_TOOL TOOL=0 NEXT_POS="103.736,113.736" ; T0
---
> T0
355c354
< MMU_CHANGE_TOOL TOOL=1 NEXT_POS="162.908,175.000" ; T1
---
> T1
869c868
< MMU_CHANGE_TOOL TOOL=0 NEXT_POS="147.092,168.140" ; T0
---
```

I can see the thumbnail in Mainsail, shown below...
[image]

moggieuk, 2024-04-10:
So I don't understand why the thumbnail is not displaying.. on my system it displays both before the HH pre-processing and afterwards.

Also the `diff` above shows that the only parts of the gcode file changes are as expected -- the thumbnail image is exactly the same.

ningpj, 2024-04-10:
Could be a browser caching issue....try clearing your cache and refreshing as your gcode file displays on my system with Fluidd

zhangxiaokang7788, 2024-04-10:
Yes, I compared the two  G-code files and the  thumbnail part is exactly the same. So I do not know why. 

The printing state can not display too which is showing in the second picture.

zhangxiaokang7788, 2024-04-10:
> Could be a browser caching issue....try clearing your cache and refreshing your mainsail window as your gcode thumbnail displays fine on my system with Fluidd

I clearing the cache and refreshing my fluid window ,it can not disply too.

zhangxiaokang7788, 2024-04-10:
It goes right when I enable pre-processed by Happy Hare. Is that necessary.

enable_file_preprocessor: True

ningpj, 2024-04-10:
if you are using placeholders like REFERENCED_TOOLS=!referenced_tools! then yes.  However I'n not sure why not having Happy Hare ``enable_file_preprocessor: False`` would cause mainsail/fluidd not to be able to render the thumb nail though.

I just loaded your gcode with out preprocessing the placeholders and it was displayed fine even with double byte characters on Pi's setup for english users (e.g. LANG=en_GB.UTF-8)

zhangxiaokang7788, 2024-04-10:
> if you are using placeholders like REFERENCED_TOOLS=!referenced_tools! then yes. However I'n not sure why having Happy Hare `enable_file_preprocessor: False` would cause mainsail/fluidd not to be able to render the thumb nail though.
> 
> I just loaded your gcode with out preprocessing the placeholders and it was displayed fine even with double byte characters on Pi's setup for english users (e.g. LANG=en_GB.UTF-8)

Thank you very much.

moggieuk, 2024-04-10:
I'll check with HH pre-processing turned off but I can't imagine this would make a difference...

zhangxiaokang7788, 2024-04-12:
The problem troubled again,even if I enabled file preprocessor.There is an error in moonraker log file .It take too long for preprocess.Is my machine underperforming:

`2024-04-12 19:24:00,621 [app.py:prepare()] - Upload Request Received from 192.168.1.17
Content-Type: multipart/form-data; boundary=------------------------695bee0fac0e0676
2024-04-12 19:24:10,399 [app.py:post()] - Processing Uploaded File: New_Toothless_Tail_2_PETG_6h13m.gcode
2024-04-12 19:24:11,759 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2024-04-12 19:24:11,775 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2024-04-12 19:24:11,982 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/klipper/printer_data/gcodes/New_Toothless_Tail_2_PETG_6h13m.gcode

2024-04-12 19:24:18,379 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 6.40s. Detected gcode by slicer: orcaslicer

2024-04-12 19:24:21,309 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/mmu_server.py -m -n -p /home/klipper/printer_data/gcodes -f "New_Toothless_Tail_2_PETG_6h13m.gcode") cancelled
2024-04-12 19:24:21,310 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2512, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2554, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.ServerError: Extract Metadata returned with error
2024-04-12 19:24:21,796 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2024-04-12 19:24:21,813 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2024-04-12 19:24:22,020 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/klipper/printer_data/gcodes/New_Toothless_Tail_2_PETG_6h13m.gcode

2024-04-12 19:24:28,395 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 6.37s. Detected gcode by slicer: orcaslicer

2024-04-12 19:24:31,332 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/mmu_server.py -m -n -p /home/klipper/printer_data/gcodes -f "New_Toothless_Tail_2_PETG_6h13m.gcode") cancelled
2024-04-12 19:24:31,332 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2512, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2554, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.ServerError: Extract Metadata returned with error
2024-04-12 19:24:31,819 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Running MMU enhanced version of metadata

2024-04-12 19:24:31,835 [shell_command.py:pipe_data_received()] - INFO:metadata:Object Processing is disabled

2024-04-12 19:24:32,039 [shell_command.py:pipe_data_received()] - INFO:metadata:mmu_server: Pre-processing file: /home/klipper/printer_data/gcodes/New_Toothless_Tail_2_PETG_6h13m.gcode

2024-04-12 19:24:38,458 [shell_command.py:pipe_data_received()] - INFO:metadata:Reading placeholders took 6.42s. Detected gcode by slicer: orcaslicer

2024-04-12 19:24:41,355 [shell_command.py:_check_proc_success()] - Command (/home/klipper/moonraker-env/bin/python /home/klipper/moonraker/moonraker/components/mmu_server.py -m -n -p /home/klipper/printer_data/gcodes -f "New_Toothless_Tail_2_PETG_6h13m.gcode") cancelled
2024-04-12 19:24:41,355 [file_manager.py:_process_metadata_update()] - Error running extract_metadata.py
Traceback (most recent call last):
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2512, in _process_metadata_update
    await self._run_extract_metadata(fname, ufp_path)
  File "/home/klipper/moonraker/moonraker/components/file_manager/file_manager.py", line 2554, in _run_extract_metadata
    raise self.server.error("Extract Metadata returned with error")
moonraker.utils.ServerError: Extract Metadata returned with error
2024-04-12 19:24:41,356 [file_manager.py:_process_metadata_update()] - Unable to extract medatadata from file: New_Toothless_Tail_2_PETG_6h13m.gcode
2024-04-12 19:24:41,368 [app.py:log_request()] - 201 POST /api/files/local (192.168.1.17) [_TRUSTED_USER_] 40749.10ms
2024-04-12 19:24:51,737 [proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 10.43 seconds, total blocked count: 1
2024-04-12 19:24:51,939 [proc_stats.py:log_last_stats()] - 
System Time: 1712921078.395222, Usage: 2.45%, Memory: 53900 kB
System Time: 1712921079.396378, Usage: 2.95%, Memory: 53904 kB
System Time: 1712921080.397809, Usage: 3.36%, Memory: 53908 kB
System Time: 1712921081.443564, Usage: 9.49%, Memory: 53908 kB
System Time: 1712921091.805073, Usage: 87.87%, Memory: 54748 kB
[... cut ...]
2024-04-12 19:42:11,198 [websockets.py:open()] - Websocket Opened: ID: 548177135120, Proxied: False, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36 Edg/123.0.0.0, Host Name: 192.168.1.165
2024-04-12 19:42:11,222 [websockets.py:_handle_identify()] - Websocket 548177135120 Client Identified - Name: fluidd, Version: 1.29.0-db5d50a, Type: web
2024-04-12 19:42:11,223 [file_manager.py:get_file_list()] - Updating File List <config>...
`

moggieuk, 2024-04-22:
That's interesting.  The pre-processing that HH adds is much smaller (overhead) that the native "object" pre-processing that moonraker can add, but an error could cause the issue you describe. Do you have a fast SD-card? One thing that does occur during pre-processing is a lot of sd-card read/writes...

zhangxiaokang7788, 2024-04-23:
Yes,It works very well without SD-card.I will test for more time.My SD-card is very old.

deathfrozen, 2024-04-30:
I have this problem with very large files (30 mb or more). Such files are often obtained precisely because of multimaterial printing and do not allow not only to view the thumbnail, but also other valuable information is lost - for example, the layer number, and sometimes the preliminary check of the presence of the thread before printing does not work. Is there no way to increase the file processing time?

deathfrozen, 2024-05-05:
I tried to transfer my system to a usb ssd drive and it didn't help, although the testing speed increased 20 times. I tried to disable enable_file_preprocessor and got a more distinct error - 
```
File "/home/orangepi/moonraker/moonraker/components/mmu_server.py ", line 264, in main
if (insert_nextpos and lean(tools_used) > 0) or has_placeholder:
TypeError: object of type 'NoneType' has no len()
```
 I tried to delete the link to mmu_server.py in the moonraker catalog and metadata are scanned and everything works - both the thumbnail and other parameters. 

I am not well versed in python, but in the code earlier I see in the else condition the assignment of tools_used None, but there is no check before using lan(tools_used). Is it possible that adding a check to None will avoid this error?

if insert_placeholders:
    start = time.time()
    has_placeholder, tools_used, colors, temps, materials, purge_volumes, slicer = parse_gcode_file(file_path)
    metadata.logger.info("Reading placeholders took %.2fs. Detected gcode by slicer: %s" % (time.time() - start, slicer))
else:
    **tools_used = colors = temps = materials = purge_volumes = slicer = None**
    has_placeholder = False
if (insert_nextpos and **len(tools_used)** > 0) or has_placeholder:

zhangxiaokang7788, 2024-05-06:
I got this problem with a large file,too.

salocain, 2024-05-12:
same problem

and many "time too close " or "crash"

deathfrozen, 2024-05-15:
I've done some research and even some debugging. As a result, it turned out that adding

```
[file_manager]
enable_object_processing: True
```

to the moonraker.conf file solves the problem with processing large files. This is due to the fact that in moonraker, with this option set, the timeout is 30 times longer.
[image]
I didn't figure -  how to set a timeout from a file mmu_server.py or out other options in the configuration files, and it's likely that it's impossible with the current moonraker code.

The error I wrote about earlier - https://github.com/moggieuk/Happy-Hare/issues/263#issuecomment-2094985206 it only occurs when the option
is set `
enable_file_preprocessor: False`. I've made a simple fix and I think I'll try to create a pr.

closed 2024-12-11 by github-actions[bot]