moggieuk/Happy-Hare #265: Feature Request: Endless Spool mode rethought

ac1dwalk3r, 2024-04-10:
Hi again! Just tested the Endless Spool mode and i see space for improvement. Not sure if this is possible:

At the moment, when the spool runs out, it unloads all the rest in the bowden tube (at least it tries, in my setup, when it detects the runout, its already past the ercf gear and outputs an error....i assume i'd need pregate sensors)

Since i also have a bambu AMS, i'm used to a different behaviour: when the spool runs out, it just switches to the next spool and pushes it into the tube without unloading. 

This would have advantages in happyhare too: 

-Encoder detects runout
-Printer resumes printing
-Extruder sync is deactivated(not sure if this works while printing)
-MMU switches to next gate
-Loads filament and uses the diag pin of the MMU extruder to detect when the new filament pushes against the rest of the last spool
-Reactivates extruder sync

Alternatively, doing this while the print is paused still would be an advantage

ac1dwalk3r, 2024-04-10:
I just thought of single colour....i'm printing on a vcore 500, enclosed drybox for the spools, they tend to run out on large prints, one could easily make use of nearly empty spools with that feature. Looking forward to that!

zhangxiaokang7788, 2024-04-11:
Can the  `extruder_homing_endstop` help。

1、filament runout
2、continue print until filament come to `extruder_homing_endstop` 
3、pause and load filament to toolhead sensor。（there is a little filament extrude from nozzle）
4、continue print

Belstrekkie, 2024-06-09:
This behaviour of unloading the bowden tube after filament run out in endless spool mode caused me issues with the unloaded filament getting trapped in other 'gates' of the ERCT buffer as it no longer is tied to its own buffer, resulting in either blockage of the buffers of the other gates or a break of the filament inside the selector, preventing it from moving to the next gate from the endless spool group. 

A simple solution might be to automatically decrease unload speed if filament run out is detected. Or just use the remaining filament and wait with filament change until the toolhead sensor is reached (by purging if this is before the end of the print/color change).

k1-801, 2024-07-29:
> you have to hope the entire fragment is used before the print ends otherwise you will have a bit of filament left in the extruder/bowden or if you unload at the end of a print it will fail...

It requires a way to remember which EndlessSpool's state is currently in use, some of which will indicate that the filament currently used by the extruder does NOT connect to the MMU gears, thus, can not be pulled out.
In this situation, if a toolchange is necessary, the remainder of filament has to be ejected through the nozzle first, or extracted by the user manually. Both options should be available by either giving a choice at the moment the runout is detected, or through configuration.

Ideally, we would want the slicer to mark how much filament will be used between toolchanges. In this case, we can compare how much filament was already used, and how much more would be needed, so we can decide what to do with the remainder currently in the bowden. Unfortunately, this is not currently possible (or is it?), but we can acquire this info by pre-processing gcode. Here's how I see the logic working:
* If the remaining length to print before the next toolchange is bigger than the length of the bowden, and endless spool is configured (i.e. we know for a fact that by the time of the next toolchange we will be using the newly fed bit), we activate follow-up no questions asked. This scenario may be impossible on multicolor prints on printers with long bowdens from MMU, but we must check anyway.
* If the remaining length to print is smaller than what we have left, but the difference is small enough (~10 cm, or configured by the user `purge_threshold`), we don't follow up, we continue printing with what is loaded, and we park and eject the remainder through the nozzle, wipe if needed. We do our best to ensure the next runout sensor is ignored.
* If the remaining length to print is very small (difference > `purge_threshold` or what was configured), AND Endless Spool is configured for this tool, we unload the remainder and instead of status "Buffered" we set a status "Running out". We know that this is a buffered state with a known length of filament left. We then load the next gate for this tool.
> The next time we load this tool, we also check how much is expected to print, and, if it seems reasonable to load this remainder instead of loading from a full spool, we use it up, and eject what is left through the nozzle if needed.

> If we have spoolman configured for this tool, and the remaining length in the spool is known to be short, we can use similar logic to choose between several gates for the same tool to avoid situations where we'd have to purge, and we can even mark a gate as EMPTY if we know that the length is not sufficient to load. - Spoolman part is to be thought through.
* If the remaining length to print is very small AND there is no EndlessSpool configured for this tool, we pause the print and inform the user (current behavior). We give options to continue printing with what is left and then eject the remainder through the nozzle, continue printing and then let the user extract filament from the toolhead manually, or we let the user re-load the gate with a new spool.