moggieuk/Happy-Hare #266: Save Blobifier.cfg backup on HH install/update

SkiBikePrint, 2024-04-10:
Although HH updates create a backup of the HH cfg files, it overwrites the existing files in the "addons" directory such as Blobifier.cfg and does not create a backup.  Please include backups for the cfg's in the addons directory as well.

moggieuk, 2024-04-19:
I'll certainly check this because I thought it worked.

The intent is that it does backup and then apply config and variables to new macros in the add-ons folder.  This allows for upgrade...

moggieuk, 2024-04-22:
@SkiBikePrint I checked the behavior on upgrade...

All the config files (including blobifier) are backed up in `mmu-<date>` folder.
Then, the macro variables from the previous blobifer.cfg are used to update the new blobifer.cfg config file.
If the blobifier_hw.cfg exists already then it will not be updated.

This is exactly the behavior of other HH config file upgrades.    Are you directly modifying the macros in `blobifier.cfg`?  If so this will be overwritten because (like other HH cfg files) the macros are meant to be upgraded (only config part retained).  If you are doing something really custom then you can copy and create your own uniquely named macro and point to that in the `mmu_macro_vars.cfg`

What am I missing?

SkiBikePrint, 2024-04-22:
Thanks for checking. and apologies for the exercise  I am modifying the blobifier.cfg file for a software only Trident version with no blobifier_hw.cfg so I will look at making the change to point to my own macro via mmu_macro_vars.cfg.  I'm traveling and don't have access to my machine but at the time of writing up this issue I wasn't seeing my version of blobifier.cfg.  Maybe I was just being stupid and not seeing the backup??

Please don't spend any more time on this and I will make you suggested changesm when I get back home.

Thanks again,

Glenn

closed 2024-10-28 by SkiBikePrint