moggieuk/Happy-Hare #268: -bash: ./install-sh: No such file or directory

slightlysketch, 2024-04-12:
no .install found
is there something wrong with the upload?
cant seem to pull anything?

using your script from the install walkthru so not sure why it isnt working.

slightlysketch, 2024-04-12:
well happy hare kinda trash.
been using ERCF 1.1 just fine, figured id try happy hare,
installer is broken, something wrong with .sh install file as i have it its there but the script to run it doesn't register,
can run other install scripts just fine, just not happy hare.

gotta fix this trash, when a 5 yr old software and config out performs this just based off install process.

closed 2024-04-12 by slightlysketch

ningpj, 2024-04-12:
Are you seriously after help or did you just drop by to have a moan since you haven't bothered to provide much in the way of information to go by?  

In my experience Happy Hare works exceptionally well and judging by questions and comments on the Voron ERCF discord forum, is widely used by the majority of ERCF users (1.1 and 2.0).  I use it with ERCF and Tradrack MMU's and are happy to help if you are prepared to provide more context and detail about the issue you have installing it.

Firstly though,  Happy Hare requires that you have a **reasonably current version** of Klipper and Moonraker installed. There are two reasons for this:

1) the Klipper/Moonraker directory structure was changed some time ago and the installer expects the new ``printer_data`` folder structure to be in place.  Based on your 5-yr driver comment, I'm unsure if you have updated this
2) Happy Hare uses a virtual Klipper toolhead object to do clever things (as does the native Tradrack MMU driver).  As the Klipper team have been enhancing and rolling out changes to the toolhead object over the last 4-5 months, Happy Hare and/or Klipper "may" need to be updated to maintain compatibility.

So, what version of Klipper and Moonraker do you have installed?
Do you have the following directory structure?
```
/home/<your pi user>
/home/<your pi user>/klipper
/home/<your pi user>/moonraker
/home/<your pi user>/printer_data
```
If not, Klipper and Moonraker need to be upgrade before you begin installing Happy Hare.
 
Also don't forget to removed the original ERCF 1.1 python driver (usually installed in ``/home/<your pi user>/klipper/klippy/extras`` folder) and its client macros.

Happy Hare is installed by cloning the GIT directory using the account you installed Klipper with so it can set the correct permissions and copy the files into the correct locations (I presume you already have git installed on your Pi).

```
cd ~
git clone https://github.com/moggieuk/Happy-Hare.git
cd Happy-Hare
```
Launch the installer in interactive mode to tailor it to your environment (ERCF v1.x, v2, Tradrack, Prusa), the mcu you use to run it, the features you want enabled, menus for 12864 screen, etc.  While the installer will try its best to setup the correct pins, you may need to review and tweak this.  In your case, provided you answer the questions correctly, it will set up pins for the Easy BRD and ask you to confirm its serial device.

``./install.sh -i``

Happy Hare also has a super useful, native KlipperScreen package that exposes advanced MMU visual controls if you have a 640x480 or higher resolution touch screen rather than basic 12864 display on your printer.

slightlysketch, 2024-04-12:
Well the issue stemmed from the copy and paste code. 
Typed it and it worked.