moggieuk/Happy-Hare #270: Read only files

Gr33n5murf, 2024-04-12:
As the title suggests, many files, including mmu_software.cfg, mmu_form_tip.cfg and mmu_cut_tip.cfg, are read only since the update to 2.5.1. Additionally [save_variables] was missing in mmu_software.cfg and some pin names for the bigtreetech mmb are missing. I solved the former by using WinSCP to repair the file.

Any idea how to remove the read only designation of the files?

ningpj, 2024-04-12:
Please refer to comment from dev in  https://github.com/moggieuk/Happy-Hare/issues/241#issuecomment-2039027529
Its by design, exten and customise through overrides. Also the args for tip form and cut are now set in a user updatable file https://github.com/moggieuk/Happy-Hare/blob/main/config/base/mmu_macro_vars.cfg

moggieuk, 2024-04-19:
As ningpj describes the "read-only" nature is deliberate and in keeping with other leading macro extensions (Mainsail macros, Klippain, etc).  There aren't really read-only, that's just the way Mainsail interprets them...

But the point is that they are this way to make upgrades easier but still allow for customization.  You can either just add macro calls at the designated spots with, for example:
```yml
variable_user_pre_unload_extension  : ''        ; Executed after default logic
variable_user_post_unload_extension : ''        ; Executed after default logic
```

Or, if the extension mechanism isn't sufficient then you can completely replace with macros of you own name and set this set of variables in `mmu_parameters.cfg`:
```yml
# ADVANCED: MMU macro overrides --- ONLY SET IF YOU'RE COMFORTABLE WITH KLIPPER MACROS -------------------------------
#
# 'pause_macro' defines what macro to call on MMU error (must put printer in paused state)
# Other macros are detailed in 'mmu_sequence.cfg'
#
pause_macro: PAUSE
action_changed_macro: _MMU_ACTION_CHANGED           # Called when action (printer.mmu.action) changes
print_state_changed_macro: _MMU_PRINT_STATE_CHANGED # Called when print state (printer.mmu.print_state) changes
pre_unload_macro: _MMU_PRE_UNLOAD                   # Called before starting the unload
post_form_tip_macro: _MMU_POST_FORM_TIP             # Called immediately after tip forming
post_unload_macro: _MMU_POST_UNLOAD                 # Called after unload completes
pre_load_macro: _MMU_PRE_LOAD                       # Called before starting the load
post_load_macro: _MMU_POST_LOAD                     # Called after the load is complete
unload_sequence_macro: _MMU_UNLOAD_SEQUENCE         # VERY ADVANCED: Optionally called based on 'gcode_unload_sequence'
load_sequence_macro: _MMU_LOAD_SEQUENCE             # VERY ADVANCED: Optionally called based on 'gcode_load_sequence'
```
Both these ways of customization will survive upgrades.

closed 2026-10-21 by moggieuk