moggieuk/Happy-Hare #272: MMU_START_LOAD_INITIAL_TOOL doesn't load tool on single color prints

iamnotahippy, 2024-04-18:
As the title says, MMU_START_LOAD_INITIAL_TOOL won't actually load the tool on a single color print. I'm not sure if this is intended behavior, but the culprit seems to be the line which reads:
``` {% if not using_bypass or tools|length > 1 %}```
Changing the line to:
``` {% if not using_bypass or tools|length >= 1 %}```
seems to fix the issue.

moggieuk, 2024-04-22:
I think you are correct and this is an oversight. I need to review and remind myself that there isn't a valid reason.... Some folks don't use the bypass and just load the tool/color they want to print with.  The slicer default for `initial_tool` is 0 and so it could result in frustrating behavior of undoing intent, but...

On reflection I do think that loading the initial tool (even on single color prints) might be more logical.

moggieuk, 2024-04-22:
Fixed. Update pushed.

demitrix, 2024-07-26:
The call to MMU_START_LOAD_INITIAL_TOOL now gives me this error. I'm not using bypass just a single color print. I can manually change to correct tool with T1 and resume.

MMU issue detected whilst printer is paused
Reason: Load bypass or initial tool before resuming print

mpsnp, 2024-07-26:
@demitrix do you have referenced tools in gcode? there should be at least one i guess 🤔

mpsnp, 2024-07-26:
I suppose you're removing/adding each time list of filaments in your slicer. Guess whole intention of TTG is to keep list of filaments static in slicer and change those only in case of swapping spools.
So instead of leaving only one in the list, you'd select right filament from the list for single color prints. 
Or i'm wrong @moggieuk ?

demitrix, 2024-07-26:
I didn't chance any spools in the my slicer, or gate mapping. The filament on gate 1 is also the second filament in my slicer. There is no tool change during the print so tool 1 should be the tool loaded by MMU_START_LOAD_INITIAL_TOOL, from my understanding.
Relevant g-code from a file
```
MMU_START_SETUP INITIAL_TOOL=1 REFERENCED_TOOLS=1 TOOL_COLORS=000000,000000,000000,0000FF,00FFFF,C0C0C0,FFFFFF TOOL_TEMPS=230,260,230,230,230,230,230 TOOL_MATERIALS=PLA,ASA,PLA,PLA,PLA,PLA,PLA PURGE_VOLUMES=0,60,60,307,586,459,560,60,0,60,307,586,459,560,60,60,0,307,586,459,560,80,80,80,0,540,562,661,114,114,114,256,0,272,408,60,60,60,241,213,0,210,80,80,80,272,242,60,0
MMU_START_CHECK
START_PRINT TOOL_TEMP=260 BED_TEMP=105 FILAMENT_TYPE=ASA
MMU_START_LOAD_INITIAL_TOOL
LINE_PURGE
```
START_PRINT is just handling homing, bed leveling, and preheating.

closed 2024-12-11 by github-actions[bot]