moggieuk/Happy-Hare #273: Feature Request: Bring back old "sync_unload_length" feature from HHv1

dohlin, 2024-04-18:
The old "sync_unload_length" parameter that was present in HHv1 was awesome - with that my filament ejects were 100% rock solid reliable.  With HHv2, however, a small portion of the time I get failures on eject after tip forming completes and the ERCF servo engages to pull the filament back.  In a long bowden tube setup such as mine, any slightest bit of stickage in the extruder gears can cause unload failure.  The "toolhead_unload_safety_margin" isn't really a complete replacement either, as if there's some hairs or little bits of the tip remaining in the extruder gears after tip forming then that extra margin can just end up grinding those up into the tip and buggering up its shape up even further.

For reference, this is the feature I'm referring to from HHv1:

```
How far (mm) to run gear_stepper and extruder together in sync on load and unload. This will make loading and unloading
more reliable and will act as a "hair pulling" step on unload.  These settings are optional - use 0 to disable
Non zero value for 'sync_load_length' will synchronize the whole homing distance if toolhead sensor is installed
sync_load_length: 10            # mm of synchronized extruder loading at entry to extruder
sync_unload_length: 10            # mm of synchronized movement at start of bowden unloading
```

The old behavior where a short distance of _syncronized_ unloading was possible was perfect.  Is there a specific reason this was removed?  Is there any shot at getting it back?  Thank you!!

moggieuk, 2024-04-19:
The reason it was removed was really because I felt it was unnecessary (at least when used in combination with other options) and in an attempt to simplify what was becoming complex setup with a lot of variations to debug.  Do you turn on syncing when forming tips?   I ask because there is logic to minimize the servo thrashing and thus if the extruder is synced during the tip forming it will remain synced until the filament is out of the extruder.  JFYI the most reliable setup is to have a (pre) extruder sensor.  Then the unload can move to that point in synchronized fashion before fast bowden move.

Somewhat related there is also an option called `bowden_pre_unload_test` in `mmu_parameters` which will pull an "encoder_move_step_size" (the move distance which the encoder will always detect .. 15mm default). 

I'm not opposed to bringing back a feature that works but I'm trying to understand where your filament tip is prior to the bowden move.  It sounds as though your tip forming logic maybe too fast than sometimes skips steps leaving the filament slightly short of exiting the extruder gears? If so you could try slowing down speeds. I'm also unsure why `toolhead_unload_safety_margin` isn't sufficient, because if the extruder is synced, then this is very similar to the former `sync_unload_length`

dohlin, 2024-04-19:
I don't currently have syncing when forming tips on, as every time I've used it anecdotally seems like it jams even more - although I can't scientifically prove that.

The biggest reason I suspected it was not related to tip forming slippage is that occasionally when it would bind up I would unscrew the side of my (old) Orbiter v2 without touching anything to see where the filament was at.  To my surprise, on more than one occasion the filament seemed like it had made it out of the gears, but the tip had become bent and mangled up, sort of like an "L" shape.  The only thing I could think of at the time was that maybe it was related to the "safety unload margin", and decreasing that to around 2-3 did seem like it helped at least a little.  I don't know if that makes any sense, or if that's even possible, that was just where my mind was going.

I can try slowing down my tip forming a little and see if it improves reliability...and I have definitely considered adding a pre-extruder sensor - that's probably next on my to-do list honestly.  It sounds like if I could get a pre-extruder sensor paired with sync'ed tip forming I could pretty much replicate the "old" functionality correct?

If so, is there any doc on what config changes I'd need to make after retroactively adding something like that?

dohlin, 2024-04-26:
> Yes to you question and pre-extruder sensor and sync tip forming. Let me know how it is going because I haven't heard any other reports that would amplify the need for the old feature (and it would be confusing, e.g. incompatible with the pre-bowden test).

Will do!  I think there's some real promise here, but I tried about 4 different sensor designs so far and every single one either doesn't trigger 100% reliably, or is just right enough that occasionally on filament homing (loading) the filament must just be bent enough to where it causes failure on load.  If I can get something with (practically) 100% reliability I might be golden, so still digging!

Side note while I have you - I don't know if these sorts of load issues happen to other people too or if it's just me...but it seems like it would be super cool for HH to be able to simply run an eject, tip cut, and then reattempt homing.  Sure, it's a workaround, but if all the logic for those functions is already created then it could only help in overall error recovery ability and reliability, right?

dohlin, 2024-04-30:
I'll close this up for now as I think I'm closing in on a reliable pre-extruder filament sensor solution.  The hardest part has been finding on that is consistently resistance-free (or as close as humanly possible) no matter what slight angle difference the filament feeds in at.  But I think I'm close so we'll consider this resolved for now.  Thanks!

closed 2024-04-30 by dohlin