moggieuk/Happy-Hare #274: Multiple servo angles

micahvdm, 2024-04-18:
So I’m designing a MMU unit that doesn’t have gates, but instead uses a servo to set an idler(like the mmu2) to the correct position(set angle) and then has a merger with a sensor at the exit like a gate sensor. 

Would it be feasible to modify Happy Hare to make this work?

I suppose the first question is can the servo currently accept multiple angles other than just up and down angle.

second question, is it possible to fake the gates so that the selector sets the servo angle instead of moving the selector to the selected filament?

thanks

moggieuk, 2024-04-19:
You have probably seen this already: https://github.com/moggieuk/Happy-Hare/blob/main/doc/conceptual_mmu.md

Support for these is how HH is designed.  That said there is still a small about of work to do to support type-B and type-C that I have not had the time to finish, but will soon.   From there it isn't far to support Prusa MMU2/3 which sounds like it is similar to your design.

Bottom line is that HH is close to being about to support what you describe (after I get my enhancements live) but you will still likely need to modify slightly...

micahvdm, 2024-04-19:
That’s awesome news! I’ll be happy to test any enhancements for you and even add some functionality for the mmu2/3 via pr. would you be able to dm me what you have so far so I can go through it?

micahvdm, 2024-04-24:
I've managed to add another option like the virtual_selector (called servo_selector) that bypasses the normal selector routines if true. Got it 98% done so far, but not sure if it'll be worth doing a PR for as its very edge case usage. Let me know if you're interested and I'll pop a link to the changes.

moggieuk, 2026-10-21:
Closing this as Happy Hare v4 now supports this kind of design directly with the `ServoSelector`.

It uses one servo to select the filament, with a separate angle per gate (`servo_gate_angles`), plus an optional release angle (`servo_release_angle`) and bypass angle. To set it up, choose "Custom Design" as the MMU type in menuconfig, pick "ServoSelector" as the selector type, then run `MMU_CALIBRATE_SERVO_SELECTOR` to record the angle for each gate. A sensor at the exit of your merger can be the gate homing endstop ("Shared exit / hub sensor", i.e. `gate_homing_endstop: mmu_shared_exit`). The Pico MMU preset uses the same selector if you want an example to start from.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk