moggieuk/Happy-Hare #275: Belay sensor never triggers on printer restart and first print

igiannakas, 2024-04-19:
There is a pretty common case where the Annex tension belay sensor never triggers when the printer is restarted.

If after a reboot the filament is expanded (sensor not triggered) it registers it as neutral as the pin has never triggered for it to update its state. So it considers it a neutral and uses multiplier 1. 

However this will result in the filament possibly always expanding if your extruder is ever so slightly under extruding compared to the ercf. 

Just woken up to an overnight print with the belay fully expanded with plenty of tension in the filament. Checking the logs it appears that the sensor never triggered. It started working when I pushed the lever manually.

**Log file:** 
[mmu_2.log](https://github.com/moggieuk/Happy-Hare/files/15038336/mmu_2.log)

I believe this is caused due to the way the sensor is initialised. 

Looking at the below it is set initially to neutral:
[image]

Then it is initialised as neutral:
[image]

However when the filament is loaded into the extruder, the sensor will expand as the filament is pushed on the extruder gears creating some tension in the filament path. If your extruder is not over extruding compared to the ERCF motor then it will never trigger the tension sensor. Worse, if it is slightly under extruding it will keep raising compression in the filament. 

If you have a dual sensor this situation can never occur. Also if you have a compression sensor, it is far less likely to happen as the filament will most likely result in filament compression on initial tool load triggering the sensor.

Ideally, to correct this and make it more reliable, the initialisation routine above needs to check what type of sensor is fitted. If:
1. Tension sensor: Set the sync_feedback_last_state to -1, triggering the sensor to start contracting the filament until the switch is hit.
2. Compression sensor: Set the sync_feedback_last_state to 1, triggering the sensor to start expanding the filament until the switch is triggered
3. Dual sensor or proportional sensor: Leave it to 0 and let the sensors do their thing on first update.

Unfortunately my knowledge of the code is limited to be able to raise a PR but I'd appreciate your thoughts on this and possibly a fix? I've manually set this like the below on my local branch to work around the issue with the annex sensor but this won't work universally:

[image]

moggieuk, 2024-04-22:
Thanks for the detailed explanation.  I'll take a look at this because it would be a corner case.  My test machine has both tension and compression sensors so maybe I missed this case....

igiannakas, 2024-04-22:
> Thanks for the detailed explanation. I'll take a look at this because it would be a corner case. My test machine has both tension and compression sensors so maybe I missed this case....
> 
> Anyway, I'll address in a day or two and push patch..

No worries! Which sensor are you using btw?

moggieuk, 2024-04-25:
Sorry. I have a customized sensor of my own.  I also have an experimental proportional feedback device.  My setup varies all the time as I develop and test Happy Hare.  Anyway, I think the fix to this is quite simple and I'm going to fix today.  Expect update tomorrow...

moggieuk, 2024-04-25:
Ok. I pushed a fix for this.  When a print starts the sync feedback is correctly set based on available sensors.  I'm confident this will work but perhaps you can validate before I close out this Issue.   Just need to force an update and restart of klipper.

igiannakas, 2024-04-25:
Of course thank you! I’ll grab the latest version when the current print is done and let you know

igiannakas, 2024-04-26:
OK I've installed the latest version, force updated etc. Will be running a longish print later today and will report back.

moggieuk, 2024-04-26:
Great.  I've been thinking about my fix and it isn't perfect.  I might work for you but I need to engineer a more correct fix.  What I have done still has corner cases.  So I'm taking off the "fixed" label on this issue until I can revist.

igiannakas, 2024-04-26:
> Great. I've been thinking about my fix and it isn't perfect. I might work for you but I need to engineer a more correct fix. What I have done still has corner cases. So I'm taking off the "fixed" label on this issue until I can revist.

What are you thinking as the edge cases? So I know what to look out for or what to test/work around?

MilosMrihlad, 2025-01-01:
I have a problem with belay, for example, that when I add it to the configuration, it still looks OPEN. Even when I switch it over it is still open after checking it in the console. I checked the wiring and the switch and everything is fine there. So I don't know if maybe it's also related somehow. Moreover, I didn't read anywhere whether with ercf it is possible to use normal synchronization with the addition of belay.

closed 2025-02-18 by igiannakas