moggieuk/Happy-Hare #276: RESUME after a failed tool change skips the tool change

Sparkss, 2024-04-20:
I am having an issue with my encoder starting to fail on long prints. This is causing the Bowden pre-unload test to fail. What I noticed was that when a failure occurs during a tool swap and then Resume is called, it skips the tool swap and continues printing with the "old" filament. In the attached mmu.log there are multiple occurrences, one such example starts at 22:11:54.

As seen in the logs both the toolhead and extruder sensors show that they are empty and visually the filament end is pulled into the end of the bowden about 1 or 2 mm. When it does try to "reload" the previous filament, T0,  (also shown by entries in the log) it actually runs the ERCF gears at full speed for the length of the Bowden, as if it were trying to load from the gate/selector, when the filament is actually already just a couple of mm above the toolhead. So the ERCF ends up chewing a flat spot on the filament, which just leads to more encoder "read" issues. It does load the extruder though (with T0 again) and then starts printing, completely skipping the tool swap to T1 and the print lines that should have happened with T1.
[mmu(1).log](https://github.com/moggieuk/Happy-Hare/files/15046744/mmu.1.log)

Sparkss, 2024-04-20:
happy-hare
v2.5.1-45-ga79354d8

ERCF v2

variable_restore_xy_pos         : "next"	; last|next|none - What x,y position the toolhead should travel to after a toolchange

moggieuk, 2024-04-22:
The expected pattern is that if the MMU "pauses" because of error, you need to manually get it to the state it wants before resuming.   It is on my to-do list to add some warnings if that is not the case, but if say the MMU fails on T0 --> T1 tool change, then you are expected to load up T1 before hitting resume...

HH keeps a "state" or where it thinks the end of the filament is.  It uses whatever sensors it has to validate that position, but there are many conditions where it would be impossible to deduce that the state is incorrect.  In the case you describe, the filament could actually be anywhere in the bowden assuming the toolhead and extruder sensors don't show filament.  HH thus assumes the filament is unloaded and does full load. Two ways to solve this:
1. unload the filament completely and resume
2. load the filament completely and resume

So whilst, I agree I can add some more defensive warnings, you have to ensure that the filament state in HH is correct before resuming.  You can do this with the `MMU_STATUS` command when paused and look that this graphic:
```
[T?] < En ......... [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:96.1mm)
```
and confirm it represents where the filament is.

PS `MMU_RECOVER` command can help reset the state..

Sparkss, 2024-04-24:
Thank you for the detailed response. I guess I am just accustomed to the behavior of Prusa's MMU2/firmware. If an unload failed it would re-attempt it on a resume, the same for a load. If what I encountered is the state of ERCF/HH then I just need to get used to it and familiarize myself more with the manual remediation commands and where they can be found in the menus. Thank you very much for all of your time and hard work in creating HH and the ERCF Klipper Screen modification and for sharing them with the community.

closed 2024-04-24 by Sparkss