moggieuk/Happy-Hare #277: "Must home axis first" at start of print if not homed before starting

charredchar, 2024-04-20:
I receive an error stating I need to home when selecting a file to print if I do not home the printer before selecting the file.
Since I use Klicky-00 I have my homing routine within my start gcode which is just after the `MMU_START_SETUP` in my slicer. But the printer does not get that far into the start gcode before triggering this error. I can not move this homing before the `MMU_START_SETUP ` as the rest of my start macro depends on the specific homing used within the macro.
Removing `MMU_START_SETUP ` from my start gcode in the slicer allows the printer to start without an issue except for the fact nothing gets passed to HH.

[klippy.log](https://github.com/moggieuk/Happy-Hare/files/15047290/klippy.log)

moggieuk, 2024-04-22:
I you sure you are not calling something extra in the `MMU_START_SETUP`?  E.g. is
```yml
variable_user_pre_initialize_extension      : ""
```
set to anything?

The MMU_START_SETUP should not attempt to move the toolhead during setup (unless an error occurs and even then it will just report a warning).

charredchar, 2024-04-22:
The only option I currently have for `MMU_START_SETUP` is `INITIAL_TOOL` since I do not have the other options set up yet.

I originally had `G28` in the `variable_user_pre_initialize_extension` option (as it is there by default) but it was causing a second unneeded homing as I have one near the start of my start gcode.

I attempted to remove it (leaving it empty as your example) but I would receive a Klipper error;
`Option 'variable_user_pre_initialize_extension' in section 'gcode_macro _MMU_SOFTWARE_VARS' is not a valid literal: unexpected EOF while parsing (, line 0)`

So I changed it to `M117` just to have something there and the error of not having anything there went away.

And while I was typing this out I decided to test a little more... I found the issue and I am a little annoyed at the cause. 😅
For some reason SuperSlicer added a move before *any* of my start gcode. I've never seen this before using an MMU and I've seen this error happen more than once so I (falsely) assumed it was MMU related and not my slicer. Sorry, I should have check my gcode file first. I will continue to troubleshoot this with SuperSlicer.
But I do want to change this issue to the semi-related error with `variable_user_pre_initialize_extension` being empty noted above. Having `M117` in there is kind of more of a hack than anything, it should allow for it to be empty.

moggieuk, 2024-04-25:
This isn't actually a bug, but rather a limitation of jinja templating and evaluation.  I suspect you are still running under Python2.  In this version (which has been deprecated for 3 years now) you cannot set to a double quoted string.  I know that a single quoted string will work like:
```yml
variable_user_pre_initialize_extension      : ''
```

Because it is a "feature" of python there is nothing I can do to support double quote on Python 2.

Re Slicer issue:  SuperSlicer does have quirks -- one reason I stick with PrusaSlicer (they are annoyingly slow at taking PRs and updates but there QA is good).   But thanks for the info!

If you can validate the single quote and Python2 setup then I can close out this issue.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

v4 runs on Python 3 only and the installer writes these extension values as proper string literals, so leaving one empty now just works.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk