moggieuk/Happy-Hare #281: Issue with pause/resume client macros when MMU ENABLE=0

Saikedo, 2024-04-24:
I encountered an issue when using pause resume functionality inside mmu/optional/client_macros.cfg when MMU ENABLE=0.

After print is paused and then resumed, printer is throwing the following error. Note that this issue does not happen when MMU is enabled. I also do not have any custom pause resume macros besides what is in client_macros.

I assume this is due to some absolute vs relative state that is not being restored after the pause-resume macro is done executing.
I tried setting the state in PAUSE and then restoring the state in RESUME but that did not help. The same issue that was observed before was still happening.

Any ideas what might be getting wrong?

moggieuk, 2024-04-25:
I believe this is fixed. 

 It is an tiny update to the client_macros.cfg.   Specifically the optional "retraction" macro sets the extruder to relative but doesn't restore.  Without the HH overarching protection which is disabled when Happy Hare is disabled, this would lead to your problem.   Can you verify for me?
Specifically the change was to wrap the retraction logic with a save and restore state..
```yml
[gcode_macro _UNRETRACT]
description: Helper to extruder filament in resume to undo retract
gcode:
    {% set vars = printer['gcode_macro _MMU_CLIENT_VARS'] %}
    {% set length = params.LENGTH|default(vars.unretract)|default(1.0)|float %}
    {% set speed = params.SPEED|default(vars.unretract_speed)|default(1.0)|float %}

    SAVE_GCODE_STATE NAME=MMU_RETRACT_state

    {% if printer.extruder.can_extrude %}
        M83 ; Extruder relative
        G1 E{length} F{speed|abs * 60}
    {% endif %}

    RESTORE_GCODE_STATE NAME=MMU_RETRACT_state
```

Saikedo, 2024-04-25:
This was fixed, closing it down. Thanks.

closed 2024-04-25 by Saikedo