moggieuk/Happy-Hare #282: Can someone share their Config?

Durahl, 2024-04-25:
Can someone running their ERCF off the Printers Mainboard instead of an Addon Board share their ENTIRE Printers config package?

The broken Installer ( providing 7 Options but only accepting 6 ) doesn't account for a System running the ERCF not off an Addon Board but instead the Printers Main Board ( like a BTT Octopus V1.1 ) and I can't seem to figure it out on my own.

moggieuk, 2024-04-25:
The 6 vs 7 selection is fixed.  Best to use option 7 and then adjust.

I would suggest commenting out the `[board_pins]` section in `mmu.cfg` and then setting all the pins in `mmu_hardware.cfg` manually...  you can use aliases if you would like but you might find it easier to use explicit pin names.

Durahl, 2024-04-25:
How does the MCU or lack thereof need to be handled?
All Combinations I've tried produce errors...

`[mcu mmu]`
`serial: /dev/ttyACM1` # Config guess. Run ls -l /dev/serial/by-id and set manually	# Change to `canbus_uuid: 1234567890` for CANbus setups
[image]
`[mcu mmu]`
`#serial: /dev/ttyACM1`  # Config guess. Run ls -l /dev/serial/by-id and set manually	# Change to `canbus_uuid: 1234567890` for CANbus setups
[image]
`#[mcu mmu]`
`#serial: /dev/ttyACM1` # Config guess. Run ls -l /dev/serial/by-id and set manually	# Change to `canbus_uuid: 1234567890` for CANbus setups
[image]

including one where I add the Printers MCUs Serial and change the `MMU` to `MCU`:
`[mcu mmu]`
`serial: /dev/serial/by-id/usb-Klipper_stm32f446xx_1C003C00115053424E363620-if00` # Config guess. Run ls -l /dev/serial/by-id and set manually	# Change to `canbus_uuid: 1234567890` for CANbus setups
`[board_pins mmu]`
`mcu: mcu` # Assuming using an external / extra mcu dedicated to MMU
`aliases:`
[image]

Durahl, 2024-04-26:
I can't be the only person not having this issue and just providing a visual Guide ( Screenshot ) of all the parts that need to be commented out in the `mmu.cfg` would really help getting this fixed cause apparently just saying "_comment out the board pins_" isn't cutting it - As seen in my last Post.

I also really don't get why the solution of just using the Printers Main Board wasn't even taking into consideration when building this Software. A Trident running a BTT Octopus V1.1 can run an ERCF without any additional Boards and a V2.4 can do the same by freeing up a Stepper Driver Connection via the use of a much more common / future proof Tool Head CAN board instead of what I'd describe a niche ERCF CAN Board.

ningpj, 2024-04-27:
The majority if not all v2 users are using dedicated mcu's given the extra endstop ports required for pregate sensors.

Your main printer mcu is the default and already references so dont need the ``[mcu mmu]`` block at all as all pins (steppers, end stops and servo) can be referenced directly.  Go through ``mmu_hardware.cfg`` and remove the``mmu:`` prefix from all HH pin settings and set these to the corresponding pin/gpio you have used on your mainboard mcu just like other settings in printer.cfg you have used for your X/Y/Z steppers, endstops, etc.

For example your gear stepper block would change to something like the following (you need to replace gipo's to match the ports you have used) 
```
[tmc2209 stepper_mmu_gear]
uart_pin: gpio9
...
[stepper_mmu_gear]
step_pin: gpio7
dir_pin: !gpio8
enable_pin: !gpio6
```

ningpj, 2024-05-09:
Have you managed to get your octopus configured to run Happy Hare?

closed 2026-10-21 by moggieuk