moggieuk/Happy-Hare #286: Questions about vars.cfg

Mrmomomomi, 2024-05-07:
[image]

ningpj, 2024-05-07:
by default, Happy Hare saves calibration and run state information in a persistent file called ``mmu/mmu_vars.cfg``. The location and name of this file is defined in ``mmu/base/mmu_macro_vars.cfg`` under the ``[save_variables]`` block and needs to be writable by the pi user kipper is installed to run as.  If you are getting this error, the klipper user likely doesn't have permissions to write to ``<klipper pi user home>/printer_data/config/mmu`` OR you have a full or faulty SD card.

Mrmomomomi, 2024-05-07:
> 默认情况下，Happy Hare 将校准和运行状态信息保存在名为 的持久文件中。此文件的位置和名称在块下定义，需要由安装 pi 用户 kipper 的 pi 写入才能运行。如果您收到此错误，则 klipper 用户可能没有写入权限，或者您的 SD 卡已满或有故障。`mmu/mmu_vars.cfg``mmu/base/mmu_macro_vars.cfg``[save_variables]``<klipper pi user home>/printer_data/config/mmu`

I'm using the Raspberry Pi CM4 EMMC version.,Actually, I don't know much.,Can the great god give me a list of step-by-step solutions.

ningpj, 2024-05-07:
We will see if we can help.

``ssh`` to the Pi CM4 using the account you normally use.
send the output from the following commands to confirm the home directory for the user, folders and folder / file permissions.

``uname -a; id; pwd; ls -al``

It will return something like the following:
```
Linux vorondev 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),5(tty),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),997(gpio),998(i2c),999(spi)
/home/pi
total 164
drwxr-xr-x 21 pi   pi    4096 May  7 18:09 .
drwxr-xr-x  3 root root  4096 Feb 21  2023 ..
drwxr-xr-x  9 pi   pi    4096 Dec 12  2021 arducam
-rw-------  1 pi   pi   45305 May  7 18:09 .bash_history
-rw-r--r--  1 pi   pi     220 Feb 21  2023 .bash_logout
-rw-r--r--  1 pi   pi    3523 Feb 21  2023 .bashrc
drwxr-xr-x  8 pi   pi    4096 Apr  3  2023 BOSSA
drwxr-xr-x  8 pi   pi    4096 Jun  3  2023 .cache
drwx------  5 pi   pi    4096 Jun  3  2023 .config
drwxr-xr-x 10 pi   pi    4096 Apr  3  2023 crowsnest
drwxr-xr-x  5 pi   pi    4096 May  4 18:52 fluidd
-rw-r--r--  1 pi   pi      55 Apr  3  2023 .gitconfig
drwxr-xr-x  8 pi   pi    4096 May  4 18:53 Happy-Hare
...
```

Mrmomomomi, 2024-05-07:
> uname -a; id; pwd; ls -al

Please ignore the username, I changed it myself 

Linux voron-300 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
uid=1000(biqu) gid=1000(biqu) groups=1000(biqu),4(adm),5(tty),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),997(gpio),998(i2c),999(spi)
/home/biqu
total 108
drwxr-xr-x 20 biqu biqu 4096 May  7 14:21 .
drwxr-xr-x  3 root root 4096 Jun 18  2023 ..
-rw-------  1 biqu biqu  522 May  6 21:50 .bash_history
-rw-r--r--  1 biqu biqu  220 May  3  2023 .bash_logout
-rw-r--r--  1 biqu biqu 3523 May  3  2023 .bashrc
drwxr-xr-x  3 biqu biqu 4096 Jan 21 03:13 .cache
drwxr-xr-x 10 biqu biqu 4096 May  6 20:29 crowsnest
drwxr-xr-x  8 biqu biqu 4096 May  6 20:56 Happy-Hare
drwxr-xr-x  7 biqu biqu 4096 May  6 20:20 kiauh
drwxr-xr-x  3 biqu biqu 4096 May  6 20:22 kiauh-backups
-rw-r--r--  1 biqu biqu  561 May  6 21:27 .kiauh.ini
drwxr-xr-x 11 biqu biqu 4096 May  6 20:28 klipper
drwxr-xr-x  9 biqu biqu 4096 May  6 21:28 klipper-led_effect
drwxr-xr-x  9 biqu biqu 4096 May  6 20:22 KlipperScreen
drwxr-xr-x  6 biqu biqu 4096 May  6 20:25 .KlipperScreen-env
drwxr-xr-x  5 biqu biqu 4096 Jan 21 03:12 klippy-env
drwxr-xr-x  3 biqu biqu 4096 Jan 21 03:11 .local
drwxr-xr-x  5 biqu biqu 4096 May  6 20:30 mainsail
drwxr-xr-x  3 biqu biqu 4096 Jan 21 03:16 mainsail-config
drwxr-xr-x  8 biqu biqu 4096 May  6 20:29 moonraker
drwxr-xr-x  4 biqu biqu 4096 Jan 21 03:13 moonraker-env
drwxr-xr-x  8 biqu biqu 4096 Jan 21 03:16 moonraker-timelapse
drwxr-xr-x 11 biqu biqu 4096 May  6 20:29 printer_data
-rw-r--r--  1 biqu biqu  807 May  3  2023 .profile
drwxr-xr-x  8 biqu biqu 4096 Jan 21 03:24 sonar
-rw-r--r--  1 biqu biqu  165 May  6 20:30 .wget-hsts
-rw-------  1 biqu biqu   55 May  7 14:21 .Xauthority

Mrmomomomi, 2024-05-07:
> I'll see if Ican help.
> 
> `ssh` to the Pi CM4 using the account you normally use. send the output from the following commands to confirm the home directory for the user, folders and folder / file permissions.
> 
> `uname -a; id; pwd; ls -al`
> 
> It will return something like the following:
> 
> ```
> Linux vorondev 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
> uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),5(tty),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),997(gpio),998(i2c),999(spi)
> /home/pi
> total 164
> drwxr-xr-x 21 pi   pi    4096 May  7 18:09 .
> drwxr-xr-x  3 root root  4096 Feb 21  2023 ..
> drwxr-xr-x  9 pi   pi    4096 Dec 12  2021 arducam
> -rw-------  1 pi   pi   45305 May  7 18:09 .bash_history
> -rw-r--r--  1 pi   pi     220 Feb 21  2023 .bash_logout
> -rw-r--r--  1 pi   pi    3523 Feb 21  2023 .bashrc
> drwxr-xr-x  8 pi   pi    4096 Apr  3  2023 BOSSA
> drwxr-xr-x  8 pi   pi    4096 Jun  3  2023 .cache
> drwx------  5 pi   pi    4096 Jun  3  2023 .config
> drwxr-xr-x 10 pi   pi    4096 Apr  3  2023 crowsnest
> drwxr-xr-x  5 pi   pi    4096 May  4 18:52 fluidd
> -rw-r--r--  1 pi   pi      55 Apr  3  2023 .gitconfig
> drwxr-xr-x  8 pi   pi    4096 May  4 18:53 Happy-Hare
> ...
> ```
> 
> Also from the mainsail or fluidd user interface, please open `mmu/base/mmu_macro_vars.cfg` and attach what you have for the `[save_variables]` block.

[image]

ningpj, 2024-05-07:
Ok so its a BTT CM4 module running their Debian based linux image.
Can you please run

``ls -al printer_data/config/mmu/mmu_vars.cfg``

and 

``cat printer_data/config/mmu/mmu_vars.cfg``

Mrmomomomi, 2024-05-07:
> Ok so its a BTT CM4 module running their Debian based linux image. Can you please run
> 
> `ls -al printer_data/config/mmu/mmu_vars.cfg`
> 
> and
> 
> `cat printer_data/config/mmu/mmu_vars.cfg`

> Ok so its a BTT CM4 module running their Debian based linux image. Can you please run
> 
> `ls -al printer_data/config/mmu/mmu_vars.cfg`
> 
> and
> 
> `cat printer_data/config/mmu/mmu_vars.cfg`

My motherboard is BTT M8P core board is Raspberry Pi CM4 The username is my custom, the machine is printing, I'll try it later

Mrmomomomi, 2024-05-07:
> Ok so its a BTT CM4 module running their Debian based linux image. Can you please run
> 
> `ls -al printer_data/config/mmu/mmu_vars.cfg`
> 
> and
> 
> `cat printer_data/config/mmu/mmu_vars.cfg`
> 
> Unfortunately I need to go out for a while but will check back later.

> Ok so its a BTT CM4 module running their Debian based linux image. Can you please run
> 
> `ls -al printer_data/config/mmu/mmu_vars.cfg`
> 
> and
> 
> `cat printer_data/config/mmu/mmu_vars.cfg`
> 
> Unfortunately I need to go out for a while but will check back later.

Hello, friend of which country are yours?

Mrmomomomi, 2024-05-08:
> Ok so its a BTT CM4 module running their Debian based linux image. Can you please run
> 
> `ls -al printer_data/config/mmu/mmu_vars.cfg`
> 
> and
> 
> `cat printer_data/config/mmu/mmu_vars.cfg`
> 
> Unfortunately I need to go out for a while but will check back later.

After executing the command you sent, the same error is still reported

ningpj, 2024-05-08:
Can you copy the output from the commands above.  Still trying to verify your setup and permissions.  In all likelihood, rerunning Happy Hare ``./install.sh`` without arguments “may” fix this but need to confirm it you have an underlying permission issue.

Mrmomomomi, 2024-05-08:
> cat printer_data/config/mmu/mmu_vars.cfg

[image]

Mrmomomomi, 2024-05-08:
[image]

ningpj, 2024-05-08:
Can you please run the following commands and upload the results:

``ls -al printer_data/config/mmu/mmu_vars.cfg``

and

``cat printer_data/config/mmu/mmu_vars.cfg``

Mrmomomomi, 2024-05-08:
[image]

ningpj, 2024-05-08:
Everything looks fine.   Have you run through the calibration steps outlined in the Happy Hare documentation to calibrate your selector, servo, gear stepper and encoder?

https://github.com/moggieuk/Happy-Hare/blob/main/doc/calibration.md

Mrmomomomi, 2024-05-08:
Boss, I'll contact you tomorrow, the machine is still working, I'll try again in the afternoon if the rabbit still has an error

ningpj, 2024-05-08:
It needs to be calibrated and will need to run though the steps in the link above.

Mrmomomomi, 2024-05-08:
At the beginning, I installed and started debugging, and I knew that I started to tune the encoder, and the encoder was broken, and then I purchased parts and waited for the goods. After the arrival and loading, it was reported that the error was reported, and the machine could not run if there was an error, so I commented out the relevant files first and let the machine work

Mrmomomomi, 2024-05-08:
[image]
[image]

It's still the same error

moggieuk, 2024-05-16:
Looks like @ningpj has given you all the correct advice but it appears you didn't say "add the Happy Hare includes" when you ran  the `./install.sh -i` installer.  If you didn't do this you must add them manually.  Honestly it is best to let the installer do it for you.

closed 2026-10-21 by moggieuk