moggieuk/Happy-Hare #287: Unload fails sporadically

danyial, 2024-05-07:
I don't know why, but sometimes the unload fails and I'm sure the synced unload not works if it fails.
It tries to unload to the buffer but the filament is still inside the extruder (toolhead sensor is triggered)

```
15:15:14   DEBUG: Tool change initiated with standalone MMU tip forming
15:15:14 Tool T3 is already loaded
15:15:14     TRACE: Enabled runout detection
15:15:14     TRACE: (mmu_pre_gate_3 sensor detects filament)
15:15:14     TRACE: Running macro: _MMU_CLEAR_POSITION
15:15:14   DEBUG: Setting servo to up (filament released) position at angle: 85
15:15:15   DEBUG: Restoring toolhead height
15:15:15   DEBUG: Restored gcode state and z-hop position only (Z:3.6) after change_tool
15:15:22   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 4.4mm
15:15:22   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 3.6mm
15:33:38 > MMU_EJECT
15:33:38     TRACE: Disabled runout detection
15:33:38   DEBUG: Unloading tool T3
15:33:38     TRACE: Running macro: _MMU_PRE_UNLOAD
15:33:40 Unloading filament...
15:33:40     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Idle'
15:33:40     TRACE: (extruder sensor detects filament)
15:33:40     TRACE: (toolhead sensor detects filament)
15:33:40 [T3] < En <<<<<<<<< (*) << [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
15:33:41     TRACE: (extruder sensor detects filament)
15:33:41     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Forming Tip' OLD_ACTION='Unloading'
15:33:41     TRACE: _ensure_safe_extruder_temperature: current_temp=199.96, paused_extruder_temp=None, current_target_temp=200.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
15:33:41 Forming tip...
15:33:41     TRACE: Running macro: _MMU_CUT_TIP
15:33:47     TRACE: After tip formation, park_pos reported as: 38.5mm with 5.0mm filament remaining in extruder (extruder moved: 33.5mm, encoder measured 32.0mm)
15:33:47     TRACE: (extruder sensor detects filament)
15:33:47     TRACE: (toolhead sensor detects filament)
15:33:47 [T3] < En <<<<<<<<< (*) << [Ex << (*) <. Nz] -38.5mm (e:39.0mm)
15:33:47     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Forming Tip'
15:33:47     TRACE: Running macro: _MMU_POST_FORM_TIP
15:33:47     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Exiting Ext' OLD_ACTION='Unloading'
15:33:47   DEBUG: Extracting filament from extruder
15:33:47     TRACE: _ensure_safe_extruder_temperature: current_temp=200.45, paused_extruder_temp=None, current_target_temp=200.0, klipper_minimum_temp=170.0, default_extruder_temp=200.0, source=auto
15:33:47   DEBUG: Setting servo to down (filament drive) position at angle: 0
15:33:48     TRACE: (extruder sensor detects filament)
15:33:48   DEBUG: Reverse homing up to 110.0mm to extruder sensor (synced) to exit extruder
15:33:50     TRACE: Reverse homing to extruder sensor. Stepper: 'gear+extruder' homed after moving -6.2mm (of max -110.0mm), encoder measured 6.0mm (delta 0.2mm). Pos: @-44.7, (48.9mm)
15:33:50     TRACE: (extruder sensor detects filament)
15:33:50     TRACE: (toolhead sensor detects filament)
15:33:50 [T3] < En <<<<<<<<<|(*) .. [Ex .. (*) .. Nz] -44.7mm (e:48.9mm)
15:33:50     TRACE: (toolhead sensor detects filament)
15:33:50     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Exiting Ext'
15:33:50     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Unloading'
15:33:50 MMU issue detected. Print will be paused
         Reason: Unload sequence failed: Toolhead sensor still reports filament is present in toolhead! Possible sensor malfunction
15:33:50 After fixing, call RESUME to continue printing (MMU_UNLOCK to restore temperature)
```

Here is my config:

```
15:44:25 MMU: Happy Hare v2.5.1 running ERCF v2.0 with 6 gates (OPERATIONAL)
         Servo in UP position, Encoder reads 0.0mm
         Print state is COMPLETE. Selector is HOMED. Tool T3 selected  on Gate #3
         Gear stepper is at 100% and is not synced to extruder
         
         Load Sequence:
         - Filament loads into gate by homing a maximum of 70.0mm ('gate_homing_max') to ENCODER
         - Bowden is loaded with a fast CORRECTED 702.7mm ('calibration_bowden_length') move
         - Extruder (synced) loads by homing a maximum of 40.0mm ('toolhead_homing_max') to TOOLHEAD SENSOR before moving the last 54.0mm ('toolhead_sensor_to_nozzle - toolhead_ooze_reduction') to the nozzle
         
         Unload Sequence:
         - Tip is always formed by '_MMU_CUT_TIP' macro and tip forming extruder current is 100%
         - Extruder (synced) unloads by reverse homing a maximum of 110.0mm ('toolhead_entry_to_extruder + toolhead_extruder_to_nozzle + toolhead_unload_safety_margin') to EXTRUDER SENSOR
         - Bowden is unloaded with a fast 652.7mm ('calibration_bowden_length - gate_unload_buffer') move
         - Filament is stored by homing a maximum of 70.0mm ('gate_homing_max') to ENCODER and parking 13.0mm ('gate_parking_distance') in the gate
         
         Selector touch (stallguard) is DISABLED - blocked gate recovery is not possible
         Persistence: Gate status, TTG map & EndlessSpool groups state is persisted across restarts
         MMU has an encoder. Non essential move validation is ENABLED
         Runout/Clog detection is AUTOMATIC (8.4mm runout), EndlessSpool is ENABLED
         SpoolMan is DISABLED. Sensors: TOOLHEAD (Empty), EXTRUDER (Empty), 
         Logging: Console 4(STEPPER), Logfile 3(TRACE), Visual 1(ON), Statistics 1(ON)
         
         For details on TTG and EndlessSpool groups add 'DETAIL=1'
         
         Gates: | 0 | 1 | 2 | 3 | 4 | 5 |
         Tools: |T0 |T1 |T2 |T3 |T4 |T5 |
         Avail: | B | B | B | B | ? | ? |
         Selct: ------------| * |-------- T3
         
         [T3] < En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED 0.0mm (e:0.0mm)
```

Ambilights, 2024-05-09:
What sensor do you use on toolhead sensor?  I'm using filametrix with microswitch and ball and sometimes filament debris jams the sensor.

moggieuk, 2024-05-22:
This looks like your toolhead sensor is getting stuck...

Viewing the log you can see that the filament is "reversed homed to extruder sensor". That means it is retracted until the extruder (entrance) sensor no long sees the filament.

The safety check then spots that the toolhead sensor (inside the extruder) is still triggered and hence the pause -- there is no point in ignoring this because it will cause issues later.

Do you use the microswitch toolhead sensor? (Hall Effect switches are a disaster in my experience).  Also is the cut piece of filament somehow being pulled from the nozzle and blocking the toolhead sensor?

closed 2026-10-21 by moggieuk