moggieuk/Happy-Hare #292: Question regarding temperature during toolchange

Belstrekkie, 2024-05-13:
When printing filaments with different printing temps, the by the slicer applied M104 in the gcode is basically interpreted as a M109 during the tool change. I wonder why? For example a temperature difference of 20-30 degrees between to filaments, costs only 15-20 seconds to reach, while the entire toolchange in my case is 40-50 seconds. So why not start the toolchange directly after the M104 instead of waiting until the new temperature is reached? This would also be better for the filament with the lower printing temps, as it currently gets basically overheated, before being unloaded.

moggieuk, 2024-05-25:
I reviewed my code.  Indeed the wait should occur BEFORE the load_extruder() function with the new filament -- which would be correct.

However, I see that for some reason, it will always wait if the current_temp is < new_target_temp.   I know this was added for another reason so I need to figure out why, but I believe that is what you are seeing.   If you wanted to try you can find this in `mmu.py` (around line 6770 in the file):
```yml
        if current_temp < new_target_temp:
            wait = True
```

and comment it out like this:
```yml
#        if current_temp < new_target_temp:
#            wait = True
```

Let me know if this fixes the issue.

Belstrekkie, 2024-05-25:
Thanks! It will take a while before I can test it, because my U2C unfortunately died and I am waiting for a replacement :(

moggieuk, 2024-05-27:
When you can I'd appreciate feedback on whether this removal works.   I think it is the correct thing to do but I'd like confirmation from you before I remove it in `main` -- the temp change/wait logic is surprisingly tricky because of all the scenarios  both in print and out of print...

Belstrekkie, 2024-05-27:
No problem, as soon as I received the new canbus adapter, I'll test it (but it has to come from China, will take a while).

Belstrekkie, 2024-06-08:
I've finally been able to test your suggestion and it works perfectly for me, no more waiting and no side affects noted, thanks!

Belstrekkie, 2024-06-16:
Up until now, it worked fine with relativ small differences in target temp. However, now I'm trying to use two different types of filaments with a target temp difference of 60 degrees, which results in errors, as the low temp filament (target temp 200 degrees)  is not able to purge/flush the nozzle to clear it from the high temp filament (260 degrees) at the low nozzle temperature.

So, if possible, I would somehow need a solution that if the new_target_temp is lower than the default_extruder_temp or for example more than 30 degrees lower than the current_target_temp, the new_target_temp is only set after the initial purge and not before the filament change.

closed 2024-11-29 by github-actions[bot]