moggieuk/Happy-Hare #295: Add Encoder Checking to EREC?

markleaf131313, 2024-05-18:
Sometimes when going to do the cut action for the EREC addon, the filament doesn't make it outside to the cutter. So it ends up cutting nothing without giving any error. Would it be possible to include encoder checks for the moves to ensure that filament made it out?

moggieuk, 2024-05-22:
Can you attach `mmu.log` to really understand what is happening.  Note that EREC is fully implemented in a macro (means it can't use the normal HH mechanism. Still possible, just pointing that out).

markleaf131313, 2024-05-22:
I don't have the log and am having trouble getting it working reliably so I don't have it connected at the moment. There wouldn't really be anything in the log though. The filament gets stuck and doesn't go into the encoder, or it sprung back too far and can't be engaged anymore. So there are no errors. The cut action is run which doesn't push anything into the encoder, so nothing ends up being cut and it continues on since there is no error checking until it comes back to the gate on the next swap. It gives an error then when it sees it isn't pulling any filament.

moggieuk, 2024-07-10:
The logs would tell me if there is any logic issue (which I don’t think is the case).  The encoder is already in the gate parking, so it really shouldn’t be necessary to validate the “poke back” move.  It would have to be added to the macro so I would consider a PR with the change.

markleaf131313, 2024-07-11:
The problem is it gets stuck in the gate, never cuts it, so then it has to be manually cut. Doesn't matter I guess, couldn't get it to work so went with IDEX instead.

closed 2024-07-11 by markleaf131313