moggieuk/Happy-Hare #296: Blobifier with Front cut_tip

Dragi2k, 2024-05-19:
if you use Blobifier_Park with mmu_cut_tip and the cutting arm is in front than the toolhead will crash in to the Bucked shake arm on the way back after Cutting

moggieuk, 2024-05-25:
Flagging @Dendrowen the author of Blobifier.  May need a simple movement enhancement.. it is difficult to account for all setup possibilities.

Dendrowen, 2024-05-25:
I placed my pin at y_max - 6, which helped.

Dragi2k, 2024-05-25:
I can't move this far, it would be nice if the head would move up, over the shaker arm and then down to the blob.

I had the same issue with a print last night, were it crashed in the shaker arm

Dragi2k, 2024-05-27:
[image]
[image]
 is it possible to run it with this code ?
 eventualy add a part where it checks that the head is over a minimum hight

Dragi2k, 2024-06-01:
@Dendrowen @moggieuk i changed the cut tip to my needs, if you want have a look, i can send it to you on discord

moggieuk, 2024-06-26:
If there is need in additional functionality then it is best to open a PR because this is an enhancement to the tip cutting macro.  We can discuss on that changes..

christi-ko, 2024-06-30:
Hello is, the pull request already there? I have the same issue here. If you need someone for testing i am open to support.  Don't want to create something by myself if someone else did the work already :-)

Dragi2k, 2024-06-30:
i have the cfg working and can it send to you

the request will i start soon

christi-ko, 2024-06-30:
This would be very helpful. You can also just attach it in here so others can also try it ...

Dragi2k, 2024-06-30:
https://github.com/moggieuk/Happy-Hare/pull/328

but i dont know what the script will do if someone doesn't use bloblifier

seandboss, 2024-07-07:
> i have the cfg working and can it send to you
> 
> the request is addet

Could you send it to me please? Truly appreciated!

Dragi2k, 2024-07-07:
> > i have the cfg working and can it send to you
> > the request is addet
> 
> Could you send it to me please? Truly appreciated!

https://github.com/Dragi2k/Happy-Hare/blob/main/config/base/mmu_cut_tip.cfg
and
https://github.com/Dragi2k/Happy-Hare/blob/main/config/base/mmu_macro_vars.cfg

seandboss, 2024-07-15:
Stupid question, but how do I get this applied if the mmu_cut_tip.cfg is read only?

Dragi2k, 2024-07-15:
use winscp or something to exchange it, the file on the Printer is a shortcut

Dendrowen, 2024-07-16:
> Stupid question, but how do I get this applied if the mmu_cut_tip.cfg is read only?

The proper way to do this is to overwrite the macro in your own macro file, otherwise it gets removed on an update.

```
[gcode_macro ORIGINAL_MACRO]
rename_existing: ORIGINAL_MACRO_OLD
gcode:
  #call the old macro
  ORIGINAL_MACRO_OLD
  #add your own functionality
  M117 macro called!
```

seandboss, 2024-07-16:
> The proper way to do this is to overwrite the macro in your own macro file, otherwise it gets removed on an update.
>
> ```
> [gcode_macro ORIGINAL_MACRO]
> rename_existing: ORIGINAL_MACRO_OLD
> gcode: [...]

Thank you, I used the fact that Klipper uses the last called Macro with that name to my advantage in the end and loaded a customised macro file that overwrites these two with the new one. Works well!

Is there any intention to include these move changes in a future update?

Dragi2k, 2024-07-16:
> The proper way to do this is to overwrite the macro in your own macro file, otherwise it gets removed on an update.
>
> ```
> [gcode_macro ORIGINAL_MACRO]
> rename_existing: ORIGINAL_MACRO_OLD
> gcode: [...]

That works? I tried the same, but get an error 🤔

bigtalljv, 2024-10-20:
Hi, I have the same issue with crashing into the shaker arm.  Is there a solution  from this discussion?

Dragi2k, 2024-10-21:
> Hi, I have the same issue with crashing into the shaker arm.  Is there a solution  from this discussion?  

I will send it in the evening, but the file is hand changed on every big update

Dragi2k, 2024-10-21:
> Hi, I have the same issue with crashing into the shaker arm. Is there a solution from this discussion?

[mmu_macro_vars.cfg (1).cfg.log](https://github.com/user-attachments/files/174644
[mmu_cut_tip(2).cfg.log](https://github.com/user-attachments/files/17464489/mmu_cut_tip.2.cfg.log)
80/mmu_macro_vars.cfg.1.cfg.log)

you have to remove the .log

bigtalljv, 2024-10-21:
Well I thought of maybe a better idea.  I have a servo mounted cutting depressor arm, maybe I should just move that to where the toolhead goes for blobify-ing and shaking?  Are there specific default Y coordinates for that location?

christi-ko, 2025-09-01:
I ended up putting the cutting depressor arm in the front and having the blobifying and shaking position in thre rear of my voron 2.4.

lokertr, 2025-09-01:
@christi-ko That is how I am currently set up but when it moves back from cutting it crashes, and if the tray is in the way if will crash on the way to cut. I had been using @Dragi2k config such up until I had to update due to another issue. I had hoped something had been done to resolve it.

moggieuk, 2026-10-21:
Closing this as Happy Hare v4 has a few ways to deal with the cutter and the Blobifier being at opposite ends of the bed.

- The cut macro now approaches the pin with guarded moves and, after cutting, backs away from the pin by the safe margin before going anywhere else. Set `variable_safe_margin_xy` in `mmu_macro_vars.cfg` (menuconfig: Macro Variables > Toolhead tip cutting > "Safe margin (X,Y)") to about your toolhead size plus a few mm.
- If the straight-line path still crosses the shaker arm, set `variable_post_form_tip_position` (x, y, z-hop) to a waypoint that the toolhead visits straight after the cut, before it heads to the purge position.
- For full control of the travel path you can set `variable_user_park_move_macro` to your own macro, which is then used for the park and restore moves instead of the default straight line.

The next release also clamps these cutter moves to your axis limits more safely.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk