moggieuk/Happy-Hare #297: How to disable completely MMU?

lettore, 2024-05-20:
Hello, I sometimes need to print without using the MMU, even without using the bypass, so directly from the extruder.
I tried to use MMU ENABLED=0 but I can't go forward printing, it just stuck after my custom PRINT_START.
So I removed the mmu load config lines in PRINTER.cfg, and worked perfectly.
There's something I need to do other than do MMU ENABLED=0 or it's just a bug somewhere?

moggieuk, 2024-05-22:
Actually `MMU ENABLE=0` will disable Happy Hare (not ENABLED), but it is possible that the new print start macros aren't checking for that.  If so it is an oversight.  So, best this is to also comment the special Happy Hare commands in the slicers start-gcode as well.

BTW The "Bypass" functionality is to allow ad-hoc single color prints without disabling.

lettore, 2024-06-13:
I tried again many times but the print always get stuck after PRINT_START macro, it goes in PAUSE state but unable to resume as it says not currently printing.
I will try to debug a little to find exactly where the print get paused, anyway as I said, commenting out happy hare in printer.cfg do the trick, so there's should be no commands calling MMU in the print file, otherwise will complain some gcode macro missing.

3Dgraf64, 2024-06-13:
Commenting out is not enough.  The encoder is still active. Putting "MMU ENABLE=0" at the beginning of the start GCODE in the slicer worked great for me.

lettore, 2024-06-14:
Found out the culprit, I had to disable clog detections in config to be able to print, calling `MMU ENABLE=0` is not enough to turn off all Happy Hare functions. I didn't found a command to change MMU clog detection from terminal, maybe disabling the filament sensor from Klipper will be enough?

lettore, 2024-06-14:
Managed to make it working doing `MMU_ENCODER ENABLE=0` and then `MMU ENABLE=0`.
It's not possible to make clog detection visible as a filament sensor and be able to switch it on and off from the interface?

3Dgraf64, 2024-06-17:
Once the MMU disabled it's not possible to use the encoder alone.

lettore, 2024-06-17:
> Once the MMU disabled it's not possible to use the encoder alone. In the start gcode I enable my BTT filament sensor instead.

That's not the case, just disabling MMU is not enough to disable the clog detection. So I have to disable the Encoder before disabling the MMU, it makes no sense but it works.

moggieuk, 2024-06-26:
I'm interested in seeing the problematic `mmu.log` because disabling the MMU with `MMU ENABLE=0` should make all the runout callbacks a no-op. 

For example this is the handler for the encoder runout/clog.  The only way that `self.is_enabled` can become true is if you were to call `MMU ENABLE=1` or `MMU_RESET`

```yml
    cmd_MMU_ENCODER_RUNOUT_help = "Internal encoder filament runout handler"
    def cmd_MMU_ENCODER_RUNOUT(self, gcmd):
        self._log_to_file(gcmd.get_commandline())
        if not self.is_enabled: return
        self._fix_started_state()

        try:
            self._log_debug("Filament runout/clog detected by MMU encoder")
            self._runout()
        except MmuError as ee:
            self._mmu_pause(str(ee))
```

lorangthierry, 2024-10-08:
i habe the same issue.
17:09:41 . DEBUG: Job State: INITIALIZED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 86400.00s)
17:09:41 > MMU ENABLE=0
17:09:41 . DEBUG: Job State: READY -> STANDBY (MMU State: Encoder: Disabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 86400.00s)
17:09:41 MMU disabled
17:10:20 > _MMU_PRINT_START
17:10:20 . DEBUG: Job State: STANDBY -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 86400.00s)
17:10:20 . DEBUG: Setting servo to move (filament hold) position at angle: 92
17:10:24 Happy Hare initialized ready for print (no filament preloaded)
17:10:24 . DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 86400.00s)
17:14:27 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 7.4mm
17:14:27 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 6.5mm
17:14:28 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 5.6mm
17:14:28 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 4.8mm
17:14:28 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 3.9mm
17:14:28 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 3.1mm
17:14:29 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 2.2mm
17:14:29 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 1.4mm
17:14:29 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): 0.5mm
17:14:29 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -0.3mm
17:14:30 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -1.2mm
17:14:30 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -2.0mm
17:14:30 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -2.9mm
17:14:30 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -3.8mm
17:14:31 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -4.6mm
17:14:31 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -5.5mm
17:14:31 . DEBUG: Automatic clog detection: new min_headroom (< 8.0mm desired): -6.3mm
17:14:31 > __MMU_ENCODER_RUNOUT

MMU_ENCODER ENABLE=0
MMU ENABLE=0

in the start G-Code works for me.
Regards
Thierry Lorang

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4, MMU ENABLE=0 turns off filament monitoring and any runout or clog event while disabled is ignored, so the print carries on without disabling the encoder separately.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk