moggieuk/Happy-Hare #298: Timer too close at random times during print

mikeytdisco, 2024-05-25:
Problem is caused by two problems.  Problem 1 is a flood of syslog messages, and Problem 2 is a failure to read the file metadata.

Problem 1 is an excessive number of .debug messages when metadata is not found. There were over 8e6 lines of:

May 21 00:00:21 voron KlipperScreen-start.sh[901]: 2024-05-21 00:00:21,676 [job_status.py:update_file_metadata()] - Cannot find file metadata. Listening for updated metadata
May 21 00:00:21 voron KlipperScreen-start.sh[901]: 2024-05-21 00:00:21,677 [job_status.py:show_file_thumbnail()] - FilamentalistV1-Voron_v2_350_PIF Standard 0.4 nozzle - MMU-eSun ABS+HS Black (Higher Fan) MMU.gcode
May 21 00:00:21 voron KlipperScreen-start.sh[901]: 2024-05-21 00:00:21,677 [job_status.py:show_file_thumbnail()] - no pixbuf

in syslog.  These messages put an extreme load on the Raspberry Pi 4B.  Klipper graphs showed 400%+.  Normally only shows 40% during a normal print.  If problem 2 is not triggered, the only visual symptom is that the thumbnail is not displayed on the klipper screen.  The code that prints these messages needs to be refactored to only print the messages once per print job.

Problem 2 is in the file_manager.py _run_extract_metadata() function.  It sets a default timeout of 10 which is insufficient.  After I set the default timeout to 30, and uploaded offending gcode file, I observed these two messages:

INFO:metadata:Reading placeholders took 6.06s. Detected gcode by slicer: prusaslicer
INFO:metadata:mmu_server: Writing MMU placeholders,Inserting next position to tool changes took 5.89s

I am not sure what the default timeout should be.

When Problem 2 occurs, the metadata is not loaded.  This causes Problem 1 to put a large load on the system, in my case causing the Timer Too Close error at random times.  I'm sure that if thumbnails were disabled in the slicer, the syslog flood would occur.

moggieuk, 2024-05-25:
On Problem 1, I'm confused. The log lines are coming from KlipperScreen?  I'm not following why the issue report here?  But I agree that excessive file I/O on a rpi with less than perfect SD card would be an issue.  The generation of metadata by moonraker is indeed optional.  It can be disabled by timeout or explicitly by the user. Therefore I think that KlipperScreen should really be coded to tolerate that...

But the excessive syslog messages is definitely useful to know for others experiencing the TTC errors -- I'll add to the HH knowledge base.

On Problem 2: I do add the additional parsing to detect tool changes. In my testing this was usually about 20% of the total time the `extract_metadata` function took so didn't attempt to mess with the moonraker defined timeout.  Can you give me more details on g-code file you are pre-processing.  Ideally, can you send to me?  I'd love to run the timing on my system.

mikeytdisco, 2024-05-25:
Just the Rim Rollers:
[RimRollers.zip](https://github.com/moggieuk/Happy-Hare/files/15444824/RimRollers.zip)

As far as placing the issue in Happy-Hare instead of KlipperScreen -- I did say a few bad words to me after I submitted the issue.

moggieuk, 2024-05-26:
Thanks for the gcode.  I'll probably fix KS in my KS-Happy-Hare fork. I'll also look into increasing moonraker timeout, the problem is that it is hardcoded :-(

moggieuk, 2024-07-08:
Ok. PR was accepted and merged into Moonraker.  Going to close this Issue because this particular cause of TTC should now longer be an issue. 

https://github.com/Arksine/moonraker/pull/862

closed 2024-07-08 by moggieuk