moggieuk/Happy-Hare #308: TYPE B MMU

MG-longshot, 2024-06-06:
How do you define the multiple filament steppers and remove the selector for a type B system?

or better yet do you have a sample config for a type B system?

wuzefacke, 2024-06-08:
would be very interesting, did not find any info/solution

MG-longshot, 2024-06-08:
> would be very interesting, did not find any info/solution

No info yet. Looking at the code to implement it myself just thought they might have already made progress before I re-invent the wheel.

moggieuk, 2024-06-09:
Hi. Yes, it is in progress, has been for a while though.  Search for "TODO" in `mmu_toolhead.py` and you will see the methods for moving steppers in/out of the rail.  The config method is incomplete.  The idea would be to setup as a MultiRail using incrementing gear_steppers, then using the above method to pull all but that required from the rail.

I have a local dev branch with more progress on this but I need to finish up a few other updates first...

MG-longshot, 2024-06-09:
We are currently working on a specific project and would like to include happy hare.

Armored Turtle
https://discord.com/invite/n8dVvdxwyZ

you can message me on the discord as Longshot1970

Thank You

wsidebottom, 2024-06-14:
I would also like to do this using a stepper per color and a filament sensor on each.

moggieuk, 2024-09-04:
Just wanted to keep this Issue alive.  Work on Type-B support is in full swing.  To do it write has meant a restructure of Happy Hare and it is taking longer than I expected...

I apologize to that delay @MG-longshot but I also got bombarded with a TTC issue that I just sunk 60 hours into and some teething issue on the latest HH release.

timboll, 2024-09-14:
really interested in this, and willing to test any early versions as they become available. chiming in to stay subscribed for updates.

moggieuk, 2026-10-21:
Closing this as type-B MMUs are fully supported in Happy Hare v4, and you can set one up from the installer.

Run `./install.sh -i` from your Happy-Hare directory and pick your MMU type. There are ready-made presets for several type-B designs, including Box Turtle, Night Owl, QuattroBox, KMS and EMU. For your own design, choose "Custom Design" and set the selector type to "VirtualSelector". That gives you one gear stepper per gate and no selector. You then add your per-lane entry/exit sensors under "MMU Features / Additions" > Sensors, and pick the gate homing endstop (per-gate exit sensor, or a shared hub sensor) under "Endstops and Bowden movement". The installer generates the full config for you.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk