moggieuk/Happy-Hare #310: MMU3

raulpuente, 2024-06-07:
Hello, i'm using Klipper with mk3s+ and i have mmu3. Is it supported?

Thank you

m4gn3to, 2024-06-07:
I'm in the same boat.
I have the MMU2s modded to support mmu3 firmware....

I don't know if it is supported but I can help with my printer and mmu if the devs want...

moggieuk, 2024-06-09:
Similar question to Issue #309.    I stalled on work to add support.  Its really close but needs a little rework in making the selector and servo functionality more modular.  The selection process on the MMU3 isn't just move a "selector" stepper, then  grip filament.  It really combines the two my moving the selector stepper to either an engaged position or slightly off to release filament.

I hope to pick up this work soon, it would help if I have an MMU3.  Maybe Prusa would send me one!?!

raulpuente, 2024-06-09:
Nice!, If you need help trying something, please don't hesitate to tell me.

m4gn3to, 2024-06-09:
Yes!
I can help too.
I can test whatever you want.

m4gn3to, 2024-07-04:
Can you publish a new branch with what you already have so we can try to help?

jfestrada, 2024-07-04:
Hi @moggieuk I can help or at least try to help with the development for the MMU3, I was involved on the support of MK3s+ with RatOS and I can tell to you that there are a ton of people interested on the support of the MMU3 with your project, so if you have a development branch, I can try to help there, also I have a deep knowledge of the prusa firmware for the MK3S+ and the MMU3.

ErdbeermarmeladebrotmitHonig, 2024-08-15:
> I hope to pick up this work soon, it would help if I have an MMU3. Maybe Prusa would send me one!?!

Do you have one now? Or have you any update on this?

m4gn3to, 2024-10-24:
Any updates?
Any way we can help?

eoyilmaz, 2025-02-09:
I'm the developer who ported [MMU3 to Marlin](https://github.com/MarlinFirmware/Marlin/pull/26635), I think this is going to be my next project.

m4gn3to, 2025-02-10:
Great!!
I hope this means a big step forward for the project.
I haven't even been able to flash the Klipper firmware to the MMU board...

m4gn3to, 2025-02-11:
Is there any way we can help you?
You plan to use the original MMU board and flash klipper?

agravelot, 2025-02-11:
> I'm the developer who ported [MMU3 to Marlin](https://github.com/MarlinFirmware/Marlin/pull/26635), I think this is going to be my next project.

Same here

eoyilmaz, 2025-02-11:
>  You plan to use the original MMU board and flash klipper?

To be honest, I have no idea, I don't know how Happy Hare is interfacing with the other MMU boards or with Klipper, I was trusting that a serial communication protocol as in Marlin would have been enough.

calini, 2025-02-12:
I've got a klipperized Ender and an MMU3 clone sitting around, I'd love to contribute if I can

m4gn3to, 2025-02-13:
If you know the serial protocol of the original MMU board then I think you can use:
[gcode_macro GET_DEVICE_STATUS]
gcode:
    SEND_SERIAL PORT=/dev/ttyUSB0 MSG="GET_STATUS\n"
    RESPONSE = RECEIVE_SERIAL PORT=/dev/ttyUSB0 TIMEOUT=1.0
    M118 Device Status: {RESPONSE}

but I don't know if the MMU board will accept that communication through usb port or we will need to use the serial port and any kind of connection (serial to usb adapter maybe) ??

Or maybe using a python script that interacts, receives the gcodes from klipper / happyhare, send them through serial and response to klipper through the klipper api ....

Just thinking out loud...

jfestrada, 2025-02-13:
Even if you use a serial connection to communicate with the MMU board, we need to adapt Happy Hare to work with the selector and idler because currently it is not supported. Obviously the best way is to use klipper directly on the MMU board too. In may point of view these are the tasks that should be covered: 

- Flash klipper on the MMU board (so there is no need to buy and additional board)
- Add the support for the idler with the configuration options i.e. degrees between bearings
- Add support for specific options of the selector i.e. filament blades, presence of filament (finda sensor), length of the selector travel, etc ...

I'm pretty sure that there are many others tasks to accomplish but if we go step by step we could do it.

Br.
Juanfra

eoyilmaz, 2025-02-15:
> Or maybe using a python script that interacts, receives the gcodes from klipper / happyhare, send them through serial and response to klipper through the klipper api ....

Yes, with my limited knowledge, I was thinking the same.

> Even if you use a serial connection to communicate with the MMU board, we need to adapt Happy Hare to work with the selector and idler because currently it is not supported.

The selector and the idler should be irrelevant from Klipper/Happy Hare perspective. It is all controlled by the MMU board and as far as I know other than some error messages the selector/idler statuses are not exposed to the controller directly. You can setup the MMU firmware to control an arbitrary number of materials, and all it requires to switch between them is a message from the serial port (something like: L0*73). Then the MMU board takes the action and responds with a status message about the last action and then continues to responds with the current status messages, which is interpreted by Marlin and acted upon. So, in Marlin, the MMU board and the printer firmware are doing a very well coordinated choreography.

On the other hand, with my limited knowledge about Klipper + Happy Hare, I might not be understanding some key concepts, and might be thinking in the wrong way.

Freakazo, 2025-02-20:
Hi I just got a MMU3 as a gift, but I'm running klipper on a Prusa MK3S+ on a raspberry pi. I'm still assembling the MMU3 and am trying to figure out how I'd be able to use it software wise ahead of time to prevent my printer having any long periods of downtime. Seems like I'll have to help contribute to make it happen or just wait until a solution comes up.

I'm trying to understand the topology and how things communicate. Is my understanding is correct:?
 1) Where does happy hare run? I've got a raspberry pi with klipper, I think I use that to flash klipper on the mainboard as well. Does klipper on the raspberry pi talk to klipper on the mainboard for the prusa?
 2) How do we communicate to the MMU3 board.
   2.a): would happy hare talk to that board using the MMU3's protocol, e.g. sending through the required message to change from filament 1 to 2.
   2.b) or is happy hare more of a firmware that runs on the MMU3's board directly controlling the servo's, which is communicating with the raspberry pi through a protocol?
3) Is there a preference between the two ways?

m4gn3to, 2025-02-20:
Hi.
1.- Happy hare runs in the raspberry, and uses klipper to comunicate with the boards. Yes it also can flash klipper on the boards.
2.- The ideal solution is to flash klipper on the MMU board to use klipper protocol, we are trying to acheave that. A posible solution is to use the Prusa's MMU firmware and comunicate through serial commands...
The other problem is that Prusa's MMU does not have any servo, it uses 3 steppers and happy hare is not able to use it (for now)

eoyilmaz, 2025-02-24:
### Building Klipper for MMU Board

So, as I see, flashing Klipper firmware to MMU board is not possible anymore due to the size limitation.

The latest Klipper code and a normal config builds to a 53 kB firmware for m32u4 and I can't reduce the size under 40kB by disabling everything and enabling `-Os` optimization, this is way over the 32 kB that is available on the m32u4 MCU. So, we are left with the 2 options:

.

### Option 1: Using the original MMU firmware and Happy Hare communicates the MMU board over a serial connection using MMU protocol

This will most probably require one to detach the UART cable from the printer board and attach it to the Raspberry Pi (in to the correct pins btw). And this will most definitelly fry the Raspberry Pi as it is 3.3v logic while the MMU board is 5V logic... and it happened to me while I was trying to debug the serial communication during my initial development for Marlin... So, a 5V to 3.3V logic converter will be needed, and I use one in my current setup between the SKR 3 EZ board and the MMU board.

On the other hand, I'm not sure if one can tweak the MMU firmware to make it to communicate over the USB. Though, I looked at the MMU firmware and the USB serial can be enabled to carry debug messages, so it might still be possible.

.

### Option 2: Replacing the MMU board with a more sensible 32bit board and use Klipper + Happy Hare as usual

That sounds the more sensible way of doing this, as it will remove a bunch of technical difficulties both in hardware and software side. This will at least allow us to concentrate on updating Happy Hare for the MMU configuration, which as people pointed out doesn't have a servo motor and as I understand Happy Hare logic needs modifications.

I have a couple of SKR Mini E3 v3 boards that I can utilise for this purpose.

m4gn3to, 2025-02-25:
Hi.
As @jfestrada said, in his repo you can see the config we are using to flash klipper to MMU board:
https://github.com/ComunidadMaker3D/CM3D-MMU-klipper/tree/dev/config/mmu_board_prusa
The size is not the problem. The real problem is that prusa decided to use an "intermediate" chip to comunicate with the "enable" an "dir" pins of the drivers.
If we are able to flash klipper to the MMU an use it "as is" many people will be very happy, they can use their MMU units with klipper without need to buy anything.
Anyway, we need to modificate the logic of Happy Hare to make it compatible with the "movements" of MMU2s/MMU3, this 2 works are independent and needed to be done.
The final goal would be using the "magic" of Happy Hare with the MMUs units

jfestrada, 2025-02-26:
@eoyilmaz finally I purchased an SKR PICO in order to substitute the mmu board from Prusa, why? just because the intermediate chip, the size is not a problem because you can remove some parts of klipper that are not needed. @trevjonez already developed the klippy extra modules for that but the integration into klipper finally was not "accepted" [PR 1047](https://github.com/Klipper3d/klipper/pull/1047). In my opinion will be better to focus on the MMU2/3 functionality instead of try to solve/hack the poor design that Prusa made for this board.

eoyilmaz, 2025-02-26:
> In my opinion will be better to focus on the MMU2/3 functionality instead of try to solve/hack the poor design that Prusa made for this board.

@jfestrada I think the same...

eoyilmaz, 2025-02-26:
I started working on the code, and I have a question; does Happy Hare need to be supporting Python 2 or can I use Python 3 only syntax?

agravelot, 2025-02-27:
Might be usefull : https://github.com/KalicoCrew/kalico/pull/560

eoyilmaz, 2025-02-27:
@agravelot a very interesting project, as I see it has lots of interesting features added on top of upstream Klipper, it is Python 3 only and they even do `black` formatting, which I had a similar PR to Klipper last year, about PEP8 conformation and it was rejected.

eoyilmaz, 2025-03-09:
Let me do some status updates...

So, because I'm not very familiar with Klipper and Happy Hare at all, I spent some time to familirize myself with them by installing and running the old DIY MMU config on my MMU3 which is solely depending on Klipper and custom GCode macros. btw, I'm happily surprised that the Klipper GCode Macros are basically Jinja2 templates, which I'm very familiar with. I skipped installing Klipper to the MMU board, and used an SKR Mini E3 v3 board instead. After fidiling around with the board config and macros for a couple of days, the setup was properly working, albeit not very robost compared to the original MMU firmware. But, this helped me to setup the stepper cables for the SKR Mini E3 v3 board, run the macro with sensorless homing, setup the required bowden tube lengths, motor currents, slot distances etc. So I know that my setup is working in its current form and when I switch to Happy Hare if I have any issues, I will not need to look into the hardware setup at all. This will help me debugging the problems much more easier... at least the plan is that.

Then I switched back to the Happy Hare code base, and now going over it. As I understand, we need to implement a custom `LinearSelector` that also handles the idler movements (as mentioned in above comments). Still going over the code, reading and understanding the code structure etc..

While doing the things mentioned above, I'm introducing a lot of changes, mostly related the to the fact that the code is not very... Pythonic... in many ways. And unfortunatelly, proven with my past experience on many projects that I contributed in, most of the times these kind of changes are not very... welcomed... by the project owners. Let's see how this one will go.

m4gn3to, 2025-03-10:
Great!!
We are trying to run de old DIY MMU too, can you share your config?  We are using BTT SKR Pico boards... 
Thanks in advance!

eoyilmaz, 2025-03-10:
@m4gn3to sure: https://github.com/eoyilmaz/klipper-configs.git

Here is the SKR Mini E3 v3 pinout schema to help you map my setup to your SKR Pico board:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20E3%20SKR%20MINI%20V3.0_PIN.pdf

m4gn3to, 2025-03-10:
Ok Thank you !!!!!

LuisDiazUgena, 2025-04-22:
Hi!

So I just readed the whole thread and trying to understant the current state of the project. As far as I understood:

- Is possible to run it with stock mmu3 board with klipper, but the serial implementation is not done
- current state is swaping mmu board with an SKR mini e3 board and running [this ](https://github.com/eoyilmaz/klipper-configs.git)for configuration

m4gn3to, 2025-04-24:
Yes, more or less.
With the stock mmu3 board we could not flash klipper directly to the board, so eoyilmaz was trying to "send" serial commands to it, but I don't know how far he has go...
With the skr mini or any other board with klipper flashed and starting with the configuration you point to, it is "posible" to make it work, but we had "personal problems" and can't keep working on it...

LuisDiazUgena, 2025-04-25:
Thanks @m4gn3to 

I just got my bear running on klipper and I'm trying to adjust everything to my liking, then I will start working on seting up the mmu3. I also have a pico board like you have. Did you do something regards the configuration files before you had to stop working on it?

m4gn3to, 2025-04-29:
My main problem was the speed... when you ask it to make home, it takes "forever".
I can send commands correctly and the speed is fine, but when I use variables it takes too much longer, I don't know why.
I had to check the hommings, there were places where you need it to be sensorless and does not stop...

Freakazo, 2025-05-12:
Howdy,
So I've been working on adding a serial bridge to klipper, so that you can setup additional MCU's to pass-through coms with each other, it's inspired by previous attempts, hopefully I'll be able to get it merged upstream.
```
[mcu]
serial: /dev/serial0
restart_method: command

[mcu mmu3]
bridge_mcu: mcu
usart_number: 2
baud: 76800
# All commands for mmu3 will be routed through the bridge_mcu to serial port 2
```

So this should work on a stock mk3s+ and MMU3 _(plus the pi running klipper)_.
Just got a [working version](https://github.com/Freakazo/klipper/tree/transparent_serial_bridge) of this done, so I'll start to work on dealing with things like the shift register on the MMU3 control board.

eoyilmaz, 2025-05-13:
> eoyilmaz was trying to "send" serial commands to it, but I don't know how far he has go...

unfortunatelly I've been pulled to a different project right now and I'm not able to reserve much time to this project. But it is still on my list...

Freakazo, 2025-06-13:
Update from my side, everything is still very much in a draft state but its making progress 🤞.

For Happy Hare:
- Added support for the idler, I'm implementing this as a class that has the same interface as `LinearSelectorServo`, calling it `LinearSelectorIdler`. It pretty much just checks which gate is selected when the `servo_down()` command is called to know the exact position to go to, and added in the calibration commands. The idler is implemented as the "Z" axis on the MMU toolhead/kinematics.
- Todo here is to update the installer script to give sane defaults for MMU3, add filament in selector checking (Pinda sensor on MMU3), and do some actual printing at some point.

For Klipper:
- As mentioned in the edits in the previous comment, I think moving the shift register support from firmware to happy hare _(extras)_ will have to happen, otherwise people will have to keep using custom firmware on the stock MMU3 board.
- The serial bridge implementation is still giving me headaches and is pretty unreliable at the moment. I've added bitmasks and had to play around a lot with the size of serial bridge messages/serialqeue settings/init procedure to get it working semi reliably. It still requires me ~4 firmware restarts just to get everything initialized. Then during runtime there's still timing too close/failure to allocate move errors. I have a High-Low dual direction voltage converter board for serial communication, I might switch to using this and connecting directly to the RPI if I get stuck on the serial bridge.

ErdbeermarmeladebrotmitHonig, 2025-11-03:
Hi,
this weekend I tried to bring the original MM control board from Prusa to work. Therefore, I flashed Klipper onto the MK3S and onto the MM Control board. For the small ATmega32U4 I disabled most of the features (SPI and software SPI is needed!!!) in order to fit the Klipper firmware on it.

Next, I implemented a driver for the 74HC595 shift register using software SPI. You can find this implementation here: [sr74hc595.p](https://github.com/ErdbeermarmeladebrotmitHonig/klipper/blob/master/klippy/extras/sr74hc595.py) Acutally, this is not cleaned up yet, it is just the first working example and I am not sure whether all of the functions are required.

Using the following config, I was able to control the 10 LEDs (5 green, 5 yellow):
```
[mcu mmboard]
serial: /dev/serial/by-id/usb-Klipper_atmega32u4_12345-if00

[respond]
default_type: command

[sr74hc595 sr1]
spi_software_sclk_pin: mmboard:PC7
spi_software_mosi_pin: mmboard:PB5
spi_software_miso_pin: mmboard:PE2 # dummy pin
latch_pin: mmboard:PB6
chain_count: 2
invert: false

[output_pin led_out_1_1]
pin: sr1:Q7
value: 0

[output_pin led_out_1_2]
pin: sr1:Q6
value: 0

[output_pin led_out_2_1]
pin: sr1:Q15
value: 0

[output_pin led_out_2_2]
pin: sr1:Q14
value: 0

[output_pin led_out_3_1]
pin: sr1:Q13
value: 0

[output_pin led_out_3_2]
pin: sr1:Q12
value: 0

[output_pin led_out_4_1]
pin: sr1:Q11
value: 0

[output_pin led_out_4_2]
pin: sr1:Q10
value: 0

[output_pin led_out_5_1]
pin: sr1:Q9
value: 0

[output_pin led_out_5_2]
pin: sr1:Q8
value: 0
```

There are two more problems, which need to be solved. First, these pins can now only used for digital outpus like shown above, but they cannot be used as pins for a stepper motor. However, on the MM control board the DIR and EN pins for all three motors are wired to the shift register.
Second, the Klippers stepper module does not allow to have STEP and DIR pins for one stepper motor comming from different chips. The wiring on the MM control board however controls all STEP pins direclty from the MCU and the DIR and EN pins via the shift register.

I am not sure how to solve this problems. I don't want the solution to be too intrusive to the Klipper source code, so I don't want to change the Pins, MCU or stepper source files from Klipper.
I thought of an additional module, which let's you group pins from different chips in a new "virtual" chip. But I am not really sure whether this would solve the problem or not ...

Here is the config for the stepper motors which is NOT WORKING right now:
```
[manual_stepper pulley_stepper]
step_pin: mmboard:PB4
dir_pin: !sr1:Q0
enable_pin: !sr1:Q1
microsteps: 16
rotation_distance: 22.857
velocity: 20
accel: 10
endstop_pin: ^mmboard:PF6

[tmc2130 manual_stepper pulley_stepper]
cs_pin: mmboard:PC6
run_current: 0.650
sense_resistor: 0.110

# Y : MMU3 idler with the 5 ball bearings
[tmc2130 manual_stepper idler_stepper]
cs_pin: mmboard:PB7
run_current: 0.580
sense_resistor: 0.110
stealthchop_threshold: 999999
diag0_pin: ^mmboard:PF0
#driver_sgt: 128

[manual_stepper idler_stepper]
step_pin: mmboard:PD6
dir_pin: !sr1:Q4
enable_pin: !sr1:Q5
microsteps: 16
rotation_distance: 128
velocity: 100
accel: 80
endstop_pin: tmc2130_idler_stepper:virtual_endstop

# X : MMU3 selector
[tmc2130 manual_stepper selector_stepper]
cs_pin: mmboard:PD7
run_current: 0.580
sense_resistor: 0.110
stealthchop_threshold: 999999
diag0_pin: ^mmboard:PF1
#driver_sgt: 96

[manual_stepper selector_stepper]
step_pin: mmboard:PD4
dir_pin: !sr1:Q2
enable_pin: !sr1:Q3
# 80 step/mm
microsteps: 16
rotation_distance: 8
velocity: 30
accel: 200
endstop_pin: tmc2130_selector_stepper:virtual_endstop
```

Zangief1977, 2025-11-07:
https://github.com/Prutsium/3D-Druckerplausch-Klipper/commits/main/

JiriPansky, 2025-11-26:
@eoyilmaz
First of all, thank you for the conversion and adaptation to Klipper.
After trying your version for Marlin (I had trouble making the right filament tip), I'm now trying the Klipper version. It's working on the table so far, but I have a few questions about the technical setup:
1) did you use direct drive extruder? Or bowden extruder?
2) The BTT Smart filament sensor is used in the configuration. where is the sensor installed? with the direct drive extruder it has to be in front of the extruder? how far from the extruder? what happens to the remaining distance between BTT SFS and the extruder? If the BTT SFS triggers, the filament is still not in the extruder... with Prusa's PINDA, the sensor controls directly when the filament is gripped by the extruder.
3) Which filament cutter is used?

Alexande76, 2025-12-09:
Hi ErdbeermarmeladebrotmitHonig can you post a mini guide on how to configure mmu3 prusa for klipper mainsail. I can install happy_hare but the configuration is in error thanks

[image]
[image]