moggieuk/Happy-Hare #311: How to install Happy-Hare to Creality K1c?

tent228, 2024-06-12:
May I ask how to install happy-hare to the rooted Creativity K1? I once attempted to copy some *. py to the corresponding extra directory and successfully drove the encoder and erec, but did not attempt anything else. However, running install.sh directly did not succeed. I installed the bash command to modify root user authentication in the sh file, but it prompted that klipper.service was not detected. How can we continue?

m4gn3to, 2024-06-13:
Do you have klipper installed?
I don't know Creality K1c but as far as I know you need an external "CPU", like a raspberry, running klipper. After that you can try to install Happy-Hare (in that "CPU") to control your mmu

ningpj, 2024-06-16:
The installer includes arguments you can use to override common settings like the target install directory as well as the name of the service for non default installations of klipper.  use ``-a <name of your klipper linux service>`` to change the prefix.  it appends ``.service`` so ``-a K1_Klipper`` will check for ``K1_Klipper.service``

```Usage: ./install.sh [-a <kiauh_alternate_klipper>] [-k <klipper_home_dir>] [-c <klipper_config_dir>] [-m <moonraker_home_dir>] [-b <branch>] [-r <repetier_server stub>] [-i] [-d] [-z]

-i for interactive install
-d for uninstall
-b to switch to specified feature branch (sticky)
-z skip github check (nullifies -b <branch>)
-r specify Repetier-Server <stub> to override printer.cfg and klipper.service names
-a <name> to specify alternative klipper-service-name when installed with Kiauh.
-c <dir> to specify location of non-default klipper config directory
-k <dir> to specify location of non-default klipper home directory```

k1-801, 2024-07-17:
> You "may" also need to override other settings

Unfortunately, despite the Creality K1 series running a version of Klipper under Linux, the system is way too different from a "normal" Klipper installation on a Pi. There is no `apt`, there is no `systemd`, klippy runs under root.

Furthermore, klipper that comes pre-installed on the machine is heavily modified by Creality in ways unknown to humanity, and I never tested this on the original firmware. I am using a "tenso mod" on my K1 Max, so I had to replace the firmware with "clean" Klipper, so, although this should work, it may fail for other people.

There is STILL a way to install HappyHare on it, but it involves a custom install script, ran with proper directories.
1) The helper script is required 
2) One needs to install Entware through helper script, and then Bash through Entware (opkg). I believe git is also installed through opkg.
3) The adapted install script should be placed in the cloned repo, replacing the original. The version should be updated from 2.51 to the one that was in the original (currently 2.6). The script can be found here: https://pastebin.com/aM3kAAJ6
4) The script runs as: `bash Happy-Hare/install.sh -k /usr/data/klipper/ -c /usr/data/printer_data/config/ -m /usr
/data/moonraker/moonraker/ -i`

@moggieuk would it be possible to make some sort of a switch to detect if the install script runs on a MIPS machine, and, if so, adapt the procedure according to the script in the pastebin, keeping in mind the issues listed above (preferably with default paths also adapted to the proper location for the K1 series as listed above) ?

FelixOstermeier, 2024-08-06:
I'm trying to install Happy Hare with the modified script on my K1C, however my klipper directory is not under usr/data. Any idea why there is no klipper directory there, even though I installed moonraker etc with the helper script?

k1-801, 2024-09-13:
Technocally, we already have a PR :)
#388 was already linked & tested, it works (but the board firmware needs to be built with appropriately old sources - not newer than december 2023)

closed 2024-09-21 by moggieuk

jacksky6, 2024-09-29:
@k1-801 Can you share the MMU package configuration file of K1? Mainly the data of the tip forming. When I test the filament unload, it is always easy to get stuck in the extruder.

k1-801, 2024-10-03:
> @k1-801 Can you share the MMU package configuration file of K1? Mainly the data of the tip forming.

I am currently on a trip, will send the data at the weekend.
The tip forming is tricky to configure, but I found the default settings work fine with ABS if Ramming is turned off fully in the slicer (it turns out, the instructions for Prusa Slicer are outdated - these days you need to open the graph abd reduce the counter on the right to 0).

I believe, the only changes I made that had any effect were the cooling tube position (reduced) and gear-to-nozzle distance (didn't affect tip forming, but significantly improved unloading reliability.

k1-801, 2024-10-07:
> Mainly the data of the tip forming.

I only use ABS, so my settings may not  be valid for you if you use other materials.

```
toolhead_extruder_to_nozzle: 65
toolhead_entry_to_extruder: 8 ; ( https://www.printables.com/model/686275-k1-extruder-with-filament-runout-sensor-remixed )
toolhead_ooze_reduction: 18 ; ( Calibrate yourself, this is a "safe" value)
```

Additionally, I disabled ramming completely, as no matter what I did, it was leaving a blob of filament on the nozzle before it would even begin to unload:

```
variable_ramming_volume            : 0
variable_ramming_volume_standalone : 0

variable_cooling_tube_position  : 15
variable_cooling_tube_length    : 10
variable_initial_cooling_speed  : 10
variable_final_cooling_speed    : 40
variable_cooling_moves          : 3
```

PR #388 (2024-09-21): Changes for Creality K1 series
The Creality K1 series of 3d-printers runs on a MIPS CPU, under a heavily trimmed Linux system akin to that on WiFi routers. The current install script will not run on them.
Furthermore, on those printers, there are no users other than root present, the directories for klipper are always different from what you expect on a Pi-based system, the services run as init scripts (with no systemd present).

This PR suggests changes necessary for checking if the srcipt runs on such a system and adjusting the folders accordingly. It addresses the question in #311 and is si [... cut]