moggieuk/Happy-Hare #312: Internal error during connect: 'MmuToolHead' object has no attribute 'requested_accel_to_decel'

WKn1975, 2024-06-14:
I try to fix in different way, but it doesn´t matter to give the configs this requested value - no change.

Any Ideas welcome.

I know this is deprecated since month.
in toolhead.py there are rules for that, but no solution.

Thankjs for help

BR

moggieuk, 2024-07-08:
You will need to update Klipper.  Klipper made some internal changes to the toolhead code and I needed to update HH to keep up. Trying to also be backward compatible creates too many test combinations..

WKn1975, 2024-07-08:
Thank you for answer
HH was uptodate as I create the issue.
For Klipper it is more complicated, because it is a custom repositiry
for an Sovol SV 04 IDEX Printer
https://github.com/Bully85/Sovol-SV04-Klipper

So I think the solution is setup a "standard Klipper" and loose IDEX or waiting for an update for this machine, because at the moment I´m not able to solve it self.

I also waiting for my delivery of SV08, and for this I can hopefully use the ERCF - there are already existing official Klipper repos online existing.
BR Werner

moggieuk, 2024-07-10:
Ah I see. I suspect the custom Klipper must be periodically updated so hopefully it would be resolved soon.  I'm happy to accept a PR to make this work .. I'm sure a backward compatibility check is possible but I suspect this may occur again in the future as Klipper evolves.  I'dd take one quick look as see if a backward compat check is possible..

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

This came from an older custom Klipper build; I added a compatibility fix at the time, and the v4 toolhead code no longer uses that setting at all.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk