moggieuk/Happy-Hare #313: Internal error on command:"_MMU_PRINT_START" on Danger Klipper

jackjohnsonuk, 2024-06-14:
I get a `Internal error on command:"_MMU_PRINT_START"` followed by `MCU 'mcu' shutdown` on Danger Klipper:Bleeding-Edgev2.

Speaking to "Rogelz" from the Danger Kilpper team, he's advised me that it's to do with the new way they're calling e_pos.  They've tried to help me by suggesting an amend on e_pos, as below:

```
 gear_rail.set_trapq(extruder.get_trapq())
 e_pos = extruder.last_position
 # rogerlz's fix
 if isinstance(e_pos, list):
   e_pos = e_pos[0]
 gear_rail.set_position([e_pos, 0., 0.]
```

That seems to solve HappyHares initial complaint, but brings up a new one: "Active extruder does not have a stepper"

I've attached the klippy log below following the initial fix from above.

I appreciate this is a small problem, generated by "upstream" changing some pretty substantial working, so understand if it's beyond your scope to try to problem-solve, for a very small minority.

I appreciate your help/consideration anyway. Thanks for the project so far :)

[klippy_9.log](https://github.com/user-attachments/files/15844474/klippy_9.log)

moggieuk, 2024-06-26:
At the moment I don't have the time to support Danger Klipper. Although I love some of the new features I worry about this fork from what is a very stable klipper. I really hope the two projects can merge at some point.

That said, I will look at this when I get time. Probably a simple change.  Note that Happy Hare deliberately configures the extruder without a stepper initially and then adds one that is capable of homing.  You can disable this be adding:
```yml
homing_extruder: 0
```
in your `mmu_parameters.cfg`.  This MIGHT work around this issue

moggieuk, 2024-07-10:
I did test HH with Danger Klipper and everything worked.  I read again your Issue and you are talking about a specific branch "bleeding edge" of DK!   A development branch isn't something I can support.  I might have opportunity if it becomes mainstream DK but then I would really hope for some help from the community. 

I'll keep this ticket for future reference..

closed 2024-12-23 by jackjohnsonuk