moggieuk/Happy-Hare #314: Incorrect number of values for PURGE_VOLUMES

lettore, 2024-06-15:
Hello,
I'm using Orca Slicer printing a simple dual color with just one material swap, so no purge tower.
I added this in Machine start Gcode `MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!`
But when starting the print it stops with error.
`19:20:42 !! Incorrect number of values for PURGE_VOLUMES. Expect 1, 6, 12, or 36, got 4`
My Gcode file have this line `MMU_START_SETUP INITIAL_TOOL=0 REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!` so it looks like this line is modified by Happy-Hare while loading.
In the Gcode on the printer `PURGE_VOLUMES=!purge_volumes!` becomes `PURGE_VOLUMES=0,321,159,0`, I have no clue where it takes this values, the only way for print is to remove `PURGE_VOLUMES` from the Gcode.
It's not possible to add a logic to not create wrong data that prevent successful printing?

ningpj, 2024-06-15:
the number of colors defined in your slicer needs to match the number of carts your ERCF has even if you dont plan in using them.  E.g. if 8 you need to define 8 in orca

lettore, 2024-06-16:
It makes sense if the slicer is filling those numbers, but as it's done by Happy Hare, it should not be possible, as it correctly recognizes only 2 filaments, to add on its dummy data for others? I mean the data necessary for the 2 tools is there, for the other unused tools why the slicer should be filled if 6/8/12 filaments just for Happy Hare to be happy if they will be not used in printing?

ningpj, 2024-06-16:
The Happy Hare Moonraker preprocessor pulls the color breakdown from generated slicer variables. Currently the preprocessor and happy hare logic requires the slicer to pass the same number of gates as your ERCF/MMU has configured whether you use them or not.  I have 12 gates so have the following defined in orca for the printer.  Unused gates have a default color assigned.  

[image]

moggieuk, 2024-06-26:
Note I've also patched Happy Hare to issue a clear warning of this mismatch on print start.

closed 2024-11-29 by github-actions[bot]