moggieuk/Happy-Hare #316: Happy-Hare installation fails if home directory path has dots in it

avas, 2024-06-17:
Hey 🙂
I've set up Klipper on my printer using `a.vas` as my Linux username. So my home directory is `/home/a.vas`, and my printer_data is located in `/home/a.vas/printer_data`. So when I try to install Happy-Hare on that machine, `install.sh -i` just crashes when backing up my existing configuration:
```bash
a.vas@voron24:~/Happy-Hare $ ./install.sh -i

...

Reading default configuration parameters...
Copying original printer.cfg file to /home/a-20240617_232602.vas/printer_data/config/printer.cfg-old
cp: cannot create regular file '/home/a-20240617_232602.vas/printer_data/config/printer.cfg-old': No such file or directory
a.vas@voron24:~/Happy-Hare $ pwd
/home/a.vas/Happy-Hare
```

It probably fails because `nextfilename` calculates wrong filepath, and it probably does so because it doesn't seem to play well with paths that have more than one dot (i.e. they have another dot before the last one in `printer.cfg`).

I managed to work around this issue like this:
```diff
function nextfilename {
    local name="$1"
    if [ -d "${name}" ]; then
-        printf "%s-%s" ${name%%.*} $(date '+%Y%m%d_%H%M%S')
+        printf "%s-%s" ${name} $(date '+%Y%m%d_%H%M%S')
    else
-        printf "%s-%s.%s-old" ${name%%.*} $(date '+%Y%m%d_%H%M%S') ${name#*.}
+        printf "%s-%s.%s-old" ${name%.*} $(date '+%Y%m%d_%H%M%S') ${name##*.}
    fi
}
```
...but not sure if this is the right way to fix it.

moggieuk, 2024-06-26:
Yeah a dot in a path would be unusual -- you are the first!!

Could you create a PR for this. Happy to incorporate.

avas, 2024-07-02:
Yup, it works for me now. Thank you!

closed 2024-07-02 by avas